Infineon/XMC-for-Arduino #1: Support for the XMC4500 Relax

nmaas87, 2017-03-01:
Dear Infineon engineers,

thanks a lot for giving out a XMC Boardspackage for Arduino!
Could you please add the XMC4500 Relax and XMC4500 Relax Lite series as well? I would love to experiment with this boards - from this new perspective.

Thanks a lot for your great work 👍

Infineon, 2017-03-02:
Thank you very much for your great feedback! Very nice that you are already looking and asking for the other boards.

At the moment, we are having hands full enabling different sensors within the Arduino environment for the embedded world. After that based on the feedback of the community and your input we will decide for the next steps of development.

Best regards,

The Infineon Team

Infineon, 2017-03-28:
We are currently preparing the next steps and facing the problem that the XMC 4500 Relax Kit does not support VCOM with the on-board debugger (currently no FW supporting this is planned).
We need to check how to proceed here to allow serial communication within the Arduino IDE.

mhollfelder, 2017-07-19:
We finally decided not to integrate the XMC4500 right now as the missing VCOM makes an Arduino integration hardly possible right now.
I hope you understand this :)

We will keep this in mind and see whether we will find a workaround or not in the future.
Sorry for that!

Best regards,

Manuel

closed 2017-07-19 by mhollfelder