Infineon/XMC-for-Arduino #3: serialEvent() not working on XMC 4700

ventilator, 2017-03-27:
I tried to implement an Serial Event Handler (https://www.arduino.cc/en/Tutorial/SerialEvent)
The XMC implementation does not call serialEvent from a scetch.
The HardwareSerial.cpp does not implement a call of its function void serialEventRun(void) from line 32.
I guess somewhere further below (in void HardwareSerial::IrqHandler ?) there is a missing call of serialEventRun()

Expected behavior:
XMC would write back "Hello!" on the console if a Serial input is given.
Seen behavior:
Noting happens.

Consider the following code
`
void setup() {
  // put your setup code here, to run once:
  Serial.begin(250000);
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(1000);
}

void serialEvent()
{
  Serial.println("Hello!");
}
`

mhoechner, 2017-03-28:
Hi,

the serialEvent is actually not a real Interrupt call back of the HardwareSerial. 
It is just a function, that is called after the "loop()"-part of every Sketch, if there is serial data available.

Also refer to the [Arduino Forum](https://forum.arduino.cc/index.php?topic=166650.0) for details.

In the current implementation the check if there is serial data available is missing. 
I added it here #6.

Infineon, 2017-03-28:
We will collect the pull requests in branch 'develop' and bring a new release by end of the week fixing this issue.
In the meantime, please use the develop branch to circumvent the problem.
Thanks to everyone for contributing here, great work!

Best regards,

The Infineon Team

mhollfelder, 2017-04-07:
The branch 'develop' has been merged into the 'master' branch and a new [release](https://github.com/Infineon/XMC-for-Arduino/releases/tag/V1.0.2) for Arduino is available which should fix this issue.
Please note that also other issues have been fixed, see the release notes for more information.
In case there are more problems, let us know :)

closed 2017-04-07 by mhollfelder