Infineon/XMC-for-Arduino #4: analogRead() not working for XMC 2Go

thihaElec, 2017-03-28:
Hi,

The following code is not working in XMC 2Go. Could you please test on your side?

`void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
//  analogReadResolution(10);
//  wiring_analog_init();
}

void loop() {
  // put your main code here, to run repeatedly:
    int sensor = analogRead(0);
    Serial.println(sensor);
    delay(3000);  
}`

mhoechner, 2017-03-28:
Hi thihaElec,

your example is not working because you try to use Pin 0 as an analog Input.

But for the XMC2GO the only analog Input Pins are Pin 12 and 13.

Here you will find a detailed description of the XMC2GOs pinout:
[https://github.com/Infineon/XMC-for-Arduino/wiki/XMC-2Go](https://github.com/Infineon/XMC-for-Arduino/wiki/XMC-2Go)

thihaElec, 2017-03-28:
Hi,

Thanks for your prompt reply. It works. (Honestly, I did try with 12 too but didn't work at that time. anyway..) Thanks

Best regards,

closed 2017-03-28 by thihaElec