Infineon/XMC-for-Arduino #5: SD.open() not returning File object

thihaElec, 2017-03-28:
Hi,

I am testing example program (Datalogger.ino). The file is created but the function is not returning the object. Thus, I couldn't write or read to the file. I am using XMC 2Go.

Best regards,

`#include <SPI.h>
#include <SD.h>

const int chipSelect = 3;
SdFile root;

void setup() {
  // Open serial communications and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  Serial.print("Initializing SD card...");

  // see if the card is present and can be initialized:
  if (!SD.begin(chipSelect)) {
    Serial.println("Card failed, or not present");
    // don't do anything more:
    return;
  }
  Serial.println("card initialized.");
}

void loop() {
  // make a string for assembling the data to log:
  String dataString = "";

    int sensor = analogRead(12);
    dataString += String(sensor);
    Serial.println(sensor);

  // open the file. note that only one file can be open at a time,
  // so you have to close this one before opening another.
  File dataFile = SD.open("datalog2.txt", FILE_WRITE);

  Serial.println(dataFile);

  // if the file is available, write to it:
  if (dataFile) {
    dataFile.println(dataString);
    dataFile.close();
    // print to the serial port too:
    Serial.println(dataString);
  }
   if the file isn't open, pop up an error:
  else {
    Serial.println("error opening datalog.txt");
  }
  delay(3000);  

}
`

Infineon, 2017-03-28:
To help you, would you please be so kind and tell us your hardware setup for bug tracing?
This makes it easier to find the bug and give advice or fixes.

Thank you very much for your input.

thihaElec, 2017-04-20:
Hi,

Sorry for late reply. I had connected the SD card module (MISO,MOSI,SCLK,CS) to XMC-2GO (MOSI,MISO,SCK,SS). I powered SD card with external 3.3V supply. The card can be initialized . So "SD.begin(chipSelect)" is cleared. But it seems like there is no return pointer from "File dataFile = SD.open("datalog2.txt", FILE_WRITE);". But I observed datalog2.txt was created. Please let me know if you need more information. This is my last piece of jigsaw in my project. Thanks! 

Best regards,

thihaElec, 2017-05-18:
fixed on new update.

closed 2017-05-18 by thihaElec

mrsoeira, 2018-05-21:
Hi,

I seem to be running into a similar issue. How did you solve it, thihaElec?

Thanks in advance

techpaul, 2018-05-21:
What version of XMC-for-Arduino are you using?

Tools -> Board ... -> Boards Manager

  Enter "xmc" in Fileter box should display version installed

Many issues have been fixed in last major release around Easter to V1.1.0