Infineon/XMC-for-Arduino #7: MODBUS Library Not supported

yuvaraaj123, 2017-04-14:


mhollfelder, 2017-04-20:
Hi,

could you please provide more information?

* Which board did you use?
* Which library for MODBUS?
* A minimum example sketch that we can reproduce the problem

Thanks and best regards,

Manuel

mhollfelder, 2017-07-19:
We have adapted a MODBUS library implementation, but we need to test it and integrate it properly.
As soon as we have done this, we will integrate it with a new release.

Best regards,

Manuel

mhollfelder, 2017-10-18:
We got to fix a problem so that many public libraries should work now with the XMC core.
If you face issues with your implementation, please open a new issue.

Otherwise, I am happy to support you with specific error messages here on your library as well.

Best regards,

Manuel

closed 2017-10-18 by mhollfelder