Infineon/XMC-for-Arduino #8: Poor Linux support for XMC2Go

hajosc, 2017-06-22:
On F25, figured the config doesn't work with the F25 packaged Arduino 1.6.4. Required upgrade to Arduino 1.8 to get it to find the compiler.
Then, JLink rpm installs to /opt which is not in the library path, hence uploading to the board doesn't work. Need to tweak the environment variable.
Finally, JLink couldn't connect to the board. Tried on another PC with Windows -> the board works.

That's where I deinstalled everything and went back to Arduino boards which work out of the box. Too much trouble debugging a tool chain...

mhollfelder, 2017-06-22:
We are aware of these problems, same problems we have for MacOS where the JLink installs at the correct path, but the XMC Flasher cannot find it.
Currently, we're trying to figure out how to solve these issues and release a working version for Linux. The strange thing is that for Linux it was working on our test system, but now it's not working anymore.
We need to fix the XMC Flasher to get rid of these problems...

Sorry for the troubles and inconvenience!

DonPY, 2017-09-29:
Any update?   I have a H-bridge Kit2Go and XMC1100 boards that I can't program in OSX.

mhollfelder, 2017-09-30:
Yes, we got the XMCFlasher recently running on Linux, now we are looking for an OS X Mac to test it there as well.
We will release a new version with a modified XMCFlasher as soon as possible.

Sorry for the delay regarding the fix of this issue.
Would be great if you could test it as well and give feedback here.

Best regards,

Manuel

DonPY, 2017-10-03:
Manuel,

I would be happy to try it out.   Please let me know.

Thanks

mhollfelder, 2017-10-03:
I released a new version V1.0.6 with a new XMC Flasher 1.2.1 which should fix many issues for Mac OS and Linux. 
The main problem was that the flasher could not find the shared library for the SEGGER J-Link.
 
We hope that the issue is resolved, but we could not test it for every OS version on the market.
Could you test it and give feedback for your system? Would be great :)

Thank you very much and best regards,

Manuel

DonPY, 2017-10-04:
Manuel,

Thanks so much for your efforts on this!   I can confirm I can program and operate both LEDs on the H-Bridge Kit 2GO board!

Thanks!

macOS Sierra Version 10.12.6

mhollfelder, 2017-10-18:
Great, thank you very much for the feedback!

I waited to close this issue until I've gotten more feedback for this issue on other platforms as well:
if additional problems arise, please do not hesitate to open a new issue.

Hopefully, many problems have been resolved by the fix.

Best regards,

Manuel

closed 2017-10-18 by mhollfelder

PabloPL, 2017-12-06:
It's not working for me on Linux (opened new issue https://github.com/Infineon/XMC-for-Arduino/issues/21).