Infineon/XMC-for-Arduino #9: XMC1100 Serial class problem

KarlArtman, 2017-07-04:
I stumbled across a subtle problem in the Serial class whereby a receive character can get caught in the USIC buffer. Since we only get an interrupt on change of state and currently the Serial class only reads one character from the USIC, one character can remain queued. This tends to show itself when debugging and you hit a breakpoint, then restart the code with serial data coming in. My solution was the following code in the Serial.cpp isr:
        XMC_UART_CH_ClearStatusFlag(_XMC_UART_config->channel, (XMC_UART_CH_STATUS_FLAG_ALTERNATIVE_RECEIVE_INDICATION |
                                    XMC_UART_CH_STATUS_FLAG_RECEIVE_INDICATION));
        while(_XMC_UART_config->channel->RBUFSR & (USIC_CH_RBUFSR_RDV0_Msk | USIC_CH_RBUFSR_RDV1_Msk))
          {
            _rx_buffer->store_char(XMC_UART_CH_GetReceivedData(_XMC_UART_config->channel));
          }

mhollfelder, 2017-07-10:
Thanks a lot!
We are currently preparing a new release and we will integrate the fix of this bugs, too.

Best regards,

Manuel

mhollfelder, 2017-07-19:
This issue should now be resolved with the new [release](https://github.com/Infineon/XMC-for-Arduino/releases/tag/V1.0.3) for 1.0.3.

Many thanks from our side!

Best regards,

Manuel

closed 2017-07-19 by mhollfelder