Infineon/XMC-for-Arduino #10: XMC1300 Bootkit in commits

KristoffRey, 2017-07-12:
Hi, I saw that there is a variant for the XMC1300 Boot Kit but it does't appear in the latest release as the README indicates.

Is there a way for me to include the board into Arduino manually? Thank you!

mhollfelder, 2017-07-12:
Hi,
 
you can activate it just by uncommenting (remove #) the board in the [boards.txt](https://github.com/Infineon/XMC-for-Arduino/blob/master/arm/boards.txt) file:

####################################################
#XMC1300_Boot_Kit.name=XMC1300 Boot Kit
#XMC1300_Boot_Kit.upload.tool=xmcprog
#XMC1300_Boot_Kit.upload.speed=115200
#XMC1300_Boot_Kit.upload.resetmethod=ck
#XMC1300_Boot_Kit.upload.maximum_size=65536
#XMC1300_Boot_Kit.upload.wait_for_upload_port=true

#XMC1300_Boot_Kit.communication=usb
#XMC1300_Boot_Kit.protocol=dragon_isp
#XMC1300_Boot_Kit.program.protocol=dragon_isp
#XMC1300_Boot_Kit.program.tool=xmcprog
#XMC1300_Boot_Kit.program.extra_params=-Pusb

#XMC1300_Boot_Kit.serial.disableDTR=true
#XMC1300_Boot_Kit.serial.disableRTS=true

#XMC1300_Boot_Kit.build.mcu=cortex-m0
#XMC1300_Boot_Kit.build.f_cpu=32000000L
#XMC1300_Boot_Kit.build.board=ARM_XMC
#XMC1300_Boot_Kit.build.board.version=1302
#XMC1300_Boot_Kit.build.board.type=T038x0200
#XMC1300_Boot_Kit.build.board.v=0200
#XMC1300_Boot_Kit.build.core=./
#XMC1300_Boot_Kit.build.variant=XMC1300
#XMC1300_Boot_Kit.build.board_variant=XMC1300_Boot_Kit
#XMC1300_Boot_Kit.build.flash_size=64K
#XMC1300_Boot_Kit.build.flash_ld=linker_script.ld
####################################################

Afterwards, it should work.
Please let me know if you face issues.

Best regards,

Manuel

KristoffRey, 2017-07-13:
Thank you for the fast reply, do I simply clone/fork the repository, make the changes and then make a new release in order for it to be searchable in Arduino's board manager?

mhollfelder, 2017-07-19:
Yes, you could do it like this, but then you also need to add a new .json pointing to that release.

Easier would be to go to your `\Arduino15\packages\Infineon\hardware\arm\1.0.2` folder of the boards manager and remove the `#` from the `boards.txt` file of the current release. 
Depending on your OS and Arduino installation, this location may be different. For me, it is located under `C:\Users\<Username>\AppData\Local\Arduino15\packages\Infineon\hardware\arm\1.0.2`.

Hopefully, this helps you :)

Best regards,

Manuel

mhollfelder, 2017-07-19:
I added a new release which enables the use of the XMC1300 Boot Kit, please update from Arduino IDE to 1.0.3 and it should work :)

Best regards,

Manuel

closed 2017-07-19 by mhollfelder

KristoffRey, 2017-07-20:
Alright, thank you!