Infineon/XMC-for-Arduino #11: Issue in the TLE94112 Arduino library maybe related to the XMC110 library

alicemirror, 2017-07-19:
I have found an issue and added a patch described in the Issue \#1 of the TLE94112LE library
**TRUE and FALSE undefined with Arduino boards, IDE version 1.8.3+ (with workaround)** 

Maybe the worth to take a look if this library issue depends on something wrong in these sources (I suppose)

Enrico

mhollfelder, 2017-07-19:
The problem seems to be that FALSE/TRUE is not correctly defined in the lib when used with other architectures, i.e. not the XMC boards. 
We added a fix with conditional preprocessor directives and I will release a new version in some minutes.
Please test and check whether this will solve the problems for you as well.

Sorry for the troubles from our side.

Best regards,

Manuel

alicemirror, 2017-07-19:
Hello Manuel,

thank you. I will note this in the Road Test on Element14 I am working on and as you release the new version I will test immediately with the modified library. I expect that nothing change in no one boards (XMC110 and Arduino) if I define or undefine the preprocessor directive I have put in the workaround. I will notice the result here.

Enrico

mhollfelder, 2017-07-19:
Released it right now, please test it.
I will also comment in the Road test on Element14, sorry for the delayed reaction and response, but somehow the release has been postponed so far due to the other changes.

Best regards,

Manuel

alicemirror, 2017-07-19:
No problem Manuel, Hardware is great :)

alicemirror, 2017-07-19:
Manuel, I updated the XMC1100 library to 1.0.3 you just released
I tested the modified library with #define and #undef the preprocessor directive and it was ok. Then I installed the origianl TLE94112 library and it worked fine too.

Please close also the issue I opened initially on the TLE94112 library repository. Thanks.

Thank you for your effort. Have a nice day, here in Spain it's time to sleep :)

Enrico

mhollfelder, 2017-07-19:
Perfect! I closed the other issues as well.

Here, it's also time to sleep ;)

Best regards,

Manuel

closed 2017-07-19 by mhollfelder