Infineon/XMC-for-Arduino #14: I2C Wire.begin() non-blocking issue - XMC1100 + TLE94112LE

alicemirror, 2017-09-01:
After the last **1.0.4 Arduino IDE library update** by  @mhollfelder  everything is working fine with the I2C bus. Tested with a Raspberry Pi Compute Module 3 on a multiple I2C peripheral bus it works like a charm.

Problems arises when the **XMC1100 Boot Kit is integrated with the TLE941120 DC motor controller**. In the **Setup()** function there is different behaviour depending on where the **Wire.begin()** call is placed. In detail I have setup a use-case easy to reproduce:

To check the presence of the XMC1100 board on the I2C bus from the Linux terminal I used the command

`i2cdetect -y 1`

**Wrong case**
```
#include <Wire.h>
#include <Streaming.h>
#include <TLE94112.h>
[...]

//! I2C Slave address. 
#define SLAVE_ADDRESS 0x04
//! Motor control class instance
MotorControl motor;
[...]

// ==============================================
// Initialisation
// ==============================================
void setup() {
  // Serial is initialised at high speed.
  Serial.begin(38400);

  // Initialize i2c as slave
  Wire.begin(SLAVE_ADDRESS);

  // define callbacks for i2c communication
  Wire.onReceive(i2cReceiveData);
  Wire.onRequest(i2cSendData);

  // >>>>>>>>>>>> TLE LIBRARY INITIALISED AFTER THE I2C BUS <<<<<<<<<<<<<<
  // initialize the motor class
  motor.begin();  
}
```

**Working case**
```
#include <Wire.h>
#include <Streaming.h>
#include <TLE94112.h>
[...]

//! I2C Slave address. 
#define SLAVE_ADDRESS 0x04
//! Motor control class instance
MotorControl motor;
[...]

// ==============================================
// Initialisation
// ==============================================
void setup() {
  // Serial is initialised at high speed.
  Serial.begin(38400);

  // >>>>>>>>>>>> TLE LIBRARY INITIALISED BEFORE THE I2C BUS <<<<<<<<<<<<<<
  // initialize the motor class
  motor.begin();  

  // Initialize i2c as slave
  Wire.begin(SLAVE_ADDRESS);

  // define callbacks for i2c communication
  Wire.onReceive(i2cReceiveData);
  Wire.onRequest(i2cSendData);
}
```

mhollfelder, 2017-09-02:
Thank you very much for this report.

For the working example, does the motor controller board as well as the I²C still work simultaneously?
Or does then only the I²C work and the motor board is dead?

I have an assumption but this input would help me to clarify the situation.

Best regards,

Manuel

alicemirror, 2017-09-04:
Hi Manuel! Good question. Frankly I have not tested as I was setting up a series of devices in a I2C bus connected to a Raspberry PI Compute Module 3 master and when I saw finally the I2C XMC board in the I2C devices list it was too late to continue testing. In this project development phase I have not yet reached the testing of the motors themselves but - maybe erroneously - I have assumed that what was working before was working after too. Now I saw your issue #15 so I suppose that something was again issuing the system working. I can test if today later.

Anyway the compiling and testing comparison was between the XMC1100 Boot kit and Arduino UNO Rev 3, both with the TLE94112LE installed and using a previous software (the test platform) working fine with motors. You can find the full original source here [TLE94112LE](https://github.com/alicemirror/TLE94112LE) while the related documentation article is here: [TLE94112 Interactive Testing Platform](https://www.element14.com/community/community/project14/diytestequipment/blog/2017/07/26/tle94112-interactive-testing-platform)

Enrico

mhoechner, 2017-09-12:
Hey Enrico,
even after the changes from #15 it might not work to use the XMC1100 as an I2C slave and a SPI master at the same time. 
The Problem is, that for both interfaces the same Hardware peripheral (USIC) is used, as the XMC1100 only has two USIC channels (one of them is used as UART and the other one is used for SPI and I2C). 

With the current changes in the Wire and SPI Lib we tried to fix this by changing the functionality of the USIC during runtime. But we have to do some further testing on this.
Just let us know, once you face any new issue. 

Best Regards,
Moritz

mhollfelder, 2017-09-12:
Hi Enrico, 

we were discussing this internally and I assume the order of executing these functions relates also to initializing the communication.
As Moritz explained, we have the issue that XMC1100 has limited capacity of peripherals dedicated to communication. The XMC4700 board would not face these issues, but the XMC1100 is limited here. We will see what we could do :)

Thank you very much for your feedback and best regards,

Manuel

mhollfelder, 2017-12-12:
Hi Enrico,

just wanted to follow up here and ask whether you found a workaround for the problem in your project?
Does this issues needs to be solved or did it work? If not I will close it, but feel free to reopen it anytime.

Best regards,

Manuel

closed 2018-02-15 by mhollfelder