Infineon/XMC-for-Arduino #16: Function micros() is not working properly

apfeifer, 2017-09-13:
Hi, i have some issues by running some code on the XMC 2Go.
One of the issus is that i recognied that the function micros() is not working properly.
It returns only milliseconds as smallest value.
The microseconds are allways zero.

Here small code example:
`
unsigned long s1 = 0;
unsigned long s2 = 0;

void setup() {
  Serial.begin(115200);
}

void loop() {
  s2 = s1;
  s1 = micros();
  Serial.print(s1);
  Serial.print('\t');
  Serial.println(s1 - s2);
  delayMicroseconds(1);
}
`

> 9945000	2000
9946000	1000
9947000	1000
9948000	1000
9949000	1000
9951000	2000
9952000	1000
9953000	1000
9954000	1000
9955000	1000
9957000	2000
9958000	1000
9959000	1000
9960000	1000
9962000	2000
9963000	1000
9964000	1000
9965000	1000
9966000	1000
9968000	2000
9969000	1000

mhollfelder, 2017-09-13:
Thank you very much for reporting this problem/missing feature.

If I remember correctly, this should be related to the sysTick and there was a reason for this behavior.
I am not sure whether it really is a bug or a missing feature, but we will clarify this and come back with an answer.

Best regards,

Manuel

mhoechner, 2017-09-14:
You are right. With the current implementation the "micros()" function has a resolution of one millisecond.
The reason for this is, that the system tick counter is only updated every millisecond.
You can easly change this in "XMC-for-Arduino / arm / cores / wiring_time.c " (line. 24) by updating the values of "SYSTIMER_TICK_PERIOD" and "SYSTIMER_TICK_PERIOD_US" to:
```c
//****************************************************************************
// @Macros
//****************************************************************************
/**< SysTick interval in seconds */
#define SYSTIMER_TICK_PERIOD  (0.000001F)

/**< SysTick interval in microseconds */
#define SYSTIMER_TICK_PERIOD_US  (1U)
```

The actual reason why we did not implement it like this from the beginning was, that we did not want to have a routine interrupting the user code every microsecond.
We will see how/if we can change this.

Best regards
Moritz

apfeifer, 2017-09-14:
Hi guys,

thanks for the fast reply.
Unfortunatly i tried this already, and it does not work...
Or, nothing works after this change...

Hopefully its my fault and i made a mistake.
I changed the in the file
"C:\Users\{username}\AppData\Local\Arduino15\packages\Infineon\hardware\arm\1.0.4\cores\wiring_time.c"

//****************************************************************************
// @Macros
//****************************************************************************
/**< SysTick interval in seconds */
//#define SYSTIMER_TICK_PERIOD  (0.001F)
#define SYSTIMER_TICK_PERIOD  (0.000001F)

/**< SysTick interval in microseconds */
//#define SYSTIMER_TICK_PERIOD_US  (1000U)
#define SYSTIMER_TICK_PERIOD_US  (1U)

After compiling and uploading the hex-file to the dev-board no sketch works.
Not even a simple led-blinking...

Best regards,
Anton

mhoechner, 2017-09-15:
No, you did not make a mistake. 
I only tested it for the xmc4700 and as this controller has a faster CPU, no problem occurred.
The xmc1100 seems to be too slow to generate the Interrupt every microsecond.

So to make the "micros()" function properly working, we have to change the implementation.

Thanks again for your report. I hope we can soon provide a solution.

Best regards
Moritz

apfeifer, 2017-09-16:
Ah ok, yeah that would be awesome! Thanks in advance!

mhollfelder, 2017-10-18:
Sorry for the delayed response, the issue will be fixed in our major release 1.1.0 according to the change.
We are waiting for other open developments to provide this major step.

If the request is urgent, please let me know here to help you immediatly

Best regards,

Manuel

closed 2017-10-18 by mhollfelder