Infineon/XMC-for-Arduino #19: Documentation Required (Wiki) XMC 1300 Pin Out

Similian206, 2017-11-06:
Hi,

this project is very promising as it makes transitioning a lot easier. 
I just upgraded to xmc 1300, since I have seen it in the Arduino IDE as supported Board.

Now I was wondering that the PinOut in terms of Arduino pins is. 
I did not find any docs in this regard - can we have a picture like we have one for XMC1300. 
Or am I missing something and there was a way to map this from the XMC1100 board? 

Also if xmc1300 really works this should really be documented here since that is amazing.

Thanks a lot !!!!

mhoechner, 2017-11-08:
Hi,

the xmc1300 implementation is still in a working state. We have to do some further tests and write the documentation.
I hope we can finish this soon. But anyhow it should work properly.

For now you can have a look at the file:

> XMC-for-Arduino / arm / variants / XMC1300 / config / XMC1300_Boot_Kit / pins_arduino.h

Here the used Pins are described inside the "mapping_port_pin" variable.

Best Regards

Similian206, 2017-11-16:
Hi, 

thank you for the directions. I have tested some examples and already ported some of my projects.
It seems to work very well for now. 

Thanks for the directions.

techpaul, 2018-03-30:
For those using XMC1300 Boot Kit see my Wiki for a table of pin out which may be helpful as in pin header layout

https://github.com/techpaul/XMC-for-Arduino/wiki/Hardware-Pins---Pinout-XMC1300-Boot-Kit

As from pins_arduino.h

sherylll, 2018-09-13:
Hi,

I just saw it and it looks good. As nothing new has come up for this board for some time, I simply redirected the main wiki to @techpaul 's.

closed 2018-09-13 by sherylll