Infineon/XMC-for-Arduino #20: Setting up HardwareSerial and or implementing SoftwareSerial

Similian206, 2017-11-16:
Hi there,

I have been testing some of my arduino projects with an XMC 1300. 
The project port very well, so thanks a lot for setting up this project.

I found that some arduino libraries use a SoftwareSerial as a parameter.
Would we have to implement Software Serial to use such libraries like bportaluri/WiFiEsp?

WiFi.init(&MyCustomSoftwareSerial); // the init takes a &Stream as parameter

If HardwareSerial is a Stream I guess I could pass that, yet how to set it up?
Or SoftwareSerial adruino lib only seems to be missing interrupt.h how could I set one of those up?
or would the Wire be a legitimate third option?

Regards
Sim

mhoechner, 2017-11-22:
Hi,

yeah... a SoftwareSerial Library is not yet implemented for the xmc.
Existing SoftwareSerial Libraries will propably not work, as they usually have a controller-specific implementation.

As the HardwareSerial also is a Stream, I guess it could work.

BR
Kalle

mhollfelder, 2017-12-12:
Hi Sim,

did you further test this or are you facing issues in what you are trying to do?
If so, feel free to ask here and I will try to help you/give support.

As pointed out, the SoftwareLibrary needs to be implemented/ported to XMC and it has not been done so far.

Sorry for the delayed reply from my side.

Best regards,

Manuel

closed 2018-02-15 by mhollfelder

Nyceane, 2018-07-08:
I need help on this too...