Infineon/XMC-for-Arduino #21: Unable to flash any sketch to XMC2GO on Linux

PabloPL, 2017-11-26:
Board: XMC2GO
OS: Ubuntu 17.04
Arduino IDE: 1.8.5 (fresh download)
Arduino IDE XMC addon: 1.0.6

Tested using arduino blink sketch, and i'm not able to flash sketch using Arduino IDE (i'm able to do this using ozone from segger).

From IDE log i only see
```java -jar /home/pablo/.arduino15/packages/Infineon/tools/XMCFlasher/1.2.1/XMCFlasher.jar -p /tmp/arduino_build_73071/sketch_nov26a.ino.hex -device XMC1100-0064 
Picked up JAVA_TOOL_OPTIONS: 
--------------------------
Infineon XMC Flasher Lite
Copyright Infineon Technologies 2017
--------------------------
Linux
Operating System: Linux

Initialisation
--------------------------
Linux
```

Quick workaround was to
```export IFX_JLINK_PATH=/opt/SEGGER/JLink/```
So it looks like it's searching in wrong directory for segger libraries.

```ls /opt/SEGGER/``` is returning
```
lrwxrwxrwx 1 root root   23 paź  6 17:07 JLink -> /opt/SEGGER/JLink_V620e
drwxr-xr-x 8 root root 4096 paź 13 18:07 JLink_V620e
drwxr-xr-x 3 root root 4096 lis 10 14:25 ozone
```
After this i was able to flash board
```
Picked up JAVA_TOOL_OPTIONS: 
Loading JLink Library... 
--------------------------
Infineon XMC Flasher Lite
Copyright Infineon Technologies 2017
--------------------------
Linux
Operating System: Linux

Initialisation
--------------------------
Linux
JLink Library Version: V6.20e

Configuring Device
Device Name : XMC1100-0064
Interface   : SWD
Speed       : 4000
File Upload
--------------------------
Uploading /tmp/arduino_build_370759/sketch_nov26a.ino.hex
... Done
--------------------------
... Finished Succesfully
```

PabloPL, 2017-11-26:
After doing
```
ln -s /opt/SEGGER/JLink/libjlinkarm.so /usr/lib/libjlinkarm.so
```
I was able to flash sketch to board.
```
Picked up JAVA_TOOL_OPTIONS: 
--------------------------
Infineon XMC Flasher Lite
Copyright Infineon Technologies 2017
--------------------------
Linux
Operating System: Linux

Initialisation
--------------------------
Linux
Loading JLink Library... 
JLink Library Version: V6.20e

Configuring Device
Device Name : XMC1100-0064
Interface   : SWD
Speed       : 4000
File Upload
--------------------------
Uploading /tmp/arduino_build_283508/sketch_nov26a.ino.hex
... Done
--------------------------
... Finished Succesfully
```

mhollfelder, 2017-12-12:
Sorry for the delayed reply to your issue and thank you very much for reporting this problem.

The mentioned problem is strange as we changed the search paths for the flasher some time ago and we thought the problem has been fixed.
We will test it again and try to find out where the problem comes from.

Best regards,

Manuel

bluetiger9, 2018-05-26:
Creating the symbolic link from @PabloPL's comment fixes the problem.

Seems that without having symlink, the uploader tries to load the 32-bit library:
```
$ ./arduino
...
Exception in thread "main" java.lang.UnsatisfiedLinkError: jlinkarm.JLinkARMLibrary.JLINKARM_GetDLLVersion()I
	at jlinkarm.JLinkARMLibrary.JLINKARM_GetDLLVersion(Native Method)
	at Infineon.JLinkDebugger.GetDLLVersion(Unknown Source)
	at Infineon.XMCDevice.Program(Unknown Source)
	at Infineon.upload.main(Unknown Source)
# BridJ: dlopen error when loading /opt/SEGGER/JLink_V512g/libjlinkarm_x86.so.5.12.7 : /opt/SEGGER/JLink_V512g/libjlinkarm_x86.so.5.12.7: wrong ELF class: ELFCLASS32
```

Maybe this helps to fix the issue.

Cheers,
Attila

techpaul, 2018-06-15:
Is this resolved with release V1.1.1 using newer gcc version?

If not should this be added to Wiki for installation or older version installation?

StephiB, 2018-06-15:
HiUnfortunately no change. Same compiler error as in 1.1.0. Still no input files found.Best regards StephanVon meinem Huawei-Mobiltelefon gesendet-------- Originalnachricht --------Betreff: Re: [Infineon/XMC-for-Arduino] Unable to flash sketch to XMC2GO on Linux (#21)Von: Paul Carpenter An: Infineon/XMC-for-Arduino Cc: Subscribed Is this resolved with release V1.1.1 using newer gcc version?
If not should this be added to Wiki for installation or older version installation?

{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [Infineon/XMC-for-Arduino] Unable to flash sketch to XMC2GO on Linux (#21)",
"sections": [
{
"text": "",
"activityTitle": "**Paul Carpenter**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@techpaul",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"Infineon/XMC-for-Arduino\",\n\"issueId\": 21,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"Infineon/XMC-for-Arduino\",\n\"issueId\": 21\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/Infineon/XMC-for-Arduino/issues/21#issuecomment-397578628"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 279643027\n}"
}
],
"themeColor": "26292E"
}

techpaul, 2018-06-15:
Well perhaps we need to add to Wiki the fix suggested by PabloPL above of

"After doing

ln -s /opt/SEGGER/JLink/libjlinkarm.so /usr/lib/libjlinkarm.so

I was able to flash sketch to board.
"

PabloPL, 2018-06-15:
Soon we should be able to use PlatformIO  - https://github.com/Infineon/platformio-infineonxmc

PabloPL, 2018-06-15:
@techpaul 
> Is this resolved with release V1.1.1 using newer gcc version?
I think problem is with XMCFlasher, not with gcc version. For some reason, it's not able to correctly load/use JLink software to flash board.

techpaul, 2018-06-15:
Well I have added a note about this on my Wiki for my fork of the package at

https://github.com/techpaul/XMC-for-Arduino/wiki/Linux-64-bit-Flasher-to-board

Until  the Segger installation gets resolved

PabloPL, 2018-06-20:
@techpaul Is there repo for XMCFlasher so we could for example create PR with fix ? This looks like something very easy to do.

techpaul, 2018-06-20:
Also @mhollfelder 

There is no repo for this to my knowledge.

As this is an Infineon utility based or wholly part of the Segger JLink family, this has to be passed up the support chain for the Linux 64 bit installer to be fixed. Or part of the package beyond what I see being fixed

sherylll, 2018-09-13:
Since it seems to be a segger- or XMCFlasher-related problem, I'll close this issue.

Since PlatformIO does not depend on XMCFlasher, if this does not happen with PIO, it should de a problem of XMCFlasher (which to my knowledge is not open-sourced).

closed 2018-09-13 by sherylll