Infineon/XMC-for-Arduino #22: Error downloading Infineon .json url in Arduino IDE

webstone13, 2018-01-10:
Has anyone programmed XMC microcontroller by Infineon Technologies using Arduino IDE?

I tried as per the steps given in the GitHub repo of Infineon. Below is the link of the same:

[https://github.com/Infineon/XMC-for-Arduino/wiki/Implementation-in-Arduino-IDE](url)

After I copied the .json url in the preferences, I am getting following error:

http://downloads.arduino.cc/packages/package_index.json file signature verification failed. File ignored.

Index error: could not find referenced tool name=arm-none-eabi-gcc version=4.8.3-2014q1 packager=arduino
Index error: could not find referenced tool name=arm-none-eabi-gcc version=4.8.3-2014q1 packager=arduino
Index error: could not find referenced tool name=arm-none-eabi-gcc version=4.8.3-2014q1 packager=arduino
Index error: could not find referenced tool name=arm-none-eabi-gcc version=4.8.3-2014q1 packager=arduino
Index error: could not find referenced tool name=arm-none-eabi-gcc version=4.8.3-2014q1 packager=arduino
Index error: could not find referenced tool name=arm-none-eabi-gcc version=4.8.3-2014q1 packager=arduino
Index error: could not find referenced tool name=arm-none-eabi-gcc version=4.8.3-2014q1 packager=arduino

What can be a possible solution for it? I am running Arduino IDE 1.8.5 on Windows 10. Is this has anything to do with it?

mhollfelder, 2018-01-10:
Hi,

the problem is that there seems to be an issue with file verification of the Arduino package index file (when it is downloaded):

`http://downloads.arduino.cc/packages/package_index.json file signature verification failed. File ignored.`

Therefore, this file is ignored by your Arduino IDE instance.
When you are trying to download the XMC package, the Arduino IDE tries to find the ARM compiler tool which is referenced in the `package_index.json` file. As this one is ignored, the installation cannot find it.

I tested it here with Arduino 1.8.5 and it worked properly.
Have you tried using the [Arduino 1.8.5](https://www.arduino.cc/download_handler.php?f=/arduino-1.8.5-windows.zip) standalone and creating a portable folder inside to see whether it is working as expected?

Best regards,

Manuel

webstone13, 2018-01-11:
Issue resolved. I did a re-installation of XMC board under Tools->Boards->Board Manager.
It is working perfectly fine now.

Thanks @mhollfelder for your kind help.

closed 2018-02-15 by mhollfelder