Infineon/XMC-for-Arduino #23: Tx and Rx pin for UART in XMC 2GO

webstone13, 2018-01-11:
As shown in the image at the link below:

[https://github.com/Infineon/Assets/blob/master/Pictures/XMC%202Go_PO.jpg](url)

 the UART Rx and Tx pin mentioned are P2.6 and P2.0 respectively. However, in the user manual released by Infineon, the UART Rx and Tx pins mentioned are P2.1 and P2.2. Please find the user manual here (on page 11 of 15):

[https://www.infineon.com/dgdl/Board_Users_Manual_XMC_2Go_Kit_with_XMC1100_R1.0.pdf?fileId=db3a3043444ee5dc014453d6c75078c6](url)

Which one is correct? I think the one in the user manual is the right one.

mhoechner, 2018-01-11:
Hi,

actually both are correct. 
For the XMC it is possible to assign the UART Rx and Tx to different Pins.

In case of this implementation, by default Serial (UART) uses pin P2.1 and P2.2, which enables the connection to the host PC. (Serial Monitor)
If you want to use the on-board Rx and Tx Pins (P2.6 and P2.0), you have to change the "pins_arduino.h" file under:
> XMC-for-Arduino / arm / variants / XMC1100 / config / XMC1100_XMC2GO / 
In this file you have to uncomment line 186 and comment line 185, which will change the configuration of the UART Pins.

Be Aware, that you can only use either the on-board Pins **or** the Pins connected to the Host PC.

BR

mhollfelder, 2018-01-12:
Actually, that's something we need to document properly and I will follow up on this immediately.

I know, this is confusing at first glance, but due to the structure of the peripherals of the XMC µC series it gets clearer when having a look in the Datasheet/Reference Manual, e.g. [Datasheet XMC1100](https://www.infineon.com/dgdl/Infineon-xmc1100_AB-DS-v01_06-EN.pdf?fileId=5546d4624a0bf290014a4bdaff9325bd) and [Reference Manual XMC1100](https://www.infineon.com/dgdl/Infineon-xmc1100-AA_rm-UM-v01_01-EN.pdf?fileId=5546d46255dd933d0155e31753b077af). 

But even still, there are lots of pages to read for the full picture and it's clearly out of scope of an Arduino-like approach. So, please don't hesitate to approach us for further questions regarding this.

Best regards,

Manuel

webstone13, 2018-01-13:
Thanks @mhollfelder and @mhoechner clearing the doubt.
I have started exploring XMC microcontrollers for one of my projects and I hope I will keep coming back.

HMenger, 2018-01-19:
I asked myself if it was possible to have  2  HarwareSerial , Serial on USB and Serial1 on Pins, at the same time, using USIC0 CH1 , but loosing SPI and I²C of course....

After doing  some research in the Datasheet and Refence Manual (and some trial and error) , i can say
 YES WE CAN  ;-)

With some modifikations in 
- pins_arduino.h
- HardwareSerial.h
- HardwareSerial.cpp

- TX1-pin is connected to P2.11 
- RX1-Pin to P2.10 

please have a look at attached files
this should work on other xmc1xxx boards too ( maybe the pins must be changed )

best regards
Hubert

[image]

[HardwareSerial1 for XMC2GO.zip](https://github.com/Infineon/XMC-for-Arduino/files/1647931/HardwareSerial1.for.XMC2GO.zip)

mhollfelder, 2018-02-15:
@HMenger: Thank you very much for your input and help here :)

If someone needs help or more input, please let me know and reopen this issue.
We are currently preparing a new release and I will add this information in a wiki page once we are finished.

Best regards,

Manuel

closed 2018-02-15 by mhollfelder

ironstrike, 2019-05-31:
> I asked myself if it was possible to have 2 HarwareSerial , Serial on USB and Serial1 on Pins, at the same time, using USIC0 CH1 , but loosing SPI and I²C of course....
>
> After doing some research in [...]
Okay, Can you say what kind of modification is to be done on these files
* pins_arduino.h
* HardwareSerial.h
* HardwareSerial.cpp
Because currently I'm doing a project on XMC1100 2 Go , Where I'll use both the UART channels one for Serial monitor another for sending sensor data to ESP8266 module. For using second UART channel how am I suppose define it, either by HardwareSerial or by SoftwareSerial.

techpaul, 2019-05-31:
Read the source files in the zip

techpaul, 2019-05-31:
A hint about which type of serial is in the filenames of

pins_arduino.h
HardwareSerial.h
HardwareSerial.cpp

xzsfd2457, 2019-05-31:
[image]

ArkajyotiChatterjee, 2019-05-31:
There is nothing called Serial1 defined in pins_arduino.h for XMC1100 Bootkit. Please ensure the setting that I mentioned in issue 96(I attached a screenshot).
Also, stop asking the same issue repeatedly. @techpaul  already clarified the whole procedure before. Try to follow the steps outlined.

techpaul, 2019-05-31:
Ahh students suddenly realising they have to actually do their projects at last minute