Infineon/XMC-for-Arduino #24: PHPoC WiFi Shield connection with XMC 1100 bootkit

webstone13, 2018-01-19:
Hi there,

PHPoC WiFi Shield for Arduino connects it to WiFi networks. Since, XMC 1100 bootkit has same from factor as of Arduino and almost similar pin out diagram, can PHPoC WiFi Shield be used with XMC 1100 bootkit?

Or for that matter, can Arduino Uno shield be used with XMC 1100 bootkit?

mhollfelder, 2018-02-15:
Hi,

As the PHPoC WiFi shield uses SPI, it should be compatible with the XMC1100 Boot Kit out of the box :)

With some limitations, shields for Arduino can be used with the XMC1100 Boot Kit in general.
The board itself is 5 V compatible, similar as the Arduino Uno. However, there are some limitations in terms of:

- Only two communication channels can be used in parallel, e.g. UART and I²C or I²C and SPI, etc.
- The on board UART (TX/RX) and the UART to the PC (VCOM) are swapped, i.e. when you want to use the on-board UART you are not able to send to the PC and vice versa (can be overcome by swapping the pins on the board with some cables)
- Number of PWM signals, i.e. the XMC1100 has less independent PWM signals as the Arduino Uno

The XMC4700 Relax Kit does not have these kind of problems.
If you face specific issues, just let us know and we can come back to the problem in a more detailed manner.

Best regards,

Manuel

mhollfelder, 2018-03-02:
I will close this issue, if you have additional questions, feel free to open it again.

Best regards,

Manuel

closed 2018-03-02 by mhollfelder