Infineon/XMC-for-Arduino #25: Support for other boards / mcu's

HMenger, 2018-01-19:
Is there a plan to add other boards for example XMC1400  Boot Kit ?

I intend to use XMC1403 or  XMC1404 for a new projekt with CAN/LIN communication and it would be very helpfull for rapid prototyping .

Best regards
Hubert

mhollfelder, 2018-02-05:
Dear Hubert,

Currently, we have no plan to add these boards as well right now.

Would you use another board with CAN/LIN as well, e.g. the XMC4700?
If so, we could evaluate the CAN/LIN for the existing Arduino implementations.

Best regards,

Manuel

HMenger, 2018-02-05:
Dear Manuel,

thanks for your response,
Although the 4700 will be a little overpowered for my projekt i would give it a try ;-)
Which board would you recommend ? Is the XMC4700 Relax Lite Kit OK? , or should it be the 5V version.
I have an XMC4500 Relax Lite Kit, but that is not added to arduino yet , will be sooner or later ?

The idea was to use the XMC1404 as  CAN-LIN Gateway /LIN-Master and  XMC1100/1200/1300 as LIN-Slave(s) 

Best Regards,
Hubert

dmothes, 2018-02-06:
Hey Hubert, 
I'm working in my spare time on xmc 1200 and 1400 and do try to integrate these in my spare time. The question for me is, if I  do a fork of this repo or I could pull my code. But this questions goes to Manuel.

HMenger, 2018-02-06:
Hi dmothes ,that is great news .
I guess it is a lot of work, how do you make it? Did  you take a Variant e.g XMC1300 and modify to fit with XCM1200 Board?

best regards 
Hubert

dmothes, 2018-02-06:
Hi Hubert,

yes that's what i do, a copy of the XMC1300. Normaly this is the easy 
way. The complex thing ist to find some good practical configurations 
for the pinouts, which should be useful. The XMC have a big matrix to 
handle multiple things on there pins.

Regards
Dirk

HMenger, 2018-02-14:
Hi Dirk,
do you have something working already? 
I hope you have a lot of spare time to  work on the boards : -)

regards
Hubert

HMenger, 2018-02-15:
Hi Manuel,
my XMC4700 Relax Kit arrived today , is  there allready a sample  program for CAN avaiable.

regards
Hubert

mhollfelder, 2018-02-15:
Hi Hubert,

I assume you want to have an Arduino example for CAN, right?
If you are looking for Arduino examples, we do not have examples here right now.

However, there are examples for the DAVE toolchain available [here](https://www.infineon.com/cms/en/product/promopages/aim-mc/dave_downloads.html) for the XMC4700.

Best regards,

Manuel

HMenger, 2018-04-18:
Hi Manuel,

i made some modifikation in boards.txt to add XMC1200 and XMC1400 bootkits and added the VARIANT folders for those boards in my setup.
I made Easy Start Projekts with DAVE , copied the Start files from there to the Variant-Folders, and modified the "pinsarduino.h" for the GPIO-pins, LED definition and so on. 

The XMC1200 works so far - GPIO, AnalogReead , Serial .. (didn't test all up to now) but sure needs still some work in Libraries (I2C, SPI,...)

The XMC1400 works also with basic functions Digital I/O,  AnalogrRead,...
Serial dosn't work up to know , i guess due to the different way the interrupthandlers are named and handled., the Startupfiles look quite different from other boards.
Serial works if i use the XMCLib-functions.

BTW.
To have  a more comfortable Programming Enviroment  i am currently testing VSCODE with ARDUINO extension, wich works good so far.

regards
Hubert

techpaul, 2018-04-18:
The startup/pins-arduino./linke/.ld files may look different because some things have been changed in XMC-for-Arduino to make thingshave less unnecessary items and some things simplified or made safer, like stack placement and variables defined standardised across boards.

Those changes may not have been passed to Dave/XMCLib maintainers yet.

Serial is most likely not working due to some pin configuration or interuipt/Event Routing mainly defined in pins_arduino.h and cores/HardwareSerial.cpp

I am nor part of Infineon, but from my profile on here you can contact me if you want someone else to look over your files, as I do not have those boards I can only make judgements based on experience so far and datasheets/user manual descriptions.

HMenger, 2018-04-18:
Hi Paul,
thanks for your response,
Serial is working now, i had to add this in HardwareSerial.cpp:

```
#if XMC_BOARD == XMC1400_Boot_Kit // XMC1400
void IRQ9_Handler( void )                        // IRQ-Handler according XMC1400.h / startup_XMC1400.S
{
Serial.IrqHandler();
}
#else
void USIC0_0_IRQHandler( void )
{
Serial.IrqHandler();
}
  #if (NUM_SERIAL > 1)
  void USIC1_0_IRQHandler( void )
  {
  Serial1.IrqHandler();
  }
  #endif
#endif // XMC1400
```
 i found this : 

> 5.9.3 Interrupt handler definition (overriding default handler)
> C-Start defines the default interrupt handler for all of the CPU exceptions and device interrupts. For the
> interrupt enabled above, user applications may define a final handler to meet their particular needs.
> XMC1100/1200/1300:
> For XMC1100/1200/1300 devices, the interrupt handler name is prefixed by the module name.
> void LEDTS0_0_IRQHandler(void){}/*overrides the default interrupt handler for
> Node-29*/
> XMC1400:
> For the XMC1400 device, the interrupt handler name is generic since the node may be assigned to different
> modules.

in the appending appnote
[Infineon-AP32326-XMC1000-C_Start_and_Device_Initialization-AN-v01_00-EN.pdf](https://github.com/Infineon/XMC-for-Arduino/files/1924902/Infineon-AP32326-XMC1000-C_Start_and_Device_Initialization-AN-v01_00-EN.pdf)

How can i send my files to you?

regards Hubert

techpaul, 2018-04-18:
Glad you got that going..

Been busy today hence later reply. 

My email is on profile, a zip file is easiest, and let me know what area you are conecentrating on or need second set of eyes to look over.

HMenger, 2018-05-21:
Finally i had some time to work on the XMC1400 Boot Kit.
So far Serial,( debug) I2C , SPI is working and i can send/receive CAN-Messages with some code i took and modified from MCAN example.
I made some test with I2C (SH1106) and SPI (SSD1309)  with  128x64 OLED displays using u8g2 library, and found that SPI workes much to slow. ( 20 to 80 times slower than expected )
I will open a seperate issue for this.

best regards 

Hubert

sherylll, 2018-09-13:
Hi please checkout this:
https://github.com/Infineon/XMC-for-Arduino/tree/develop/arm/variants/XMC1400

However it requires further testing, by this time only Serial & LED are guaranteed to work.
If you're still working on XMC1400 you could send a seperate PR.

closed 2018-09-13 by sherylll