Infineon/XMC-for-Arduino #26: Masses of compiler warnings on empty project

techpaul, 2018-02-12:
Using Arduino 1.8.5 and 1.0.1  as latest available from Board Manager

Standard test I do for any Arduino Board Package is

1. Start with standard EMPTY project with empty  setup and loop
2. Set preferences -> Compiler Warnings to ALL
3. Select new board
4. Click verify (compile NO download)
5. See if ANY warnings or errors come up
6. If no errors repeat steps 3 to 5 for any other board in the board package

This checks for any hidden traps and that board package selection has no weird errors as well.

Ran on board XMC1100 and got MASSES of warnings some look like hidden traps in some circumstances, but should not be there. Examples

**A) redefine true**

 ../Application Data\Arduino15\packages\Infineon\hardware\arm\1.0.1\cores/wiring_constants.h:34:0: warning: "true" redefined [enabled by default]

 #define true            1
 ../\Application Data\Arduino15\packages\Infineon\hardware\arm\1.0.1\cores/Arduino.h:31:0,
.../applic~1\arduin~1\packages\arduino\tools\arm-no~2\480e15~1.3-2\lib\gcc\arm-none-eabi\4.8.3\include\stdbool.h:43:0: note: this is the location of the previous definition

 #define true true

**B) redfine false**
In file included from C:\.......\Application Data\Arduino15\packages\Infineon\hardware\arm\1.0.1\cores/Arduino.h:165:0,

                 from C:\.....\Temp\arduino_build_342383\sketch\sketch_feb12a.ino.cpp:1:

C:\......\Application Data\Arduino15\packages\Infineon\hardware\arm\1.0.1\cores/wiring_constants.h:35:0: warning: "false" redefined [enabled by default]

 #define false           0

In file included from C:\Documents and Settings\Paul\Local Settings\Application Data\Arduino15\packages\Infineon\hardware\arm\1.0.1\cores/Arduino.h:31:0,

                 from C:\.....\Temp\arduino_build_342383\sketch\sketch_feb12a.ino.cpp:1:

c:\......\applic~1\arduin~1\packages\arduino\tools\arm-no~2\480e15~1.3-2\lib\gcc\arm-none-eabi\4.8.3\include\stdbool.h:42:0: note: this is the location of the previous definition

 #define false false

**C)  Defined but not declared functions**
C:\.....\Local Settings\Application Data\Arduino15\packages\Infineon\hardware\arm\1.0.1\cores/wiring_digital.h:74:24: warning: inline function 'digitalToggle' declared but never defined [enabled by default]

     extern inline void digitalToggle(uint8_t pin);

C:\.....\Local Settings\Application Data\Arduino15\packages\Infineon\hardware\arm\1.0.1\cores/wiring_digital.h:67:27: warning: inline function 'digitalRead' declared but never defined [enabled by default]

     extern inline uint8_t digitalRead(uint8_t pin);

C:\....\Local Settings\Application Data\Arduino15\packages\Infineon\hardware\arm\1.0.1\cores/wiring_digital.h:58:24: warning: inline function 'digitalWrite' declared but never defined [enabled by default]

     extern inline void digitalWrite(uint8_t pin, uint8_t val);

[Then repeated again]

**D) Missing braces**
C:\.....\Local Settings\Application Data\Arduino15\packages\Infineon\hardware\arm\1.0.1\cores\wiring_analog.c:40:5: warning: missing braces around initializer [-Wmissing-braces]

     XMC_VADC_GLOBAL_CONFIG_t vadc_global_config = {0};

Many other types of error and then repeats at link stage something like 60 warnings/errors listed.

techpaul, 2018-02-12:
Well I have done some work here which I will create a local master and do a Pull request when I get time, most of the problems are incorrect union and/or struct initialisations, but a few howlers.

inline of digitalRead, digitalWrite and digitalToggle is best left to the compiler/linker to optimise, as to be proper inline the functions need to be in the header not the C file. Due to the structure of the headers TREE from Arduino.h down to variant files and the mixed up parts it would require

- Major reordering of Ardino.h

- SPLITTING pins_arduino.h in variants tree being split into pins_arduino.h and serial_arduino.h, due to what is defined in Arduino.h, wiring_digital.h HardwareSerial.h and the inline functions requiring definitions of pin mapping, GND and other attributes, and pins_arduino.h needing HardwareSerial.h defined first.

Other issues involved things like Tone.cpp having multiple functions with local variable _timer being signed and unsigned in different functions and passing the value around so one function being passed in unsigned value (disableTimer) uses unsigned has paramater passed from one function at least that is signed internally, then doing the check on UNsigned value of
   if( _timer >= 0 && ......

So once I do a pull request it still needs Tone.cpp tested by someone (preferably sanity checked as it looks a mess.

Total errors now TWO down from over 60 as follows

........\Arduino15\packages\Infineon\hardware\arm\1.0.1\cores\wiring_time.c:180:22: 
    warning: unused parameter 'Temp' [-Wunused-parameter]

 void delay_cb( void* Temp )
                      ^
.......\Arduino15\packages\Infineon\hardware\arm\1.0.1\cores\xmc_lib\XMCLib\src\xmc_vadc.c:233:71:
    warning: unused parameter 'conv_type' [-Wunused-parameter]

                                           const XMC_VADC_GROUP_CONV_t conv_type, const uint32_t set_num)

Both need more time than I have available now

mhollfelder, 2018-02-15:
Hi techpaul,

Thank you very much for your work and reports here.

Actually, the latest version should be 1.0.6 which should also be available in the boards managers.
However, I tested your findings with the version 1.0.6 and it is valid there as well.

As we are preparing a new release, I will implement your changes in the release to fix the errors there as well.
I think we missed checking all these things when we published the first release and didn't follow up on this later on, sorry for that!

Thank you very much again and best regards,

Manuel

techpaul, 2018-02-15:
I was using package V1.06 and IDE V1.85

Series of staged pull request being lined up mostly affecting one or two files ata time.

See fork in my repository, next 3 committs and a combination tag will be made available in my repositor and as pull request.

Tidied up pins_arduino.h so all boards are styled the same for UART config along with conditional assembly with a define at TOP of file for choosing Serial on board or Serial debug instead of 6 lines being commented or uncommented. So for XMC100 and XMC1300 same method.

Current stages awaiting push from here gives reduction in warnings to 8 (few more stages to do) even so 75 lines of errors instead of 585 is a vast improvement. Actually the fixes actually reduce the code size slightly as well, on XMC1100 boot about 20 bytes.

techpaul, 2018-02-16:
There is now a pull request for 7 stage, commit that clears 95% of the warnings the remnants needs someone who has time to sort the USB and wiring_time.

This involves more time to track back through the Event unit works with library calls when really you either need to pass a parameter or know how not to pass one from event unit. The USB section needs someone who has time to follow through the CMSIS system and USB jump table in xmc_usbh.c.

The remaining I do not have time at momemnt for but at least  the warnings are now only

XMC1100  2 errors
XMC1300  1 error
XMC4700 1 error as per XMC1100 and 12 unused parameters in USB

Back to completing project and maybe another improvement to follow.

techpaul, 2018-02-19:
As the pull request has been incorporated into development then I am closing this issue for now.

Hope to be seen in next release

closed 2018-02-19 by techpaul