Infineon/XMC-for-Arduino #37: wiring_time module excessive functions and delaymicroseconds

techpaul, 2018-03-03:
Wiring_time includes all sorts of event timerlist as a linked list with all the insert/delete and other management. However the only use of it appears to be for delay function for millisecond delay.

With lots of handler linked list timer objects being accessed as functions and from handlers suggesting more objects should be actually volatiles.

Personally all the code for linked list management, could be better replaced by a fixed array with known offsets set at compile time for all possible systick  timer events, and probably find there is only ONE.

Delaymicroseconds while inlined with NOP loops is only useful for at most a 10 microsecond delay beyond that it disables interrupts that can really mess things up. I have times where 100 to 200 microsecond delays are needed.

A 10 microsecond delay means for XMC1xxxx at 32 MHz we use 320 clock cycles or 160 NOPs .
Whilst on XMC4xxx this is 1440 clock cycles and 1470 NOPs

Realistically if there is a timer/events map for what XMC-for-Arduino uses it could do with a hardware timer delay and interrupt for > 10 microsecond delay on XMC1xxx, and use this method only for XMC4xxx

techpaul, 2018-03-03:
Is there any form of documentation for at least XMC100 Boot KIt as to resources used and available for interrupts, event unit, timers (CCU), so at time I can look at how to implement a hardware delaymicroseconds ????

mhoechner, 2018-03-07:
hi techpaul,

First of all, I fully agree ... there is definitely room for improvement in the wiring_time.
Really appreciating your work on this.

Regarding your question here:
Until now the code always was the golden reference for "documentation".
In the future we hopefully will find time to add more and more to our wiki.

The current implementation uses all CCU channels of XMC1100 Boot kit for PWM/analogWrite.
For your purpose you could use channel 1 of CCU0 (CCU40_CC41), which is currently used for PWM1 at Arduino Pin 4 
-> As for "standard" Arduino Boards there is no PWM on this Pin, it wont have any impact on potentially added external hardware.

Best Regards
Moritz

techpaul, 2018-03-07:
See Issue #38 for comments about improved systick handling for wiring_time improvements, the scheduling is an adaptation of code I first wrote nearly 30 years ago and has appeared in many guises.

At least with CCU40 channel 1 I stand a  chance of probably on all boards removing from PWM list and making pin GPIO only. Making it common to all platforms. Finding what resources are available interupts, events, timers etc is not easy with all the code files to look over.

delayMicroseconds is a core function required in many places for things like Liquid Crystal library, other standard lkibraries and my own PS2 keyboard (and potentially PS2 mouse) libraries.

On sub 20MHz processors a NOP delay loop is far easier than hardware with interrupt, once you get above 20MHz on most processors the interrupt overhead is less noticeable and even if it ends up like Uno/Mega boards being a minimum of actually 4 or 2 micro seconds for XMC1xxx that is liveable with for most purposes. On Liquid Crystal there is a minimum delay for 4 bit interface that is 1 micro second but will work if longer, however default libray has 100 micro second delay between nibbles and is way slow, so I made a new one.

The reason I look at things like this is many school projects I get involved in mentoring use these functions.

techpaul, 2018-03-09:
See Pull #42 for details

techpaul, 2018-03-30:
As most of this is now incorporated in V1.1.0 of board library this issue is now closed

Future update will look at using Hardware timer for microsecond delay for better accuracy and allowing more interrupt tasks to operate.

closed 2018-03-30 by techpaul