Infineon/XMC-for-Arduino #38: Serial and systick events handling

techpaul, 2018-03-04:
Whilst double and triiple checking previous patches I discovered a couple of bugs, also I was trying to track down the serial transmit issues, when using simple test code

```
#include <Arduino.h>

extern uint8_t tone_pins[ NUM_TONE_PINS ];
extern uint8_t tone_irq[ NUM_TONE_PINS ];
extern uint32_t timer_ids[ NUM_TONE_PINS ];
extern volatile long timer_toggle_count[ NUM_TONE_PINS ];

void setup()
{
Serial.begin( 115200 );
//tone( LED1, 1 );
//tone( LED2, 4, 10000 );
//tone( LED5, 10, 5000 );
//tone( LED6, 20, 3000 );
delay( 45 );
Serial.println( "pin\tirq\tid\tCount" );
for( int i = 0; i < NUM_TONE_PINS; i++ )
  {
  Serial.print( tone_pins[ i ] );
  Serial.print( "\t" );
  Serial.print( tone_irq[ i ] );
  Serial.print( "\t" );
  Serial.print( timer_ids[ i ] );
  Serial.print( "\t" );
  Serial.println( timer_toggle_count[ i ] );
  }
}

void loop() {
}
```
First problem unless there is about 45 ms delay between Serial.begin and the first Serial.print or serial .write, no serial output occurs.

Second problem if multiple systeick events are setup, which are only done for delay and tone functions, only the last is actioned until complete, then the next until completion.

Basically Systick handlers doing scheduling of tasks has so many pointers and lack of correct scanning of linked lists that it only processes the last event, not the whole list. This has obviously only been tested with one event in play at any time. ther are countless other issues with what should be volatile and less bloat as the maximum events being handled is known at compile time, so does not need linked list but fixed array with enable flags.

Next stage as soon as I get some time is to port some of my own scheduling code that is known to work to replace the linked list fiasco. Also I will see about making RX as well as TX using the FULL 32 word FIFO to allow the hardware to complete whilst interrupts are turned off for delaymicroseconds and no doubt other places. All things Systick related will use milli seconds as minimum resolution as 1 ms is the smallest increment possible anyway.

On >16 MHz processors and lots of I/O 4 tone pins is adequate, whilst on >100 MHz processors 16 tone pins should be possible easily.

Once I get a stage that is working well for systick events at 1 kHz and allow for different numbers of tone outputs on different platforms I will probably tag a level of my fork of the repository as V1.06a.

Hopefully I might stumble across the serial issues as well as a better way to do hardware based delaymicroseconds.

techpaul, 2018-03-05:
I have a majorly rewritten wiring_time and tone modules that works with above example (modified for remaining variables.

Running the task loop with FOUR tone signals and multiple calls to delay with some rough timing tests gives one pass through scheduler loop and run the call back tasks with execution times ranging from
**5 to 20 microseconds**, Depenfing on number of tasks in each loop pass

Leaving most of the 1 ms available for other processing in user code. The code is adaptable and changing new define in pins_arduino.h for a specific processor for example from 4 to 8 will ripple through all files. This may be useful for those using XMC4xxx series.

Unfortuantely to be compliant with Arduino delay() and its knock on effects the interval or delay time is uint32_t, so we could sit there for 50 years if required. There are better ways to do that in software and hardware.

More on serial startup delay, serial and analogRead interaction when I get to chase that down.

This patch and a few others hopefully will be pushed later this week.

As this is a major change there is a LOT of comments especially in wiring_time.c This is the extra added at the top of the file
```
/* Includes Co-operative Scheduler for XMC-for-Arduino

   Using COMPILE time scheduling table 

Version V1.00
Author: Paul Carpenter, PC Services, <sales@pcserviceselectronics.co.uk>
Date    March 2018

Co-operative scheduler library that has some support functions and main 
schedule function meant to be run as the Systick scheduler event handler, the 
support functions should be used to get status on, start, stop, or configure tasks.

The schedule code is designed for MINIMAL overhead and SPEED.

The time interval of calling the schedule loop is determined by Systick timer 
interval currently 1 ms, and calls a selection of tasks determined by two
parameters for each task

    callback    function address
    parameter   INTEGER to pass to function
    
From this we get TWO types of function calls when a task event happens

1/ callback and parameter are set gives a function call of

    int callback( int ID, int16_t param )
            function to use as this task that accepts TWO integer parameters
            and MUST return an integer

   (even if parameter not set it is initialised to task_id AT compile TIME)
    
2/ Special case if ANY task has its callback set to NULL a special variable
   in wiring_time is set to TRUE. This enables the delay() function to work
   for ms delays. As a side effect it stops runaway tasks for callback set to 
   NULL. 
                            
Callback functions in list
--------------------------
    For different processors speeds it is suggested to keep the number of 
    total tasks (callback functions) to following limits -
    
    Speed               Max Tasks
    < 50 MHz            8
    > 50 MHz < 100 MHz  12
    > 100 MHz           16

Callback function structure
---------------------------
    Task is actually calling a function that should be defined as 
    function that returns int and takes TWO integer parameter

    e.g.    int func( int TaskId, (int16_t Parameter )
    First parameter is TaskId,
    Second parameter is user defined parameter (int16_t)

    If interval of a task has to change call setInterval during task execution
    NOT to STOP task use return code

Callback functions Contents
---------------------------
    Callback functions are effectively interrupt routines so care should be taken 
    about which variables and flags set in these functions are volatiles.
    
    Functions MUST be VERY SMALL as many tasks have to run in less than 1 ms as 
    well as the main code.
    
    Do NOT use blocking or slow callbacks that use AT LEAST the following Ardunio
    functions
        delay           (this would become recursive)
        .print, .write, println on Serial or Liquid Crystal
        analogread
        PulseIn
        Liquid Crystal  ANY function
        I2C or SPI      ANY function
        SD/MMC          ANY function

    All of these and maybe others are SLOW, block waiting for I/O or time
    delays often MUCH longer than 1 ms

Callback Function Return value
------------------------------
    The returned value is the NEW status for that task which tells scheduler
    what to do next time. Values are

      < 0 User error status (stops task execution)
        0  Task stopped (if needed to be run will have to be started by Start)
      > 0 Next status this can be used by call back function for state machines
          Task runs again at the interval set

Scheduling Functions
--------------------
setInterval Set a task's NEW interval and schedule new time from now if not
            already running
getInterval Get a task's interval
setParam    Set a task's callback function parameter (signed int16_t)
getParam    Get a task's callback function parameter (signed int16_t)
getTime     Get a task's next time to execute
getStatus   Get a particular task status word
StartTask   Start a task (if not already running)
FindID      get ID of first task from task address
*/
```

techpaul, 2018-03-09:
See Pull #42 for results for wiring_time

Delaymicroseconds is one of next tasks

techpaul, 2018-03-10:
Currently the serial side of needing a delay after serial.begin  before using the serial port, is looking like somewhere in the USB debugger, programmer changeover to serial port mode, or host side.

Resurrecting a TTL to RS232 converter to use a real serial port and do side by side coimparisons.

techpaul, 2018-03-11:
Tests confirm having to put a delay before using serial functions after a begin is debugger or further up.

When using a putty session on a real com port alonside the arduino IDE serial monitor the USB vcersion is corrupted. The Arduino IDE is one of the few serial monitors that will stay running while programming or power cycling the unit (XMC1100 boot kit). Most close when USB serial port disappears.

Screen shots of WithDelay

[image]

Screen shot of NoDelay

[image]

This needs to be looked at by different folks who understand the Segger Jlink and debugger and possibly the Arduino interface to that for changing over to allow serial.

techpaul, 2018-03-11:
Further tests indicate the issue is related to integration of programmer and virtual com into IDE for serial monitor. Possibly with how the IDE sends target resets and does not check responses or timesout expecting a set sequence to occur then messes up.

techpaul, 2018-03-18:
Further evidence there is a problem between the Virtual COM driver and/or the Segger JLink when coming out of programme mode Basically a stream of garbage characters after programme mode before the micro is reset

the images are of side by side views from USB port (COM30) and a TTL converter on COM2. 
First using Arduino Serial Monitor,

[image]

 second using Putty on USB.
[image]

mhollfelder, 2018-11-05:
Sorry for replying with such a delay - I had this issue on my agenda, but forgot to reply.

I am facing currently some more issues with the VCOM port - do these issues still persist on your side?
Just would like to get a fresh update for next steps.

Thank you very much and best regards,

Manuel

techpaul, 2018-11-05:
Just back in I will look at doing same tests again as soon as possible.

closed 2019-07-15 by techpaul