Infineon/XMC-for-Arduino #39: Suggestion on wiki

techpaul, 2018-03-04:
I have now created a wiki with my notes and what I am currently working on.

These notes maybe helpful to others and you might want to add link to it on your wiki

[Techpaul Wiki](https://github.com/techpaul/XMC-for-Arduino/wiki)

techpaul, 2018-03-20:
[Techpaul Wiki](https://github.com/techpaul/XMC-for-Arduino/wiki) has just  had some major updates and reformating for more readability and details of what has been added and improved in V1.06c version there awaiting merging into new release candidate

mhollfelder, 2018-11-05:
If okay for you, I will now merge your wiki with the existing one and copy relevant information.
Just would like to align with you before doing so - there is lot's of relevant information for all XMC users.

Best regards,

Manuel

techpaul, 2018-11-05:
That is fine by me more will be added when I get a few moments for updates

closed 2019-09-13 by techpaul