Infineon/XMC-for-Arduino #44: XMC 4700 Pin Out

Andiiiy, 2018-03-14:
Hey Manuel,
it seem to be a mismatch at pin out at XMC 4700 (pins_arduino.h)!

#define A0   17 (at the picture Arduino pin 22) 
#define A1    18 (at the picture Arduino pin 21)
#define A2   19 (at the picture Arduino pin 20)
#define A3   20 (at the picture Arduino pin 19)
#define A4   21 (at the picture Arduino pin 18)
#define A5   22 (at the picture Arduino pin 17)

Regards Andreas

techpaul, 2018-03-15:
Actually both types of pin numbers are wrong and hopefully in V1.1.0 will show as 

#define A0 0
#define A1 1
#define A2 2
#define A3 3
#define A4 4
#define A5 5
.......

Arduino uses analogread by passing in the ADC channel number not the pin number.

This was in one of my previous pull requests for tidying up analog, if you want I can find the one loater

Andiiiy, 2018-03-15:
Fixed with new version 1.1.X .... special thanks to techpaul! Great work ...

closed 2018-03-15 by Andiiiy