Infineon/XMC-for-Arduino #46: USB and onboard Serial XMC2GO

H1PP011, 2018-03-27:
Althougt it says so in the wiki page of the XMC2GO, by default there is only serial, which is the usb serial, and no serial1. I already found out, that the onboard serial is somehow disabled in [this](https://github.com/Infineon/XMC-for-Arduino/blob/master/arm/variants/XMC1100/config/XMC1100_XMC2GO/pins_arduino.h)  core file, but i couldn't get it to work.

techpaul, 2018-03-27:
You can only use one or the other for RX and TX serial data. So if using USB serial you cannot use the serial pins. If you wish to use the pins on the XMC2GO for serial comms and NOT use USB serial you need to change the HardwareSeial1 line at end to use the other configuration structure.

There are some fixes to pins_arduino.h to make the swapping easier that are currently being incorporated into the next release, I believe currently incorporated into develop branch. Getting the newer pins_arduino.h for XMC2GO would probably be enouigh and the details on changing are in comments at top iof file.

mhollfelder, 2018-03-27:
As techpaul said, only one serial can be used at a time (if you'd like to use other communication protocols as well).
The XMC1100 has one USIC peripheral with two channels, i.e. only two independent protocols are supported simultaneously. As the on-board and the USB serial use different pins, both cannot share one channel.

The next release v1.1.0 will add a menu to the Arduino IDE to enable quick switching of the two solutions. 
I will release the new version today or latest tomorrow after last checks. For the time being, please use the develop branch and have a look in the pins_arduino.h of the XMC2Go variant. 

If you need further help, please let me know.

Best regards,

Manuel

techpaul, 2018-03-27:
If you get the latest XMC2GO pins_arduino.h from develop branch you might find this page helpful
[https://github.com/techpaul/XMC-for-Arduino/wiki/Serial-Debug-XMC1xxx-Series](url)

H1PP011, 2018-03-27:
Thank you for your answers,
I had known how to change between both of them, but I thought you could also enable both of them.
I need to receive something from the onboard rx and send trough usb tx, but i don´t need onboard tx and usb rx. Is there something like SoftwareSerial?

techpaul, 2018-03-27:
Standard SoftwareSerial relies on hardcoded asm instructions for AVR, unless you can find an alternative 3rd party library that is C/C++ only.

Alternatively you could TRY reconfiguring the pins used for Serial debug and change the RX pin to the onboard pin. Then you will need to do this change everyfuture  library update.

Or do a hardware mod on your board to disconnect the USB RX and physically connectr the hardware RX board pin to correct track

mhollfelder, 2018-03-27:
This is possible as the USIC channel could share both pins:

- TX could go to USB serial
- RX could come from the on-board serial

I attached a changed file for the pins_arduino.h of the XMC2Go from the develop branch.
I changed the `#ifdef SERIAL_DEBUG` to `#ifndef SERIAL_DEBUG` at the respective places to make this connection possible. By undefining `SERIAL_DEBUG`, you would now swap TX and RX of the USB serial and on-board serial.

I hope this works as I could not check it right away here :)

Best regards,

Manuel

[pins_arduino.zip](https://github.com/Infineon/XMC-for-Arduino/files/1852073/pins_arduino.zip)

mhollfelder, 2018-03-28:
The release will come tomorrow, not today as I need to wait for HW to test some more features.

In the meantime, I updated the develop branch as there were some problems during compilation coming from the merges.
If you face issues, please let me know.

Best regards,

Manuel

mhollfelder, 2018-03-30:
There were some more changes, but finally the new release v1.1.0 has been released.
Please let me know if you need anything or you encounter additional problems.

Best regards,

Manuel

mhollfelder, 2018-04-13:
Did it work our or do you need additional help?
If not, I will close this issue, but feel free to open it again.

Best regards,

Manuel

H1PP011, 2018-04-13:
At the moment it's fine. Thanks for your help!

Nyceane, 2018-07-08:
Hey guys, the v1.1.0 still does not have softwareserial... any idea on how to use it?

techpaul, 2018-07-08:
BTW most folks have updated to V1.1.1 and minor patches by now.

SoftwareSerial by Arduino is written ONLY for AVR platforms as it uses specific Assembler and hardware in a specific way that only works on AVR micros.

You are welcome to submit a version that could work on XMC series.

It is way down the list of priorities for moist contrubutors, ability to change other I2C or SPI to Serial is more likely but the pin definitions for each type of micro is complex to do. Personally I have RTC then Wire then HardwareSerial improvements then seeing how to add any digital pin to interrupt as higher priorities.

closed 2018-09-13 by sherylll

xzsfd2457, 2019-05-31:
> BTW most folks have updated to V1.1.1 and minor patches by now.
>
> SoftwareSerial by Arduino is written ONLY for AVR platforms as it uses specific Assembler and hardware in a specific way that only [...]

can we get Your code pls??

techpaul, 2019-05-31:
All past and future improvements are incorporated into Pull requests on here and are shown as open issues. I do not think you know how Github works.

I am not paid to do this work I am a contributor with many years of embedded experience.

Currently I hv many PAYING projects on the go at same for things like
1. IoT remote sensor networks
2. Laser gun targetting systems
3. Electric Wheelchair/walking aid controllers

These are products that go into production and have to have all sorts of certification tests, so PAYING work takes priority over these improvements.