Infineon/XMC-for-Arduino #51: New Wiki errors XMC1100 H Bridge 2 Go page

techpaul, 2018-04-11:
In key features says can be powered by 3V3.

Manual for the board says external 5V (to save the on board regulator) they actually suggest using board pin 29 for +5V and 28 and/or 27 for GND.

In last paragraph there are mentions of XMC2Go and H bridge in last sentence

I would suggest change "XMC2Go" to "H-Bridge 2Go"

Also last "H bridge" to "H bridge chip"

Will check through other parts later

mhollfelder, 2018-04-13:
Dear Paul,

Thank you once again for reporting this and giving feedback.

Actually, independently whether you power it via 3.3 V or 5V, the final voltage for logic is 3.3V, i.e. either one goes 5V (USB | Ext. 5V) -> 3.3 V or directly 3.3 V, there should be no difference. 

I documented these points in the wiki after comparing it with the schematics in the respective manual, but if I oversaw something, please let me know.

The other points have been changed by me (I used IC to show that it is an integrated circuit).

Thank you very much and best regards,

Manuel

techpaul, 2018-04-13:
Glad it got fixed., thanks for updating

Will close issue

closed 2018-04-13 by techpaul