Infineon/XMC-for-Arduino #52: Version 1.1.0 breaks Serial.available()

sehraf, 2018-04-13:
Since version 1.1.0 Serial.available() will always return 0.

After updating my code stopped working. I figured out that Serial.available() causes the problems since it won't return something != 0.

The following code will never send 'a':
``` C
#define PIN 0x0e
//#define PIN 0x0f

void setup() {
  Serial.begin(115200);
  // put your setup code here, to run once:
  pinMode(PIN, OUTPUT);
}

void loop() {
  // put your main code here, to run repeatedly:
  digitalWrite(PIN, HIGH);
  delay(500);
  digitalWrite(PIN, LOW);
  delay(500);

  if (Serial.available())
    Serial.write('a');
  else
    Serial.write('b');
}
```

Downgrading to 1.0.6 makes it work again and upgrading to 1.1.0 breaks it again.

techpaul, 2018-04-13:
There is something else wrong as I have just copied your code and tried it and works using setup

     XMC1100 Boot Kit
    Arduino IDE V1.85
    XMC-for-Arduino V1.1.0
    WinXP host (32 bit)
   Standard serial Debug setup (menu item "Tools -> Serial Output Selection : PC")

I see continual 'b' written to serial output until a character is sent to the board then continual 'a'

Also using one of my standard test functions testdefines in
      .. ..../Arduino15\packages\Infineon\hardware\arm\1.1.0\tests

What board are you using for Arduino pin of 14 or 15 ?

techpaul, 2018-04-13:
Also your code OK on XMC1300 Boot Kit

That is the extent of current boards I have here

sehraf, 2018-04-13:
Sorry forgot to mentioned that: 
 I'm using
* XMC2GO board (with XMC1100)
* Host is Linux (Archlinux)
* Arduino IDE 1.8.5

Pin 14  and 15 are the two LEDs on the board.

techpaul, 2018-04-13:
At the moment I cannot see what is wrong as works fine on two boards I have here.
Hopefully someone with one of those boards can help you with confirming the error first then we can chase it down..

You might want to use built-in defines for LED pins on XMC2Go they are LED1 and LED2 to make code easier to read. For other info that might help see my wiki as well as one here

https://github.com/techpaul/XMC-for-Arduino/wiki

Normally when checking Serial.available I use

     if( Serial.available( ) > 0 )

To avoid issues on some other manufacturers boards where their implementation that returns a signed integer.

Other suggestion is to save Serial.available result to a variable and print it out as well

sehraf, 2018-04-13:
Thanks for the link. I wasn't aware of the defines :wink: 

I'll stick to XMC-for-Arduino V1.0.6 for the moment. Thanks for your help!

mhoechner, 2018-04-13:
hi sehraf,

there was a wrong input source for the rx pin. I fixed it within #53.

BR
Moritz

mhollfelder, 2018-04-13:
Dear sehraf,

Thank you very much for this report.

As my colleague Moritz could fix it, I will merge it and include it in the next release.
The next release will come shortly to fix the other issues as well.

If you need more help, please let us know.

@techpaul: Thank you very much for supporting here proactively

Have a nice weekend and best regards,

Manuel

sehraf, 2018-04-13:
Your welcome!
Thanks for the quick response and fix :+1:

mhollfelder, 2018-04-13:
You're welcome, no problem!
Fixed in current branch 'master', please just copy pins_arduino.h for XMC 2Go and overwrite it locally to get the changes right now.

Let us know if you need more help.
I will close this issue, feel free to reopen it again if you face issues.

Thank you very much,

Manuel

closed 2018-04-13 by mhollfelder

sehraf, 2018-04-13:
It's working again!

techpaul, 2018-04-13:
Glad to hear that, puts hands up to say sorry it was one of my patches that caused that to fail on looking at changes.

Sorry for causing you hassles