Infineon/XMC-for-Arduino #58: SPI works very slow with OLED-Display and u8g2 library

HMenger, 2018-05-21:
I made some test with I2C (SH1106) and SPI (SSD1309)  with  128x64 OLED displays using u8g2 library, and compared with 2 other Boards ( Teensy 3.2 and TeensyLC ) i have here.
While I2C deliveres more or less the same result for filling the screen on all plattforns -> 1kB needs about 33ms to be sent to the Display on 400000Hz, i found that SPI workes much to slow .
on the XMC1404 it takes more then 60ms and on the XMC2Go more than 200ms, while on the teensy boards it is only 2,7ms ( Teensy 3.2@ 48MHz ) and 3,5 ms (TeensyLC@48MHz) .
I will do some more test with XMC1300 and XMC4700 boards.

best regards

Hubert

HMenger, 2018-05-21:
Made another trial using Adafruit SSD1306 library with seems to work better with XMCbaords,  

here the result  : 
XMC2Go ~ 13,4 ms
XMC1404 ~ 11,3 ms
TeensyLC ~ 3.0 ms
Teensy3.2 ~1,6 ms

this is how the time for  transfer the data to the display is measured:
  microsold = micros();
    display.display(); // send data to display
  microsnew = micros();
  microsdiff = microsnew - microsold;

so it seemes u8g2 lib doesn't use HW-SPI  for some reason, with the XMC boards , although it is written that it uses Arduino library for I2C and SPI .

Is it possible to speed up SPI to reach the allmost same speed as on Teensy boards?

regards
Hubert

techpaul, 2018-05-21:
A quick look at u8g2 on github and just looking at src directory last fix status list various fixes were for changing to H/W SPI and may not have been done for all the displays.

Also note quite a few say

"reduced SPI clock speed for SH1106"  issue 551

I am not sure if the problem is this library but u8g2

HMenger, 2018-05-22:
Yes you are right , the problem depends on u8g2 lib, i just wondered why it is so slow and works fine with the Teesny boards.
So no problem with the XMC-SPI-lib.

i did some measurment with the scope and found were is the differenz in speed when using the Adafruit lib.

One thing is the clock-speed:
XMC2go       4MHz
TeensyLC      6MHz
Teensy3.2     8Mhz

the other is the  interval between two bytes are written to the bus:
XMC2Go     19µs
TeensyLC       7µs
Teensy3.2    5.5µs

I had a look at the Teensy SPI-lib and saw it is using DMA-transfer and is programmed close to the hardware .... far beyond my limited horizon 

regards
Hubert

closed 2018-05-22 by HMenger