Infineon/XMC-for-Arduino #60: Problem using XMC for Arduino in Ubuntu 16.04

odo2063, 2018-06-11:
I set up a new system with a distro based on ubuntu 16.04. 
I have installed everything in 64bit(tripple checked).
but when i try to upload the software to the MCU i get following errormessage:

# BridJ: dlopen error when loading /opt/SEGGER/JLink_V632e/libjlinkarm_x86.so.6.32.5 : /opt/SEGGER/JLink_V632e/libjlinkarm_x86.so.6.32.5: wrong ELF-Class: ELFCLASS32
Exception in thread "main" java.lang.UnsatisfiedLinkError: jlinkarm.JLinkARMLibrary.JLINKARM_GetDLLVersion()I
	at jlinkarm.JLinkARMLibrary.JLINKARM_GetDLLVersion(Native Method)
	at Infineon.JLinkDebugger.GetDLLVersion(Unknown Source)
	at Infineon.XMCDevice.Program(Unknown Source)
	at Infineon.upload.main(Unknown Source)

Since starting the jlink-console from terminal works I don't think it's a problem on seggers site, but these are just my 2 ct.

techpaul, 2018-06-11:
As you do not say which board 

Can i suggest you look at issue #21 

Which is about Linux and XMC2GO, there is a symlink there that apparently fixes problems for some users

odo2063, 2018-06-12:
Thank You very much. This works like a charm. This fixes the ELFCLASS32 Problem

To make it googleable, should also work for Ubuntu 13.04 13.10 14.04 14.10 15.04 15.10 16.04 16.10 17.04 17.10 18.04

And as it is said much too less, you guys here doing a really great job!!!

closed 2018-06-12 by odo2063