Infineon/XMC-for-Arduino #61: Compiler error when using Visual Micro/Atmel Studio7

StephiB, 2018-06-11:
Working perfect in Arduino IDE, but when compiling in Visual Micro/Atmel Studio7 the following Error occurs:

`Compiling`` 'Sketch1' for 'XMC1100 Boot Kit'
arm-none-eabi-g++*: warning: -x c++ after last input file has no effect
 
arm-none-eabi-g++*: fatal error: no input files
   compilation terminated
Build failed for project 'Sketch1'`

techpaul, 2018-06-13:
So many possibilities, from not having correct platform support in Visual Micro/Atmel Studio7 to not passing the basic extension of Sketch1.ino and expecting a default Sketch1.cpp

Arduino IDE expects the main file to be a .ino file gnu compilers assume .c or .cpp

sherylll, 2018-08-08:
At the moment only Arduino IDE & PlatformIO are supported.

closed 2018-08-08 by sherylll