Infineon/XMC-for-Arduino #62: Interrupt input pin is high ohmic after startup, but activates pull down on first interrupt

ventilator, 2018-06-13:
**Affected product:** XMC4700 Relax kit
**Expected behavior:** Internal pull down of an input pin can be configured or at least stays constant (as seen on Arduino Mega)
**Seen behavior:** Internal pull down setting changes if an interrupt is received (seen on XMC4700 Relax Kit)

**Reproduce:**

Consider this demo code from the arduino IDE which waits (attaches an interrupt) on a signal change on pin2 and changes output signal on pin 7 accordingly.

```
const byte firePin = 7;
const byte interruptPin = 2;
volatile byte state = LOW;

void setup() {
  pinMode(firePin, OUTPUT);
  pinMode(interruptPin, INPUT);
  attachInterrupt(digitalPinToInterrupt(interruptPin), blink, CHANGE );
}

void loop() {
  digitalWrite(firePin, state);
}

void blink() {
  state = !state;
}
```
If you observe this on the oscilloscope after boot up, you can see (yellow signal), that the input signal is around 5V. This signal is supplied by an pull up of an attached sensor (pull up to 5V via 1.2kOhm). The XMC kit is configured to wait on a signal change on pin 2. Once the interrupt occurs, it is recognized and accordingly an output pin is set (pin 7, blue). 

Then the sensor tries to go back to high signal, 5V but fails to do, since the XMC Relax kit has reconfigured the input pin, now activating its internal pull down (which is now in conflict with the pull up of the sensor). The sensor is only able to go up to ~2.5V, so the second change is not recognized anymore by the interrupt

Can the input pin configured high-impedance state, no pull down?

[image]

The same code running on an Arduino Mega with the same sensor setup shows that on this plattform the pull down is not activated and thus the sensor is able to reach up again to 5V.

[image]

techpaul, 2018-06-13:
Not looked at code (YET) This sounds like worng mode selection in 4700 code

Have you tried commenting out the line to set input mode, e.g.

` // pinMode(interruptPin, INPUT);
`

Note you do not need the line on a Mega as Mega/Uno etc code sets all pins to input as part of startup and initialisation in hardware.

techpaul, 2018-06-15:
Check the Pre-Release ov v1.1.1 that has some GPIO fixes that may well help

mhollfelder, 2018-11-05:
I checked `WInterrupts.c` and `WInterrupts.h` once again for the current version at V1.2.0 and could not find any hints for the reported issue on implementation level - did you test it only for the XMC4700 Relax Kit or do you have additional information?

I need to trace it down with a connected scope and a live session of the debugger - mainly to see when the flipping happens by following the full call stack.

That's a delicious problem ;)

Best regards,

Manuel

closed 2021-09-07 by jaenrig-ifx