Infineon/XMC-for-Arduino #65: Problem with Update to v1.1.1

odo2063, 2018-06-19:
Hi!

ArduinoIDE tells me there is a new Version but it won't let me install/download it.

Fehler beim Herunterladen von https://github.com/Infineon/XMC-for-Arduino/releases/download/V1.1.1/XMC_IFX_1.1.1.zip
java.lang.RuntimeException: java.lang.Exception: Fehler beim Herunterladen von https://github.com/Infineon/XMC-for-Arduino/releases/download/V1.1.1/XMC_IFX_1.1.1.zip
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:176)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Fehler beim Herunterladen von https://github.com/Infineon/XMC-for-Arduino/releases/download/V1.1.1/XMC_IFX_1.1.1.zip
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:131)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:69)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:102)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:173)
	... 1 more
Caused by: java.io.IOException: Received invalid http status code from server: 416
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:194)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:128)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:129)
	... 4 more

odo2063, 2018-06-19:
1.1.0 works fine....

techpaul, 2018-06-19:
Appears to be available now was originally a Pre-Release now it is latest version

Got download working now, take a little while as loading a new gcc compiler

odo2063, 2018-06-20:
Still not workin here(linux)....:-/

Fehler beim Herunterladen von https://github.com/Infineon/XMC-for-Arduino/releases/download/V1.1.1/XMC_IFX_1.1.1.zip
java.lang.RuntimeException: java.lang.Exception: Fehler beim Herunterladen von https://github.com/Infineon/XMC-for-Arduino/releases/download/V1.1.1/XMC_IFX_1.1.1.zip
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:176)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Fehler beim Herunterladen von https://github.com/Infineon/XMC-for-Arduino/releases/download/V1.1.1/XMC_IFX_1.1.1.zip
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:131)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:69)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:102)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:173)
	... 1 more
Caused by: java.io.IOException: Received invalid http status code from server: 416
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:194)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:128)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:129)
	... 4 more

techpaul, 2018-06-20:
Now seeing same problem here on Windows, there seems to be a package build issue

@mhoechner The package needs inspection and checks and probably seeing http codes needs to be checked for what is included as code means "Requested Range Not Satisfiable"

mhollfelder, 2018-06-22:
I found the issue (I hope):

there was a wrong file size in the JSON causing the Arduino IDE to request a different package size from the server as what it actually got. This might cause the http 416 error.

Secondly, there was an error with the tool names and links for other platforms which caused additional troubles.

Could you test it and give feedback here?

Thank you very much and best regards,

Manuel

techpaul, 2018-06-22:
Just managed a Windows platform install V1.1.1

Test compile on XMC 1100 Boot Kit compiled correctly showing gcc 5.4

nmaas87, 2018-06-22:
Can acknowledge that as well, update worked on Windows 👍

techpaul, 2018-06-22:
Time to rebase forks an resume I2C fixes

techpaul, 2018-06-22:
Just done a test with Ubuntu Linux

As Update reported a CRC error

Re run IDE nad now having V1.1.0 uninstalled, would do install successfully.

Something strange on an update or maybe a network glitch

odo2063, 2018-06-23:
Updating now seems to work on my linux (quite old ubuntu)...

but know i got an error compiling for XMC2go

/home/teste/Projekte/arduino-1.8.5/arduino-builder -dump-prefs -logger=machine -hardware /home/teste/Projekte/arduino-1.8.5/hardware -hardware /home/teste/.arduino15/packages -tools /home/teste/Projekte/arduino-1.8.5/tools-builder -tools /home/teste/Projekte/arduino-1.8.5/hardware/tools/avr -tools /home/teste/.arduino15/packages -built-in-libraries /home/teste/Projekte/arduino-1.8.5/libraries -libraries /home/teste/Arduino/libraries -fqbn=Infineon:arm:XMC1100_XMC2GO:UART=debug -ide-version=10805 -build-path /tmp/arduino_build_27528 -warnings=default -build-cache /tmp/arduino_cache_446236 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/teste/.arduino15/packages/Infineon/tools/arm-none-eabi-gcc/5.4-2016q3 -prefs=runtime.tools.XMCFlasher.path=/home/teste/.arduino15/packages/Infineon/tools/XMCFlasher/1.2.1 -verbose /home/teste/Arduino/Sensirion_SDP6x_pressure_sens-master/SDP6x/SDP6x.ino
/home/teste/Projekte/arduino-1.8.5/arduino-builder -compile -logger=machine -hardware /home/teste/Projekte/arduino-1.8.5/hardware -hardware /home/teste/.arduino15/packages -tools /home/teste/Projekte/arduino-1.8.5/tools-builder -tools /home/teste/Projekte/arduino-1.8.5/hardware/tools/avr -tools /home/teste/.arduino15/packages -built-in-libraries /home/teste/Projekte/arduino-1.8.5/libraries -libraries /home/teste/Arduino/libraries -fqbn=Infineon:arm:XMC1100_XMC2GO:UART=debug -ide-version=10805 -build-path /tmp/arduino_build_27528 -warnings=default -build-cache /tmp/arduino_cache_446236 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/teste/.arduino15/packages/Infineon/tools/arm-none-eabi-gcc/5.4-2016q3 -prefs=runtime.tools.XMCFlasher.path=/home/teste/.arduino15/packages/Infineon/tools/XMCFlasher/1.2.1 -verbose /home/teste/Arduino/Sensirion_SDP6x_pressure_sens-master/SDP6x/SDP6x.ino
Using board 'XMC1100_XMC2GO' from platform in folder: /home/teste/.arduino15/packages/Infineon/hardware/arm/1.1.1
Using core './' from platform in folder: /home/teste/.arduino15/packages/Infineon/hardware/arm/1.1.1
Detecting libraries used...
"/home/teste/.arduino15/packages/Infineon/tools/arm-none-eabi-gcc/5.4-2016q3/bin/arm-none-eabi-g++" -mcpu=cortex-m0 -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DXMC1100_T038x0064 -DXMC1100_XMC2GO -DF_CPU=32000000L -DARDUINO=1100 -DARDUINO_ARM_XMC -DARDUINO_ARCH_ARM  -DSERIAL_HOSTPC "-I/home/teste/.arduino15/packages/Infineon/hardware/arm/1.1.1/cores/xmc_lib/XMCLib/inc"   "-I/home/teste/.arduino15/packages/Infineon/hardware/arm/1.1.1/cores/xmc_lib/CMSIS/Include"  "-I/home/teste/.arduino15/packages/Infineon/hardware/arm/1.1.1/cores/xmc_lib/LIBS"  "-I/home/teste/.arduino15/packages/Infineon/hardware/arm/1.1.1/variants/XMC1100"  "-I/home/teste/.arduino15/packages/Infineon/hardware/arm/1.1.1/variants/XMC1100/config/XMC1100_XMC2GO" "-I/home/teste/.arduino15/packages/Infineon/hardware/arm/1.1.1/cores/avr" "-I/home/teste/.arduino15/packages/Infineon/hardware/arm/1.1.1/cores" "-I/home/teste/.arduino15/packages/Infineon/hardware/arm/1.1.1/variants/XMC1100" "/tmp/arduino_build_27528/sketch/SDP6x.ino.cpp" -o "/dev/null"
"/home/teste/.arduino15/packages/Infineon/tools/arm-none-eabi-gcc/5.4-2016q3/bin/arm-none-eabi-g++" -mcpu=cortex-m0 -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DXMC1100_T038x0064 -DXMC1100_XMC2GO -DF_CPU=32000000L -DARDUINO=1100 -DARDUINO_ARM_XMC -DARDUINO_ARCH_ARM  -DSERIAL_HOSTPC "-I/home/teste/.arduino15/packages/Infineon/hardware/arm/1.1.1/cores/xmc_lib/XMCLib/inc"   "-I/home/teste/.arduino15/packages/Infineon/hardware/arm/1.1.1/cores/xmc_lib/CMSIS/Include"  "-I/home/teste/.arduino15/packages/Infineon/hardware/arm/1.1.1/cores/xmc_lib/LIBS"  "-I/home/teste/.arduino15/packages/Infineon/hardware/arm/1.1.1/variants/XMC1100"  "-I/home/teste/.arduino15/packages/Infineon/hardware/arm/1.1.1/variants/XMC1100/config/XMC1100_XMC2GO" "-I/home/teste/.arduino15/packages/Infineon/hardware/arm/1.1.1/cores/avr" "-I/home/teste/.arduino15/packages/Infineon/hardware/arm/1.1.1/cores" "-I/home/teste/.arduino15/packages/Infineon/hardware/arm/1.1.1/variants/XMC1100" "-I/home/teste/.arduino15/packages/Infineon/hardware/arm/1.1.1/libraries/Wire/src" "/tmp/arduino_build_27528/sketch/SDP6x.ino.cpp" -o "/dev/null"
"/home/teste/.arduino15/packages/Infineon/tools/arm-none-eabi-gcc/5.4-2016q3/bin/arm-none-eabi-g++" -mcpu=cortex-m0 -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DXMC1100_T038x0064 -DXMC1100_XMC2GO -DF_CPU=32000000L -DARDUINO=1100 -DARDUINO_ARM_XMC -DARDUINO_ARCH_ARM  -DSERIAL_HOSTPC "-I/home/teste/.arduino15/packages/Infineon/hardware/arm/1.1.1/cores/xmc_lib/XMCLib/inc"   "-I/home/teste/.arduino15/packages/Infineon/hardware/arm/1.1.1/cores/xmc_lib/CMSIS/Include"  "-I/home/teste/.arduino15/packages/Infineon/hardware/arm/1.1.1/cores/xmc_lib/LIBS"  "-I/home/teste/.arduino15/packages/Infineon/hardware/arm/1.1.1/variants/XMC1100"  "-I/home/teste/.arduino15/packages/Infineon/hardware/arm/1.1.1/variants/XMC1100/config/XMC1100_ [... cut ...]
         ^
/home/teste/.arduino15/packages/Infineon/hardware/arm/1.1.1/libraries/Wire/src/Wire.cpp:176:9: sorry, unimplemented: non-trivial designated initializers not supported
Bibliothek Wire in Version 1.1 im Ordner: /home/teste/.arduino15/packages/Infineon/hardware/arm/1.1.1/libraries/Wire  wird verwendet
Bibliothek SPI in Version 1.1 im Ordner: /home/teste/.arduino15/packages/Infineon/hardware/arm/1.1.1/libraries/SPI  wird verwendet
exit status 1
Fehler beim Kompilieren für das Board XMC1100 XMC2Go.

techpaul, 2018-06-23:
It is a minor chnage to files to Wire and SPI i thought had been done as a hot fix see Pull Request #63 and files listed there it is a minor change

Or copy over the new versions of the two files from master repository

techpaul, 2018-06-23:
I think this issue is closed if so can you close the issue

odo2063, 2018-06-24:
Works with manually overwriting HW_SPI.cpp and Wire.cpp from master branch...
...not easy for beginners but works...i close this...
and again THANKS a lot!!!

closed 2018-06-24 by odo2063