Infineon/XMC-for-Arduino #66: Requesting 1+x bytes from I²C-Slave

odo2063, 2018-06-23:
Hi! (v1.1.0)

I got an issue with a Sensirion SDP610 device.

It sends 3 bytes of awnser, but XMC2go sends ACK(0x0) after first byte and closes the bus.(used https://github.com/daveansell/Sensirion_SDP6x_pressure_sens)

Checking against PocketCHIP, the ACK(0x0) is send after first 2 Bytes and after the third ACK(0x1).

techpaul, 2018-06-23:
Reading 3 bytes from a slave and third data byte with ACK = 1 (high/open drain) is STANDARD I2C protocol for a master to read a block of data from a slave this is to signify last byte of reading to the slave.

See https://www.nxp.com/docs/en/user-guide/UM10204.pdf

I think you have another issue

techpaul, 2018-06-23:
Quick look at **their** code suggests **they** have an issue in **SPSP6x.cpp** as endTransmission is actually when the data is sent (not on each Wire command, so may work with some boards but not others the delay to get a reading is in the wrong place.

Change from
```
    Wire.write(command);          //send the pointer location
    delay(10);
    Wire.endTransmission();                 //end
```
To
```
    Wire.write(command);          // BUFFER the pointer location (data)
    Wire.endTransmission();       // end (actually send the I2C write command and data)
    delay(10);                    // Wait foir sensor to take a reading
```
No doubt on that chip requesting a register address wakes up the device and gets it to take a reading to be available on next read, but needs a delay for that to happen.

odo2063, 2018-06-24:
tried that, doesn't change the behavior.

since the sensor does a masterhold when triggerd to send data i changed the code to
`Wire.beginTransmission(eSDP6xAddress);  //begin
delay(10);
Wire.write(command);
Wire.endTransmission();
Wire.requestFrom(eSDP6xAddress, 3);
delay(100);
data[0] = Wire.read();
...`

After recieving the "give Data"(i guess wire.request...) the sensor holds the bus for 21.7ms, then it sends the first byte gets an ACK(0x0), and then the clock stays low and the data high...i have absolutly no idea what I(?)'m doing wrong...

odo2063, 2018-06-24:
might have found something...
[image]

techpaul, 2018-06-24:
So you have this device

http://uk.farnell.com/sensirion/sdp610/sensor-pressure-i2c-500pa/dp/1696081?mckv=sRtccEcFl_dc|pcrid|99278745128|kword|+sensirion%20+sdp610|match|b|plid||slid||product||pgrid|5281560728|ptaid|kwd-85102462568|&CMP=KNC-GUK-GEN-SKU-MDC&gclid=EAIaIQobChMI8K7l4dbs2wIVrArTCh047AWzEAAYAiAAEgLoz_D_BwE

and the part datasheet (always a good thing to read)
http://www.farnell.com/datasheets/1720196.pdf?_ga=2.44539751.2116900072.1529856328-1760954961.1472845549&_gac=1.52620060.1529856328.EAIaIQobChMI8K7l4dbs2wIVrArTCh047AWzEAAYAiAAEgLoz_D_BwE

Putting delays between Wire.write and Wire.endTransmission or Wire.requestFrom and Wire.read
Will make NO differenece to how it talks to the device only stop the programme

Data sheet says 

> Each  individual  measurement  is  triggered  by  a  separate read operation.  Note that two transfer sequences are needed to perform a measurement.  First  write  command  byte  hF1  (trigger measurement)  to  the  sensor,  and  then  execute  a  read operation to trigger the measurement and retrieve the flow or differential pressure information.  

>On receipt of a header with R/_W=1, the sensor generates the  Hold  Master  condition  on  the  bus  until  the  first measurement  is  completed  (see  Section 3.3  for  timing). After  the  Hold  Master  condition  is  released,  the  master can read the result as two consecutive bytes. A CRC byte 
follows if the master continues clocking the SCL line after the  second  result  byte.  The  sensor  checks  whether  the master sends an acknowledge after each byte and aborts the transmission if it does not.

The device uses a SF04 chip which is the actuals details you need for how to talk to it. It appears to be clock stretching at point of Addressing the device not acknowledging it which WILL mess up LOTS of controllers.

I suspect you need to check the external library and put lengthy delays between Wire.endTransmission of correct sequence to trigger measurement and the Wire.requestFrom

odo2063, 2018-06-25:
starting the measurement works fine...there is no pressure difference atm so values are around zero(FFFF to 0001)...i'll test that with an "real" arduino today and report back...

odo2063, 2018-06-25:
i'm done...i just dont know whats wrong...arduino uno is missing the high too....
[image]

techpaul, 2018-06-25:
What is needed is the data sheet or programming manual for the SFO4 chip inside the sensor, which I have not been able to locate the datasheet for the product does not have enough details.

This is how the device talks to the I2C bus or what it is doing strangely

odo2063, 2018-06-25:
could it be that the xmc locks down the clock after first byte?

techpaul, 2018-06-25:
As the original (not amended code) does not work with an Uno as it is supposed to there is another issue most likely with the library.

Or a faulty device or some other aspect of not using correctly I do not have one to test or 3V board to drive it or know if I have a board capable of driving it to hand, and I do not have spare time to set one up at the moment. ( I do not work for Infineon)

There may also be an issue with clock stretching on XMC, but I would need to check the data sheets

odo2063, 2018-06-25:
uno is missing the "high" but as you can see it gets the 3 Bytes...so uno works...

odo2063, 2018-06-26:
So maybe an Infineon member could have a look at this?

techpaul, 2018-06-26:
So anyone can helkp you the following details and tests would be helpful

1.  Ask your supplier of the SDSP610 for a FULL manual for the I2C command set for the SF04 chip inside the unit. Most available online do not cover the chip and specific commands like Soft Reset, Trigger Reading, Read Temperature, Read Supply Voltage. This will help anyone determine the full requirements for the device. (Please note Sensirion state this product is not recommended for new designs on their web site)
2.  Take a series of scope screenshots showing addressing and reading,  (your last screenshots the write writing on green for the Bus signals was not easilt readable) and specifying what the V/div setting is for each trace for the 
    - Start of sequence addressing the device
    -  Gap between addressing and start of data being read
    -  The data sequence

     For XMC2GO and Uno.  Place these files in a zip file to post here

    Your screenshots suggest your I2C voltages are around 4V or 2V on device and I2C bus that should be tied to 3.3V (nominal)
3.  Capture the return codes from (assigned to variable and Serial.print them)
    - Wire.write
    - Wire.endTransmission
    -  Wire.requestFrom

     For XMC2GO and Uno.  List the details here
    These will assits in locating what the problem is for your device

I doubt anyone has that device to hand or will do, there are many thousands of I2C devices out there so their datasheet, timing shots and I2C spec is the way to detrmine what is going wrong.

techpaul, 2018-06-26:
I have sent a request to Sensirion for the chip SF04 I2C specification and command set

Let's see if anything comes back

techpaul, 2018-06-27:
Having had a LITTLE information from Sensirion

I suspect you problem is software timeouts, as Arduino (uno/Mega/etc) work on principle  of waiting for ever for events to happen blocking out anything else happening, the code probably works but not suitable for real applications with other sensors or actions to take. XMC Wire library does have a timout in numbers of loops not actual time.

I suggest you set the SDP610 to disable Hold master as explained in Application Notes

[SF_AN_Flow_DP_I2C_Pulsed_Mode_With_Hold-Master_Disabled_V2_D2.pdf](https://github.com/Infineon/XMC-for-Arduino/files/2140819/SF_AN_Flow_DP_I2C_Pulsed_Mode_With_Hold-Master_Disabled_V2_D2.pdf)
[SF_AN_Flow_DP_I2C_Continuous_Mode_With_Hold-Master_Disabled_V1_D2.pdf](https://github.com/Infineon/XMC-for-Arduino/files/2140820/SF_AN_Flow_DP_I2C_Continuous_Mode_With_Hold-Master_Disabled_V1_D2.pdf)

You may also find reading the temperature and supply voltage interesting as explained in

[ReadtemperatureAndSupplyVoltage.pdf](https://github.com/Infineon/XMC-for-Arduino/files/2140823/ReadtemperatureAndSupplyVoltage.pdf)

odo2063, 2018-07-01:
Hi!

I had some other work to do last week...

For the Inputtimings, it's all just written on the bus with no differences (Address/w; 0xF1;Adress/r;waiting for device). The Voltage of all systems is 3.3V, since it's digital i thought it's not part of the Problem.

I try disabling Clockstretching next week. but since there is no waittime in the XMC code i think there is a point to look at...

And of Course thanks for the Help Paul!!!!

techpaul, 2018-07-01:
The real problem I believe is the excessive time between asking for a reading then when trying to get a reading the address and 3 data bytes is timing out on XMC version of Wire.

The problem with voltages is often I see people using many boards including Arduino Uno, and power the device to 3V3 then add external pull-up resistors to 5V when they should be 3V3 and cause all sorts of straneg effects. 

Await your testing results.

odo2063, 2018-07-01:
wiring to 5V would destroy the SDP ;-P...

techpaul, 2018-07-01:
Depending on pull-up value having seen people use values of 10k and more.

Some people do silly things. Some people request silly things

My best example was a circuit that someone wanted to be laid out, they reckoned would work I could see that their latching hall effect switch with open-drain output, being used as a power switch to circuit would supply their 3V3 rail as open-circuit or GND !!!

Dont even start me on the people who cant wire up a simple LED circuit and using wir ended components manage to short out their current limiting resistor and wonder why their board is dead.

odo2063, 2018-07-02:
I think i'm actually quite good with hardware, it's actually the programming, especially µC, where I have a lack of expirience...;-)

..building up the setup right now...

odo2063, 2018-07-02:
THAT IS SOOOOOO STRANGE!!! Reading the setupregisters actually works. it uses the same function as getting the measurementvalues, which (still) doesn't work...

odo2063, 2018-07-02:
there is a delay of ~5ms for getting the measurement value after the readrequest which isn't there for the setupregisters...maybe someone should work that into the wirelib for XMC, allowing delays for recieving data....
[image]
Want to read AdvUserReg...
[image]
...here u go with no wait...
[image]
..I want to set Clockstretching off...and please take a measurement....
[image]
...now give me value...
[image]
...oh, please wait...
[image]
...here u can have it...no dont want it anymore...
[image]

techpaul, 2018-07-02:
There are MANY controllers and protocols where holding things for 5ms on I2C will mess things up

Try doing
     request from 0 bytes no stop
     wait 10ms 
     request from 3 bytes with stop

Their polling method whilst technically correct (while meauring the address does not exist) is the weirdet of many devices I have seen. Most are send a command to take measurement or do something, you
wait and read later any data you read may well be an old reading, poll a status bit or monitor an open-drain interrupt pin.

Wire is due for overhaul, I am doing major overhaul on RTC and associated examples at the moment that requires quite a bit of testing, next on list is Wire.

If you can edit the Wire.h file

Change WIRE_COMMUNICATION_TIMEOUT to 50000 or 500000 for now to greatly extend timeout

odo2063, 2018-07-08:
So i'm back from a workful week...i have set WIRE_COMMUNICATION_TIMEOUT to 65000u aaaaaannnnnddd......IT FINALLY WORKS!!!

THANK YOU VERY VERY VERY MUCH!!!

techpaul, 2018-07-08:
Glad it is working when I get to do thorough review of Wire (and lower libraries) I will remove timeouts as this is not in ACR style Arduino along with other issues to get better comptability.

Right try to finish RTC/Sleep/Hibernate

odo2063, 2018-07-10:
Removing timeouts completly might become a problem with faulty devices...so maybe a hugh timeout(1s?) is applicatable...

techpaul, 2018-07-10:
Standard AVR based Arduino cores and libraries have NO timeouts and few functions/libraries have even return codes for timeouts. A long clock stretch could be viewed by some as stuck bus or faulty device, but no way to recover without some form of additional Reset signal or way to power cycle the device. Standard I2C bus reset method is to toggle the clock line at least 9 cycles and send a stop condition, something you cannot do with clock stretching on.

Most notably 'standard' Arduino functions  are basically
1.  If waiting to send output (e.g. Serial.print/write), sit forever waiting for buffer or device to become ready.
2.  If interrupt data coming in and no buffer space discard data, and in some cases like HardrwareSerial receive on one error discard data with NO status, count or flag anywhere
3. Unless we can easily do an error code at time of running a function, discard error

In one case on a non-AVR board in a USB interrupt function had the method for dealing with USB device disconnect with the comment

`
// TO DO `Find a way of signalling this to higher level
`
In Arduino there is little in the way of communications/protocol stacks, there is no common error method or task scheduling everything is assumed linear coding. With very little scope for project driven configs of core/library.