Infineon/XMC-for-Arduino #68: XMC 2GO H-Bridge

AttilaRigo, 2018-06-26:
Hi All,
I tried more example code especially  IFX9201 motor control but it is not working.
Does not work at all or do I have the problem?

Thank You!

Attila

mhoechner, 2018-06-27:
Hi Attila

this commit:
[https://github.com/Infineon/XMC-for-Arduino/commit/2516f529d1fb102efc5b78abc8d496081d1f2915](url)
should solve issues with compiling "MotorControl.ino".

Let us know if you see further issues.

BR
Moritz

techpaul, 2018-06-27:
If link does not work ( as for me)

Click at top 'Code, then select 'commits' and look for commit titled 'Updated IFX9201 example' should be near the top of list

AttilaRigo, 2018-06-27:
Thanks for the quick response but it still does not work. 
I'm trying to fix the SPI collision.

AttilaRigo, 2018-06-27:
Thanks for the quick answers. unfortunately it still does not work. I'm trying to fix the SPI collision.

sherylll, 2018-09-13:
@AttilaRigo Hi for me this example compiles fine. Could you give more info about the fail? What do you mean by SPI collision exactly?

mhollfelder, 2018-11-05:
Do you still need help?
We are happy to help if you give us more information about your specific issue.

Thanks a lot and best regards,

Manuel

dizi1, 2019-06-19:
Hi All,

I could only use the forwards, backwards and stop 
i cant control the frequency and the pwm.....

For ex:  IFX9201_HBridge.forwards(50);    doesn't change the PWM or any number.
Also: IFX9201_HBridge.setPWMFreqency(5000);   - the number doesn't affect the frequency 

BTW, there is a diference between "it compiles for me" to a "the code works on the board itself".. 

 any help is appreciated. Thanks.

techpaul, 2019-06-19:
Without the begin statement you are using it is difficult to see if you are using SPI or PWM mode

Also to see if the PWM pin parameter has been set to a VALID pin for hardware PWM

 IFX9201_HBridge.setPWMFreqency(5000);  Actually calls setAnalogWriteFrequency( )

This function WILL error and not set set a frequency if the PIN for PWM is invalid for that board or frequency too large. This will be a SILENT error as the IFX routines do NOT pass this error back

 IFX9201_HBridge.forwards(50 actually calls analogWrite which can silently ERROR and not change if the pin is set wrong for that board.

dizi1, 2019-06-20:
Than you for your prompt reply.

Yes i'm using the PWM method since it's the only one that works. but only forward / backwards / stop - frequency and PWM doesn't work.

I have tried to change the PWN pin 10 to the physical number and also tried pin 8 for pwm0 vs pin 10 (default) of pwm1 - it didn't work.

Are the default pin numbers for hbridge2go are :
DIR = 6
PWM = 10
DIS = 11 

(these are arduino IDE pin numbers as per the chart here : https://github.com/Infineon/XMC-for-Arduino/wiki/XMC1100-H%E2%80%90Bridge-2Go )

Could you please check the library code and physically test them on a H-bridge2go board on Arduino IDE? 

Thanks

techpaul, 2019-06-20:
1.  I am ***NOT*** staff paid to sit here all day supporting this software. I am a contributor to this project who understands many of the levels of software.

2.  I asked what is ***YOUR*** begin statement

3.  you I can ONLY assume are using your OWN software which none of us can be psychic about what your code is doing

4. Does the example work correctly ?

5. I do not have everypiece of hardware here to stop my paying work, to assist your problem.

6. Your call IFX9201_HBridge.forwards(50) if the first time called will ***NOT*** show any chnage as it is setting the deuty cycle to the software DEFAULT duty cycle.

7. There could be MANY errors, some of which could be your misconceptions of how this works or bugs in your code, YES and it could be there is a bug in the code, but you have NOT sent anything to show this currently what you have sent is hand waving.

You actually sound like a student in your approach to this issue.

dizi1, 2019-06-20:
1. i APPRECIATE your time.

2. I only tested the EXAMPLE code not my own code, the BEGIN statement was PWM  - the frequency & PWM do not work no matter how many times i call the procedure.

3. dont assume.. i'm saying the EXAMPLE code DOESNT WORK - who ever contributed PLEASE test the damn code first on a H-Bridge2go board its only $13.. 

4. the EXAMPLE code you guys provide for arduino IDE only does FORWARD, BACKWARDS, STOP/END
The frequency & PWM settings using the EXAMPLE code doesn't work.

5. I just feel its weird contributors are writing code for hardware WITHOUT testing said code on that hardware. I have 2 boards i can work with you to fix the example code.  Happy to be a contributor too.

6. Ok, so i call it again? it doesn't work.   I asked you, are you sure with the default PIN numbers you set on the EXAMPLE code are correct - please check and post a reply.

7. I tested the EXAMPLE code you guys provide for arduino IDE - NOT my code.
Its fairly simple - BEGIN, try forwards, backwards, stop..   setting the speed (PWM) and Freq doesnt make any changes psychically.

I appreciate the time you're taking for this project - i have the utmost respect to all contributors and i'm willing to help - by working with you to test different codes on my H-bridge2go to make this project a working success.

Thank you.

techpaul, 2019-06-20:
I do ***NOT*** work for Infineon

I did ***NOT*** write the Hbridge code

I have no more spare time for this

dizi1, 2019-06-20:
Ok. Thanks for all your contribution & time.

If i can fix the EXAMPLE code i will post a solution. As it is right now it's a NON WORKING project.

dizi1, 2019-09-05:
Really sad to see this unsolved yet.
Infineon? Perhaps assign your coders to fix the code and make it WORK?.....

9Volts9er, 2019-09-09:
Hi,

I've found some problems with the H-Bridge2Go implementation. 
Please check if pull request #102 fixes your issues. 

Please also note that the default duty cycle is already 50%, so calling IFX9201_HBridge.forwards(50); won't change anything (even if you call it again). If you want to change it, provide the desired duty-cycle as parameter. E.g. IFX9201_HBridge.forwards(100); will set the duty cycle to 100%. If you call IFX9201_HBridge.forwards(); later it will be reset to the Default - which means 50%

I'm sorry if you expected the issue to be solved immediately. But being impatient is definitely not helping anybody. I can imagine that there are more important issues than this one. Even my pull request is based on the coincidence that I recently got in touch with the hardware this weekend - it's NOT a reaction to your latest message.

Best Regards,
Severin

closed 2019-09-09 by pandayswarnam

techpaul, 2019-09-09:
The comment about Duty Cycle and default values  was told in June

There is  minor bug in IFX9201.cpp
~~~
uint8_t IFX9201::DutyCycleToanlogWriteValue(uint8_t duty_cycle)
{
	uint8_t ret = ( (duty_cycle >= 100u) ? 255u : (uint8_t)((double)duty_cycle * 2.55) );
	return ret;
~~~

The value used shoulkd be uint16_t to match AnalogWrite function

The 255 values should be replaced with values determined by AnalogWriteResolution and NOT assume it is 8 bit as the max value could be 255, 1023, 4095, 6553, or other vallues. Sending 8 bit values when the resolution has changed to 12 bit will give nearly no response never getting above 16.67%

This also needs either class members or helper functions in wiring_analog.c to get what the current resolution is.

Been on back burner for me and not had a board to test on as well does not help

dizi1, 2019-09-09:
> Hi,
>
> I've found some problems with the H-Bridge2Go implementation.
> Please check if pull request #102 fixes your issues.
>
> Please also note that the default duty cycle is already 50%, so calling [...]

No it's not. You guys are a JOKE writing code to a $13 board you dont have to TEST you code!

You code is full of syntax error, library wont load , no example for first time user and after changing your code manually to a previous library version - its still not working.

ArkajyotiChatterjee, 2019-09-09:
Stop commenting and polluting the issue page @dizi1 .
Next time you will be blocked.

dizi1, 2020-08-14:
Hello,

I wonder if it's possible to read if a load is connected to the output pins?
Also, ares the latest firmware versions somehow reduce the current output? I use a 5v 2A power supply and on ver 1.2 it used to push high current up to ¬1.5A now less than 1A

Thank you

techpaul, 2020-08-14:
Not had a look at how to check pins connected

Which mode are you using SPI mode is ON/OFF and PWM mode gives speed control

I suspect that the duty cycle being used in PWM mode is lower than used to be or other issues, like lighter physical/mechanical load on motor

Are you getting same speed?

techpaul, 2020-08-14:
This is a new problem and needs to be a new issue.

Not had a look at how to check pins connected

Which mode are you using SPI mode is ON/OFF and PWM mode gives speed control

I suspect that the duty cycle being used in PWM mode is lower than used to be or other issues, like lighter physical/mechanical load on motor

Are you getting same speed?

dizi1, 2020-08-14:
Thank you for prompt reply.

I use PWN mode only. (SPI OFF)
I control speed with 0-100% 
Can you elaborate what "physical/mechanical load on motor" means?
I thought a load draws what current it needs. 
on version 1.2 same load drew upto 1.8A now below 0,8A. 

Are there any manufacturer flags to indicate a load is connected or not that is not supported yet in arduino ide ? 
Knowing if a load isnt connected, i then could trigger some kind of alert like a led blinking. Which is easy i just need which/how to read the flag.
 
Thanks