Infineon/XMC-for-Arduino #73: Troubles with I2C Communication between Boot Kit 1100 and XMC2Go

matthiasbuhl, 2018-07-01:
Hi,

i wrote two sketches (see attachments) that should demonstrate a "plug-and-play" feature for i2c. While coding the example, I used an Arduino UNO and an XMC2Go. The Node_Demo.ino was flashed to the Arduino UNO, the Device_Demo was flashed to the XMC2Go. The communication between the two devices can be seen in the attached serial monitor screenshot. I then swapped out the Arduino UNO for a Boot Kit 1100. The code wasn't altered in any way. Still, the output that can be observed in the serial monitor changed. The communication always gets stuck early on. Could somebody please review my code and let me know, what triggers this different reaction of the Boot Kit compared to the Arduino. 

[image]
[image]

[I2C_PlugNPlay.zip](https://github.com/Infineon/XMC-for-Arduino/files/2153161/I2C_PlugNPlay.zip)

techpaul, 2018-07-01:
Unfortunately in your node.ino (your master)
`  
Wire.begin(8);        // join i2c bus (address optional for master)
`

Change code to 
`  
Wire.begin();        // join i2c bus (address blank for master)
`

Currently the address is NOT optional for XMC it currently assumes this is Slave mode configuration and sets the I2C configuration differently.

This is scheduled for review when I have finished a major review of RTC/DeviceControl issues later this week, depending on other paying work interferring.

Try that and see if it works

mhollfelder, 2018-07-20:
We had also a look on it and could help meanwhile directly.
Thank you very much for reporting this issue here!

@techpaul: thank you very much for your feedback and contribution here.

Best regards,

Manuel

closed 2018-07-20 by mhollfelder