Infineon/XMC-for-Arduino #74: issue with Ardino 1.6.5 XMC1 Family and SPI and I2C C and CPP compatibility issue compilation error

jknapow, 2018-07-03:
initialization of XMC_GPIO_CONFIG_t with XMC1 Family
around line 130ish in HW_SPI.cpp SPIClass::end()

	if((XMC_SPI_config->channel->CCR & USIC_CH_CCR_MODE_Msk) == XMC_USIC_CH_OPERATING_MODE_SPI)
	{
	    XMC_GPIO_CONFIG_t default_input_port_config = {
            .mode = XMC_GPIO_MODE_INPUT_TRISTATE,
#if UC_FAMILY == XMC1
						.input_hysteresis = XMC_GPIO_INPUT_HYSTERESIS_STANDARD,
#endif
						.output_level = XMC_GPIO_OUTPUT_LEVEL_HIGH
};
The conditional statements are executed for XMC1 processors/boards, but that puts the initialization out of order with the typedef. order needs to be mode then output_level the input_hysteresis

quick fix for XMC1 boards to remove condition statement and reorder, I am sure there is a better fix that keeps it functional for all target boards.

techpaul, 2018-07-03:
Yes there are a few ways to do this in a more maintainable way.

I will try and look at these files later tonight and see if a quick fix can be issued as soon as possible.

Getting more maintainable method is on my list when I get to looking over Wire library issues all together.

jknapow, 2018-07-03:
Thanks for quick reply...over night I thought of different fix which would likely be better, haven't tried and need to review in details that it wouldn't break it for other XMC families (XMC4) would be to just reorder them in Typedef, then the conditional formatting still works as written in both the spi and wire. i might try it tomorrow.

techpaul, 2018-07-03:
See latest Pull request #75 to get files for libraries easiest to either get copy of files from pull request or copy the libraries folders for src under SPI and Wire

techpaul, 2018-07-19:
This has now been fully incorporated into release V1.1.2

This issue probably needs to be closed

jknapow, 2018-07-19:
New to forums do I go on and close it?

techpaul, 2018-07-19:
Always best if original person logs in and closes issue showing resolved/workaround

jknapow, 2018-07-20:
Resolved in  Release v1.1.2

closed 2018-07-20 by jknapow