Infineon/XMC-for-Arduino #79: Program erased on power down?

dwaq, 2018-07-30:
I have a 3D magnetic sensor to go kit (TLE493D A2B6) and can program it fine using the "XMC1100  XMC2Go" board. Once I remove power, and reapply power, it seems that the memory is erased. At least, the previously programmed sketch does not run. How can I permanently program my sketch into the memory so it runs on power up? (I'm comfortable using the J-Link Commander if that will solve my issue). Thanks in advance!

dwaq, 2018-07-30:
Nevermind, it had to do with issues in the [TLE493D-W2B6](https://github.com/Infineon/TLE493D-W2B6-3DMagnetic-Sensor) library. My sketch started correctly but the sensor was giving bad readings so it didn't appear that it was running.

I first noticed I was wrong when I did a blinky sketch and it worked after power cycling (just a tip for anyone that runs across this).

closed 2018-07-30 by dwaq

mhanuel26, 2018-08-16:
Hi @dwaq , How do you solve the issue? 

I had the same problem, intermittent works, sometimes and some not.
Most of the time I have to re-plug the device USB cable to make it work.

dwaq, 2018-08-16:
@mhanuel26 what board specifically are you using? I probably can't help past what my last comment states. Are you sure your code is good? Maybe just try the Blinky sketch and see if you can recreate the issue.

mhanuel26, 2018-08-16:
@dwaq perhaps I didn't explain myself, my example code is

```
#include <Tle493d_w2b6.h>

Tle493d_w2b6 Tle493dMagnetic3DSensor = Tle493d_w2b6();
void setup() {
  Serial.begin(9600);
  while (!Serial);
  Tle493dMagnetic3DSensor.begin();
  Tle493dMagnetic3DSensor.enableTemp();
}

void loop() {
  Tle493dMagnetic3DSensor.updateData();

  float angle = map(atan2(Tle493dMagnetic3DSensor.getY(), Tle493dMagnetic3DSensor.getX()) * (180 / M_PI), -180.00, 180.00, 0.00, 360.00);

  Serial.print(Tle493dMagnetic3DSensor.getX());
  Serial.print(" ; ");
  Serial.print(Tle493dMagnetic3DSensor.getY());
  Serial.print(" ; ");
  Serial.println(Tle493dMagnetic3DSensor.getZ());
  Serial.print("angle = ");
  Serial.println(angle);
  
  
  delay(1000);
}
```
The code works, I can see the debug console, actually when it works I see the angle change correctly. The problem is that most of the time I have to re connect the device, the USB port, because it seems the I2C freezes, and gives fixed values, 

foe example I connect the device and open console, this is what is shown

[image]

Then I close the console, unplug and plug the USB, to see this

[image]

I have seen some people complaining about the I2C, which I suspect is the problem here, but sisnce you comment that you solve the issue, that why I asked.

I am using the 3d Magnetic  Sensor 2 Go. 

On my other XMC 2Go board it actually never works the I2C. I am investigating the issue which might be related to pull up values or something.

techpaul, 2018-08-17:
Firstly as the same type of prompts are output then your program is **NOT** being erased If it was being erased NOTHING would be happening.

Secondly wrong data suggests operating the sensor incorrectly. When you flash the programme into the XMC processor the board has been powered up for **MANY** seconds and no doubt stabilised before you ran the program. When you power cycle the board, it has not had time to stabilise or warm up.

When the system starts up it takes about 10 ms to get to your setup() function, many modern complex sensors and peripheral devices either have inbuilt micros running at slower speed or have parts that rely on aspects of **PHYSICS** to settle, warm up, stabilise **before** you access the device or take readings. This will be in the datasheet for the device under Operation, programming the device or similar.

Many older devices also have **LONG** delays (in computer terms), a classic being LCD character displays based on HD44780, have to have many different types of delay as the Return Home and Clear Display command takes about 1.5ms before display is available to be written to again. Most other commands take up to 40 micro seconds on most displays. Can be longer if that display uses slower internal clock than most. 

This problem of **NOT waiting appropriate time for external device** is mentioned in issue #64 where the person asking questions put in their setup() function
~~~
delay( 500);
~~~
Then by experimentation managed to find a much smaller value for the delay that worked as well.

The issue is *unlikely* to be the board or I2C software but not following requirements for external device.
Reading the datasheet for the device on how to operate is often essential for complex devices and to know how it performs,no magic software library will auto-calibrate your setups for many types of sensors.

I deal with many complex projects as part of my other paying work, so I am used to reading datasheets and limitations of devices.

techpaul, 2018-08-17:
Personally this issue was well closed before if you have problems with other interfacing start a new issue appropriately labelled.

mhanuel26, 2018-08-17:
Hi @techpaul 
I didn't start new thread since this one originally was about the TLE493D sensor, the same I am experiencing an issue. and this one was very fresh...
It was not the same console because disconnecting the USB hangs the serial console already opened.
I close the console, disconnect and connect again USB and the the sensor readings work.
The strange thing is that after programming I disconnect the USB device, then connect again open the console to find wrong sensor initialization  which is very strange because the second time it works, but the first time after a "cold state" don't work. 
I do not open the console immediately after programming at all, because I know about this warm up things can break initialization code.

techpaul, 2018-08-17:
I will say these things one more time

1.  Start a new thread as your problem is "Problems with TLE493D on XMC...." It is NOT "Programme being erased on Power down".
2.  Try things adding as suggested a delay(500) to your setup() function to cope with *'cold stret'*
3.  Read the datasheet for the  TLE493D and how to operate it
4. Disconnecting USB while a programme is accessing especially on Windows/Mac/linux WILL cause most programs to hang Disconnecting the USB and plugging in again is adding RANDOM events to the board in question unless you are powering from a different source. I can only assume you are powering from USB as you have not said otherwise.

At the moment you are not giving evidence but general hand waving

Unless you do at least the first 3 items I will not bother replying, others might.

techpaul, 2018-08-17:
Further more raise the issue in the library for that sensor [TLE493D-W2B6](https://github.com/Infineon/TLE493D-3DMagnetic-Sensor)