Infineon/XMC-for-Arduino #82: Incomplete update to gcc V5.4

techpaul, 2018-08-06:
As reported in pull request #80 new and delete no longer work

These worked in XMC-for-Arduino V1.1.0 as evidenced by example in Device Control XMC/HeapMemoryMeasurement

Since V1.1.1 and update to gcc V5.4 the nanolib support has broken or removed the new functionality

I currently am busy on other work and other library updates to schedule this in. If someone else can find the differences before i get time, it would be appreciated.

sherylll, 2018-09-13:
Current workaround:

```
// work around to use new operator
extern "C" void *_sbrk(int incr);
void dummy_sbrk_caller() __attribute__((__used__));
void dummy_sbrk_caller()
{
  _sbrk(0);
} 
```
in main.cpp

techpaul, 2020-06-29:
We have workaround and issue resolved

closed 2020-06-29 by techpaul