Infineon/XMC-for-Arduino #85: Problem with I2C on XMC device and TLE493D sensor

mhanuel26, 2018-08-17:
@techpaul 
This is an issue I start reporting at #79 but since it was not the correct place I am starting this new thread with more proof of an issue with XMC device.

The problem is basically that sometimes I plug the board using USB connector and open the console and it works the example code as shown in #79 , but other times (most of the times) I got incorrect sensor readings. 

First of all, I am not reprogramming the device in between successful and failure of TLE493D sensor initialization. 

Ii am not 100% where the issue is, but I think is more related to XMC device than the sensor library.

When the device starts correctly I can see the initial trace of I2C signal as below

[image]
Everything works there, the sensor respond all the time as I can see in the debug console.
[image]

But when the sensor fails to read/configure correctly, the I2C trace becomes

[image]

It took my attention the clock signal.

Notice that the debug console of failure

[image]

actually don't make any new I2C trace, is actually like the I2C command stop working. 
I know the MCU is working because the debug console refresh every 1 seconds as per the delay(1000)

The debug code is

```
#include <Tle493d.h>
#include <Tle493d_w2b6.h>

Tle493d_w2b6 Tle493dMagnetic3DSensor = Tle493d_w2b6(Tle493d::MASTERCONTROLLEDMODE, Tle493d::TLE493D_A0);
void setup() {
  delay(500);
  Serial.begin(9600);
  while (!Serial);
  Tle493dMagnetic3DSensor.begin();
  Tle493dMagnetic3DSensor.enableTemp();
}

void loop() {
  Tle493dMagnetic3DSensor.updateData();

  float angle = map(atan2(Tle493dMagnetic3DSensor.getY(), Tle493dMagnetic3DSensor.getX()) * (180 / M_PI), -180.00, 180.00, 0.00, 360.00);

  Serial.print(Tle493dMagnetic3DSensor.getX());
  Serial.print(" ; ");
  Serial.print(Tle493dMagnetic3DSensor.getY());
  Serial.print(" ; ");
  Serial.println(Tle493dMagnetic3DSensor.getZ());
  Serial.print("angle = ");
  Serial.println(angle);
  
  
  delay(1000);
}

```
And the board is the Magnetic Sensor 2Go board  or the XMC MS2Go board.

The board is powering from USB. 
As you can see I have added the 500 ms delay in Setup.

Is this is OK for you @techpaul to consider I have found something here and would you kindly let me know your thoughts please.

mhanuel26, 2018-08-18:
Powering the device without USB never works. The I2C trace on my scope is always missing. I know the device is working because I have setup the PWM pin to change every second.

```
  analogWrite( 8, pwm );      // PWM4
  pwm += 20;
  if(pwm > 255){
    pwm = 0;
  }
```

I cannot see the I2C clock but I see the PWM change.

Just plugin the device thru USB without reprogramming works. 

I certainly know this is not reliable thing. Can't even make it work standalone.

Do you know if there is a way I can debug step by step the code?

mhanuel26, 2018-08-18:
I have notice a difference in the I2C pins, specifically the clock when it works and when Not.
I have tested several times to confirm. 
When it works the I2C initialization is something like this
[image]

When it does Not work
[image]

Notice the first clock "glitch", is always present when it does not work.

But even when it does not work the device seems to replay the initialization 

[image]

The I2C commands stop after this, i.e. the updateData() function does not produce any signal on the I2C bus. Perhaps some software error detection that prevents it due to some initial error.

sherylll, 2018-08-18:
> Powering the device without USB never works.

Isn't this expected because you have `Serial` in your code?

> Do you know if there is a way I can debug step by step the code?

There are different ways. You can use VSCode+PlatformIO which comes with a debugger, and it works without any setup. Or you can try out Cortex-Debug, also a VSCode extension; however it requires some extra configuration. You can also use Eclipse+Sloeber.

mhanuel26, 2018-08-18:
Hi @sherylll ,

No, I have the code without debug serial for this test such as

```
#include <Tle493d.h>
#include <Tle493d_w2b6.h>

//#define DEBUG
int pwm = 0;

Tle493d_w2b6 Tle493dMagnetic3DSensor = Tle493d_w2b6(Tle493d::MASTERCONTROLLEDMODE, Tle493d::TLE493D_A0);
void setup() {
  delay(500);
#ifdef DEBUG
  Serial.begin(9600);
  while (!Serial);
#endif
  Tle493dMagnetic3DSensor.begin();
  Tle493dMagnetic3DSensor.enableTemp();
}

void loop() {
  int err = Tle493dMagnetic3DSensor.updateData();
  if (err != TLE493D_NO_ERROR) {
    Serial.print("Error on Update: ");
    Serial.println(err);
  } else {
    float X = Tle493dMagnetic3DSensor.getX();
    float Y = Tle493dMagnetic3DSensor.getY();
    float Z = Tle493dMagnetic3DSensor.getZ();
    float angle = map(atan2(Y, X) * (180 / M_PI), -180.00, 180.00, 0.00, 360.00);

#ifdef DEBUG
    Serial.print(X);
    Serial.print(" ; ");
    Serial.print(Y);
    Serial.print(" ; ");
    Serial.println(Z);
    Serial.print("angle = ");
    Serial.println(angle);
#endif
  }
  delay(1000);

  analogWrite( 8, pwm );      // PWM4
  pwm += 20;
  if (pwm > 255) {
    pwm = 0;
  }

}
```
I will try the the debugger options you mention,

techpaul, 2018-08-20:
@sherylll you said

>>    Powering the device without USB never works.
>
> Isn't this expected because you have Serial in your code?

Depends on the board and configuration, in general for USB devices depends on the capabilities of the USB device as to how powered, and for something that may be picky about Power rails. I would choose NOT USB powered if possible as connect and disconnect of USB power can do all sorts of strange things to power, without filtering, large capacitors, reset delays, external power switching and soft-ssrts.

Most standard Arduino boards can operate with external power or USB power using external if that is available first, with FET switching of power, I have not looked at the circuit diagram for this board to see what it does on external power and USB power.

techpaul, 2018-08-20:
In this case I think the original poster has found that this is not specific to the XMC board, maybe the library and maybe more the sensor and/or circuit layout. As in the refernce [TLE493D getting stuck (maybe defective)](https://github.com/Infineon/TLE493D-3DMagnetic-Sensor/issues/6)

mhanuel26, 2018-08-22:
Hi @techpaul 
I still think the XMC has some issues or differences, It's not behaving in similar way to ATSAMD21 for example, notice I am comparing two Cortex M0 devices.

For example the following code

```
	mInterface.bus->begin();
	
	mInterface.bus->setClock(100000);
	
	delay(50);
	mInterface.bus->beginTransmission(slaveAddress);
	byte error = mInterface.bus->endTransmission();
	if (error != 0) return;
	
	digitalWrite(4, HIGH); // trigger for scope
	mInterface.bus->beginTransmission(slaveAddress);
	mInterface.bus->write(0x10);
	mInterface.bus->write(0b01010001);
	mInterface.bus->endTransmission();
```

produces the following
[image]
 scope capture (the trigger is for the actual write to address 0x10)

I have introduce the delay before all this, such as 

```
  delay(500);
  Tle493dMagnetic3DSensor.begin();
```

Actually this code works on the ATSAMD21 without this issue.

This was on the MS2Go kit, the XMC 2Go cannot even output the I2C signal with same code.
I know the pins are not dead since I have a manual reset. 

About reset, I have modify the code from @bluetiger9, to not write a High level on SDA, but instead configure the PIN as Input and let the Pull Up do the Job. The code is 

```
void i2c_Start() {
  delayMicroseconds(5);
  pinMode(PIN_SDA, OUTPUT);    // PIN_SDA LOW
  delayMicroseconds(5);
  pinMode(PIN_SCL, OUTPUT);    // PIN_SCL LOW
  delayMicroseconds(5);
}

void i2c_Stop() {
  pinMode(PIN_SDA, OUTPUT);
  delayMicroseconds(3);
  pinMode(PIN_SCL, INPUT);    // PIN_SCL HIGH
  delayMicroseconds(5);
  pinMode(PIN_SDA, INPUT);    // PIN_SDA HIGH
  delayMicroseconds(5);
}

void i2c_Bit(int data) {
  if (data)
    pinMode(PIN_SDA, INPUT);
  else
    pinMode(PIN_SDA, OUTPUT);
  delayMicroseconds(3);
  pinMode(PIN_SCL, INPUT);
  delayMicroseconds(3);
  pinMode(PIN_SCL, OUTPUT);
}

void resetSensor() {
  // 0xFF
  i2c_Start();
  for (int i = 0; i < 8; ++i) {
    i2c_Bit(0x1);
  }
  i2c_Stop();
  delayMicroseconds(20);
  // 0xFF
  i2c_Start();
  for (int i = 0; i < 8; ++i) {
    i2c_Bit(0x1);
  }
  i2c_Stop();
  delayMicroseconds(20);
  // 0x00
  i2c_Start();
  for (int i = 0; i < 8; ++i) {
    i2c_Bit(0x0);
  }
  i2c_Stop();
  delayMicroseconds(20);
  // 0x00
  i2c_Start();
  for (int i = 0; i < 8; ++i) {
    i2c_Bit(0x0);
  }
  i2c_Stop();

  delayMicroseconds(30);
}
```
The code is called from setup such as

```
  pinMode(PIN_SDA, INPUT);    // Let the pull up to rise the line high
  pinMode(PIN_SCL, INPUT);
  digitalWrite(PIN_SDA, LOW);
  digitalWrite(PIN_SCL, LOW);
  digitalWrite(TRIGPIN, HIGH); // trigger for scope
  resetSensor();
```

Then I found the pinMode(PIN_SDA, OUTPUT); instruction is doing something that is wrong.

[image]

If I comment the instruction to repeat every time in the loop such as 

```
void i2c_Bit(int data) {
  if (data)
    pinMode(PIN_SDA, INPUT);
//  else
//    pinMode(PIN_SDA, OUTPUT);
    
  delayMicroseconds(3);
  pinMode(PIN_SCL, INPUT);
  delayMicroseconds(3);
  pinMode(PIN_SCL, OUTPUT);
}
```

and since all is 0xFF or all is 0x00 on the Data, I can just go to my reset function and do

```
  // 0x00
  i2c_Start();
  pinMode(PIN_SDA, OUTPUT);
  for (int i = 0; i < 8; ++i) {
    i2c_Bit(0x0);
  }
  i2c_Stop();
  delayMicroseconds(20);
```

Then the signal becomes

[image]

Also @bluetiger9, if you use the MS2Go with external hardware you should disconnect the resistors of the board or the LED2 (better idea),  otherwise you should continue to turn on the LED2.  Otherwise the signal will be screw up by the LED network I guess, something like this 

[image]

mhanuel26, 2018-08-22:
Also about I2C issue @techpaul ,

The code

```
	digitalWrite(4, HIGH); // trigger for scope
	//write out the configuration register
	
	mInterface.bus->beginTransmission(slaveAddress);
	mInterface.bus->write(0x10);
	mInterface.bus->write(0b01010001);
	mInterface.bus->endTransmission();
	//tle493d::writeOut(&mInterface, 0x10);
	delayMicroseconds(10);
	//write out MOD1 register
	//digitalWrite(4, HIGH); // trigger for scope
	mInterface.bus->beginTransmission(slaveAddress);
	mInterface.bus->write(0x11);
	mInterface.bus->write(0b00110001);
	mInterface.bus->endTransmission();
```
produces the first write to register 0x10 a NACK, the second register write seems OK

[image]

Now if I repeat the first write as a "dummy" workaround, I get

[image]

which correspond to this snippet.
	
  ```
      mInterface.bus->beginTransmission(slaveAddress);
	mInterface.bus->write(0x10);
	mInterface.bus->write(0b01010001);
	mInterface.bus->endTransmission();
	delayMicroseconds(30);
	
	digitalWrite(4, HIGH); // trigger for scope
	//write out the configuration register
	
	mInterface.bus->beginTransmission(slaveAddress);
	mInterface.bus->write(0x10);
	mInterface.bus->write(0b01010001);
	mInterface.bus->endTransmission();
	//tle493d::writeOut(&mInterface, 0x10);
	delayMicroseconds(10);
	//write out MOD1 register
	//digitalWrite(4, HIGH); // trigger for scope
	mInterface.bus->beginTransmission(slaveAddress);
	mInterface.bus->write(0x11);
	mInterface.bus->write(0b00110001);
	mInterface.bus->endTransmission();

```
Can you let me know your thought on this please.

JonathanXu1, 2018-08-23:
Hi it seems that I too am experiencing the same issue. I've tried reinstalling the TLE493D sensor library as well as the XMC boards, but the result is always the same.
[image]
Yet for some reason the Infineon 3D 2GO application works fine. Not really sure how I can fix this problem.