Infineon/XMC-for-Arduino #86: SPI MISO pin conflict with GPIO

mhanuel26, 2018-08-23:
I have a simple sketch that uses MISO pin as GPIO, and digitalRead function is not working, it always read as 0 or LOW level.

Is there any problem with making this pin GPIO  and where is documented.

techpaul, 2018-08-27:
The first question is have you set the pin to input mode with a pinmode() function

Until mode is set on XMC boards the pins are **UNDEFINED**, So will do nothing and likely return 0 all the time

See [this wiki reference](https://github.com/techpaul/XMC-for-Arduino/wiki/Functions-and-Options-Differences)

sherylll, 2018-09-13:
Please search for the reference manual of your board, and go to the section "Port I/O Function Table" to check if analog input is possible with the pin you're trying to configure. 

The pin mapping can be found in the file `arm\variants\XMC1100\config\XMCxx00_xxxx\pin_arduino.h`

closed 2018-09-13 by sherylll