Infineon/XMC-for-Arduino #87: Will this support XMC 4800 relax kit any time soon?

theja2289, 2018-08-27:
Would you be supporting XMC 4800 relax kit? Especially if i want to use the Ethercat app

mhollfelder, 2018-11-05:
The plan is to support the XMC4800 Relax Kit as well as the XMC4800 shares most of its implementations with the XMC4700 Relax Kit.
 
However, the main issues for both microcontrollers are the more complex library implementations like Ethernet as they require more work and testing. EtherCAT is even more demanding as it requires special test setup for the SW engineers - currently, we do not have the bandwidth to manage it although we'd like to do so.

Sorry for not having better news at the moment.

Did you already work on some sketches?
We can, of course, create a branch for an improvement.

Thank you very much and best regards,

Manuel

techpaul, 2018-11-05:
Might I suggest another of your boards that can HOPEFULLY simply be added

XMC4500 Relax [Lite] Kit

Especially any Lite kits without ethernet or SD card, there are many applications which require the larger I/O capability and would love to have the on-board USB as USB to serial on a host.

As the Lite version has NO SD-MMC and NO Ethernet, so may well be easier to support.

I was going to look into it,for Lite version and educational use but other paying work has gone through the roof at the moment.

Please note the packaging for that board actually shows a photo of a 4700 Relax with Ethernet and SD connectors. something to pass up the chain as confusing product description

Plus point for passing up chain as well XMC-for Arduino means we are NOT tied to Windows as many educators cannot get students to afford more Windows systems, some corporates I work for do not want anything developed with their code on the web/Cloud, and do not want systems being used for development with so much phone homes. So using Ubuntu or similar is much preferred.

closed 2022-01-06 by jaenrig-ifx