Infineon/XMC-for-Arduino #88: HardwareSerial::flush() does not wait until data fully sent

cinderblock, 2018-11-06:
I am trying to send out data on a serial bus and to then wait for the bits to have actually been sent out on the bus. The current implementation seems to only wait for the [transmit buffer to be empty](https://github.com/Infineon/XMC-for-Arduino/blob/master/arm/cores/HardwareSerial.cpp#L208) and not actually wait for the full flush like [AVR-Arduino does](https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/HardwareSerial.cpp#L221-L222).

```C++
Serial.print("Hello World!\r\n");
Serial.flush();
while (1) toggleDebug();
```

Results in this:
[image]
You can clearly see the debug output starts toggling before the serial data is actually flushed out to the wire.

I've been playing around with the lower level `channel->TCSR` bits to see if I can't detect the end of shifting but have yet to find a bit that indicates what I need.

Any ideas?

techpaul, 2018-11-06:
I would suggest changing HardwareSerial.cpp to use different calls,  This has **NOT** been tested but is a suggested point to start
Change FROM
~~~
void HardwareSerial::flush( void )
{
while( _tx_buffer->_iHead != _tx_buffer->_iTail ); //wait for transmit data to be sent

while( XMC_USIC_CH_GetTransmitBufferStatus( _XMC_UART_config->channel) == XMC_USIC_CH_TBUF_STATUS_BUSY );
}
~~~

To
~~~
void HardwareSerial::flush( void )
{
while( _tx_buffer->_iHead != _tx_buffer->_iTail ); //wait for transmit data to be sent

while( !( XMC_UART_CH_GetStatusFlag( _XMC_UART_config->channel ) & XMC_UART_CH_STATUS_FLAG_TRANSMITTER_FRAME_FINISHED ) );
}
~~~

Alternatively
~~~
while( !(  _XMC_UART_config->channel->PSR_ASCMode & USIC_CH_PSR_ASCMode_TFF_Msk ) );
~~~

Reminder I have ** NOT** tested this but this is what I have found from datasheets and XMCLib files

cinderblock, 2018-11-07:
@techpaul Nice find!
I had been searching for that and must have skimmed over it 10 times!
What you found is the last part of the puzzle as, on it's own, what you posted does not quite work.

If we're waiting for a flag to get set, we need to make sure it is cleared first:
```C++
void clearTFF() {
  _XMC_UART_config->channel->PSCR = USIC_CH_PSR_ASCMode_TFF_Msk;
}
```
However the TFF bit is set on every sent byte so this on it's own does not quite work.

The solution that is working for me (with a rather hacked version of this code) is to first wait for the Tx buffer to have some room, then clear the flag, then wait for `TFF`:

```C++
void setup() {
  Serial.print("Hello World!\r\n");
  toggleDebug(); // Happens right after '\r' is starting to be sent.

  // Start Serial.flush()
  while (USIC0_CH0->TCSR & USIC_CH_TCSR_TDV_Msk);
  USIC0_CH0->PSCR = USIC_CH_PSR_ASCMode_TFF_Msk;
  while (!(USIC0_CH0->PSR_ASCMode & USIC_CH_PSR_ASCMode_TFF_Msk));
  // End Serial.flush()

  while (1) toggleDebug();
}
```
[image]

techpaul, 2018-11-07:
Like I said NOt tested and quick serach to find details

The first line of serial flush waits for the TX buffer (S/W) to empty, the alternative line was for the second of the lines in serial flush.

I would suggest any modification to serial flush be

Original parts
1.  Wait for TX buffer to empty
2.  Wait fortransmit buffer to empty (last byte loaded into final shift register)

Then
3. Clear TFF flag
4. Wait for TFF flag to set

I will see if I can get 5 minutes to put something together tonight

cinderblock, 2018-11-10:
> Like I said NOt tested and quick serach to find details

Yes. This is why I tested it and made modifications. 😉

mhollfelder, 2018-11-10:
Hi all,

We need to separate the TBUF register and the the shift register for the USIC in UART mode. 
The words are loaded by user software to the TBUF register and transferred to the shift register for being sent. 
If the status is `XMC_USIC_CH_TBUF_STATUS_IDLE`, we know that we can load another word to the TBUF register, but this does not indicate that the transmission is already done. 

The respective events of the shift register are: 
`XMC_UART_CH_STATUS_FLAG_TRANSMIT_SHIFT_INDICATION` `XMC_UART_CH_STATUS_FLAG_TRANSMIT_BUFFER_INDICATION`.

The event which indicates that the last data is sent from the shift register is `XMC_UART_CH_STATUS_FLAG_TRANSMIT_SHIFT_INDICATION`  

More information in the user manuel, e.g. for the [XMC1100](https://www.infineon.com/dgdl/Infineon-xmc1100-AA_rm-UM-v01_01-EN.pdf?fileId=5546d46255dd933d0155e31753b077af) in the USIC and ASC chapters.
Please check also the help for the XMCLib which you can find [here](http://dave.infineon.com/Libraries/XMCLib/XMC_Peripheral_Library_v2.1.18.zip).

I am currently working on implementing the FIFO for RX and TX as this issue showed me that we are not using the FIFO right now. 
Moreover, the UART blocks if you send data very quickly to the module as the ISR of the UART events is not implemented properly.

Once I have an update, I will let you know.
Would be great if someone could test it as well for further feedback.

Best regards,

Manuel

techpaul, 2018-11-10:
Getting FIFO's to work on USIC for ASC is a LOT of work

Back around March I removed the small amount of FIFO config that was there in init() function that was done wrong (one threshold value was too large for bitfields available)

First you must start in XMCLIB level writing to TBUFF in non-FIFO mode spin locks waiting for TBUF to be empty, but in FIFO mode writes to FIFO with NO checks for room in FIFO.

Secondly in XMCLIB level some of the bit settings for clearing and setting mode bits ASSUME some bits are always zero or one (missing flags to clear or set.not in combinations)

Then in XMC-for-Arduino level there are no checks in write or interrupt functions for space in FIFOs, or checks in interrupt routine for space in FIFO and what space in buffer to transfer max number of bytes in one interrupt call.

Similarly for RX there are no checks for how many bytes received in FIFO to transfer if enough space in RX buffer in one interrupt call.

All other HardwareSerial functions need to be reviewed for any differences in FIFO mode, like flush().

That is why my last fix on that was remove the poor FIFO implementation and put on back burner proper FIFO support.

mhollfelder, 2018-11-10:
I added an initial implementation for the TX FIFO to the develop branch with the following commit:

584c80ee43efef30675e4e39d370eb0b0594a073

The write() function will check the FIFO and buffer values respectively trigger ISR to empty the TX FIFO buffer.
I need to check that the FIFO is separated between the different USIC channels, i.e. as well for I2S and I2C on different platforms. 

Tested the code so far on a XMC4700 Relax Kit with:

```C++
void setup() {
  Serial.begin(1000000);
  while (!Serial) 
  {
  }
  // Print initial line
  Serial.println("UART test");
}

void loop() {
  // Send more data as the TX FIFO size
  Serial.print("AAAA AAAA AAAA AAAA\n");
}
```

Was running smoothly and I could debug it easily.
Will document in the wiki how to start debugging session with Visual Studio Code soon to trace down such issues/developments more efficiently with Arduino.

`flush()` is changed as well, but I could not test it so far.

Best,

Manuel

techpaul, 2018-11-11:
Whilst changing things I would suggest a change to XMCLIB file xmc_asic.c around line 384

function void XMC_USIC_CH_TXFIFO_Configure(XMC_USIC_CH_t *const channel,
                                  const uint32_t data_pointer,
                                  const XMC_USIC_CH_FIFO_SIZE_t size,
                                  const uint32_t limit)

Change line
~~~
  channel->TBCTR = (uint32_t)(channel->TBCTR & (uint32_t)~(USIC_CH_TBCTR_LIMIT_Msk |
                                                           USIC_CH_TBCTR_DPTR_Msk |
                                                           USIC_CH_TBCTR_SIZE_Msk)) |
                   (uint32_t)((limit << USIC_CH_TBCTR_LIMIT_Pos) |
                   (data_pointer << USIC_CH_TBCTR_DPTR_Pos) |
                   ((uint32_t)size << USIC_CH_TBCTR_SIZE_Pos));
~~~

To ensure the LOF and STBTEN bits are ALWAYS zeroed (not assumed zroed from reset)

~~~
  channel->TBCTR = (uint32_t)(channel->TBCTR & (uint32_t)~(USIC_CH_TBCTR_LIMIT_Msk |
                                                           USIC_CH_TBCTR_DPTR_Msk |
                                                           USIC_CH_TBCTR_SIZE_Msk |
                                                           USIC_CH_TBCTR_STBTEN_Msk |
                                                           USIC_CH_TBCTR_LOF_Msk )) |
                   (uint32_t)((limit << USIC_CH_TBCTR_LIMIT_Pos) |
                   (data_pointer << USIC_CH_TBCTR_DPTR_Pos) |
                   ((uint32_t)size << USIC_CH_TBCTR_SIZE_Pos));
~~~

closed 2022-01-06 by jaenrig-ifx