Infineon/XMC-for-Arduino #91: Missing Pin/Port functions for XMC2Go (and possibly other Boards)

cad435, 2019-03-19:
The Arduino-HAL function lacks (at least) the following Arduino-HAL functions/macros (usually defined in 'arduino_pins.h'):

digitalPinToPort
portOutputRegister
digitalPinToBitMask

these functions are used in some librarys (in my case right now Adafruits OLED-Display Library:
https://github.com/adafruit/Adafruit_SSD1306)

cad435

techpaul, 2019-03-20:
These are usally AVR specific for those writing libraries that use AVR machine code

Thy are not there for other boards when I last looked at the Arduino source for several boards.

There are many many SSD1306 repositories (some good, some bad and some ugly) and that is the FIRST one I have seen to do software bit banging SPI, Look at closed issues here for past experiences with other libraries for SSD1306 and similar displays.

The one you have chosen looks not very good to ME

cad435, 2019-03-21:
Hey,
didn't know that these are AVR-Specific, so i didn't look any further into the library.
Sorry for that.

I got it to work now, I tweaked the library to use the HW-SPI functions.

Thx for the hint!

Greetings
cad435

closed 2019-03-21 by cad435