Infineon/XMC-for-Arduino #96: How to code for UART connection between XMC1100 Boot kit to ESP32

xzsfd2457, 2019-05-31:


ArkajyotiChatterjee, 2019-05-31:
Hello,
Considering that you are using Arduino/PlatformIO IDE to program, you can connect the RX/TX of XMC1100 BootKit to TX/RX of ESP32. On ESP32 end you can use Software Serial(available already). On XMC1100 end you can either use the predefined 0 & 1 Arduino Pins for UART or define your own.
The rest is pretty similar to how you would connect an Arduino Uno with ESP32 over UART.
Also please ensure this setting in your Arduino IDE :  
[image]

ArkajyotiChatterjee, 2019-05-31:
Also, if possible, we would enable an example in the libraries folder of XMC-for-Arduino to take care of UART b/n XMC1100 Bootkit, XMC4700 Relax Kit & ESP32, ESP8266.
But, tell me why do you need to interface an XMC1100 Bootkit with an ESP32?

ironstrike, 2019-05-31:
I'm trying to send some sensor data over UART from XMC1100 Bootkit to ESP8266. In  XMC 1100 Bootkit doesn't support the SoftwareSerial. Is there any way to sen the data through UART.

xzsfd2457, 2019-05-31:
Do we need to send some sensor data from XMC1100 2GO to ESP8266  via UART ??
Give some examples, please

ArkajyotiChatterjee, 2019-05-31:
Hello ,
Please xzsfd2457, please elaborate on what Sensors you are trying to connect. As far as I know, Infineon Sensors can be directly interfaced with ESP32. But still, please do explain your design a lit bit more for me to help you out.
Regards,
Arka

ArkajyotiChatterjee, 2019-05-31:
Hello ironstrike,
The Software Serial library that you are installing and downloading from Github(I guess coming from Paul Stoffregen, can also be another module) does not support the XMC out of the box. You can do you own modifications for the XMC part.
Or you can use the hardware serial library for XMC1100 2GO to handle the UART comm. b/n XMC & ESP8266.
Get back in case if you have more doubts and you need clarifications.
Regards,
Arka

techpaul, 2019-05-31:
I will rephrase the question asked before several times

      Why do you need an XMC1100 (Boot Kit or 2GO) [b]AND[/b] ESP8266 ??

What is on the XMC1100 that is not on the ESP8266 ?

What is on the ESP8266  that is not on the XMC1100 ?

If you give up the serial to the host PC you can use it to talk to other devices like ESP8266 using standard Serial.write/.print

If you look through closed issues there is a how to add a second serial to XMC100 2GO

The way this is phrased sounds like an education project or classwork

ArkajyotiChatterjee, 2019-05-31:
Hello techpaul,
Yes, your comment makes sense. And the author is yet to throw light on the particular questions you asked.
And since there is already a closed issue on the particular topic, I would move forward to close this issue and redirect the author to the particular issue.
Thanks for pointing me to the closed issues section.
Regards,
Arka

closed 2019-05-31 by ArkajyotiChatterjee

xzsfd2457, 2019-05-31:
Need to integrate the Infineon radar sensor data to XMC2GO and send to ESP32 via UART

techpaul, 2019-05-31:
So you are using a Sense2Go NOT an XMC1100 2GO

So what are you using on an ESP32 or whatever board that needs this

You are giving so little information about what you are doing this sounds like Class/Course work

Give much more detail about your setup and what you are trying to do, at the moment we have nothing to go on or even know if your code is working on Sense2GO  to work on serial BEFORE connecting to anything else. As I often mentor 16 to 18 year olds on Computer Science Embedded projects, this really does come across as school work and expecting everything to be add this module and everything works by magic. You need to describe the scope of the whole issue and get things working a bit at a time and know the data from one section is working before integrating onto the next section.

xzsfd2457, 2019-05-31:
We did the integration with a sense2GO and XMC2GO
 but we need to send a Sense2GO data to from XMC2GO to  ESP32 via UART

ironstrike, 2019-05-31:
> Hello ironstrike,
> The Software Serial library that you are installing and downloading from Github(I guess coming from Paul Stoffregen, can also be another module) does not support the XMC out of the [...]

Thanks for the information...I'll try as you said and if there's any further doubts I'll let you know.

techpaul, 2019-05-31:
I will not answer anything further as you are expecting psychic answers from what sounds like student one line questions saying open ended things.

xzsfd2457, 2019-05-31:
ok, thank you techpaul...

ironstrike, 2019-05-31:
> Hello ironstrike,
> The Software Serial library that you are installing and downloading from Github(I guess coming from Paul Stoffregen, can also be another module) does not support the XMC out of the [...]

Arka I tried using Hardware Serial in XMC1100 2 GO but when I tried to compile it says that there was no pin declared . I commented the SERIAL DEBUG which was in pins_arduino.h even its not working any suggestion.
[image]

ArkajyotiChatterjee, 2019-05-31:
> Arka I tried using Hardware Serial in XMC1100 2 GO but when I tried to compile it says that there was no pin declared . I commented the SERIAL DEBUG which was in pins_arduino.h even its not working [...]

Hello ironstrike,
Please check the image that I attached in the morning. You don't need to declare a Serial1 from .ino file. Just ensure the setting I mentioned. But using the setting you cant see Serial output on your PC. 
Regards,
Arka