Infineon/XMC-for-Arduino #99: very many compiler warnings

mikrocoder, 2019-07-14:
Hi,

can you fix this?  I expect flawless conditions from a company like Infineon.   :-)

`
C:\Users\Worker\AppData\Local\Arduino15\packages\Infineon\hardware\arm\1.2.1\cores\xmc_lib\CMSIS\NN\Source\FullyConnectedFunctions\arm_fully_connected_q7.c: In function 'arm_fully_connected_q7':

C:\Users\Worker\AppData\Local\Arduino15\packages\Infineon\hardware\arm\1.2.1\cores\xmc_lib\CMSIS\NN\Source\FullyConnectedFunctions\arm_fully_connected_q7.c:73:90: warning: unused parameter 'vec_buffer' [-Wunused-parameter]

                        const uint16_t out_shift, const q7_t * bias, q7_t * pOut, q15_t * vec_buffer)

                                                                                          ^

C:\Users\Worker\AppData\Local\Arduino15\packages\Infineon\hardware\arm\1.2.1\cores\xmc_lib\CMSIS\NN\Source\FullyConnectedFunctions\arm_fully_connected_mat_q7_vec_q15_opt.c: In function 'arm_fully_connected_mat_q7_vec_q15_opt':

C:\Users\Worker\AppData\Local\Arduino15\packages\Infineon\hardware\arm\1.2.1\cores\xmc_lib\CMSIS\NN\Source\FullyConnectedFunctions\arm_fully_connected_mat_q7_vec_q15_opt.c:122:107: warning: unused parameter 'vec_buffer' [-Wunused-parameter]

                                        const uint16_t out_shift, const q7_t * bias, q15_t * pOut, q15_t * vec_buffer)

                                                                                                           ^

C:\Users\Worker\AppData\Local\Arduino15\packages\Infineon\hardware\arm\1.2.1\cores\xmc_lib\CMSIS\NN\Source\ConvolutionFunctions\arm_nn_mat_mult_kernel_q7_q15_reordered.c: In function 'arm_nn_mat_mult_kernel_q7_q15_reordered':

C:\Users\Worker\AppData\Local\Arduino15\packages\Infineon\hardware\arm\1.2.1\cores\xmc_lib\CMSIS\NN\Source\ConvolutionFunctions\arm_nn_mat_mult_kernel_q7_q15_reordered.c:50:64: warning: unused parameter 'pA' [-Wunused-parameter]

 q7_t     *arm_nn_mat_mult_kernel_q7_q15_reordered(const q7_t * pA,

                                                                ^

C:\Users\Worker\AppData\Local\Arduino15\packages\Infineon\hardware\arm\1.2.1\cores\xmc_lib\CMSIS\NN\Source\ConvolutionFunctions\arm_nn_mat_mult_kernel_q7_q15_reordered.c:51:65: warning: unused parameter 'pInBuffer' [-Wunused-parameter]

                                                   const q15_t * pInBuffer,

                                                                 ^

C:\Users\Worker\AppData\Local\Arduino15\packages\Infineon\hardware\arm\1.2.1\cores\xmc_lib\CMSIS\NN\Source\ConvolutionFunctions\arm_nn_mat_mult_kernel_q7_q15_reordered.c:52:66: warning: unused parameter 'ch_im_out' [-Wunused-parameter]

                                                   const uint16_t ch_im_out,

                                                                  ^

C:\Users\Worker\AppData\Local\Arduino15\packages\Infineon\hardware\arm\1.2.1\cores\xmc_lib\CMSIS\NN\Source\ConvolutionFunctions\arm_nn_mat_mult_kernel_q7_q15_reordered.c:53:66: warning: unused parameter 'numCol_A' [-Wunused-parameter]

                                                   const uint16_t numCol_A,

                                                                  ^

C:\Users\Worker\AppData\Local\Arduino15\packages\Infineon\hardware\arm\1.2.1\cores\xmc_lib\CMSIS\NN\Source\ConvolutionFunctions\arm_nn_mat_mult_kernel_q7_q15_reordered.c:54:66: warning: unused parameter 'bias_shift' [-Wunused-parameter]

                                                   const uint16_t bias_shift,

                                                                  ^

C:\Users\Worker\AppData\Local\Arduino15\packages\Infineon\hardware\arm\1.2.1\cores\xmc_lib\CMSIS\NN\Source\ConvolutionFunctions\arm_nn_mat_mult_kernel_q7_q15_reordered.c:55:66: warning: unused parameter 'out_shift' [-Wunused-parameter]

                                                   const uint16_t out_shift, 

                                                                  ^

C:\Users\Worker\AppData\Local\Arduino15\packages\Infineon\hardware\arm\1.2.1\cores\xmc_lib\CMSIS\NN\Source\ConvolutionFunctions\arm_nn_mat_mult_kernel_q7_q15_reordered.c:56:64: warning: unused parameter 'bias' [-Wunused-parameter]

                                                   const q7_t * bias, 

                                                                ^

C:\Users\Worker\AppData\Local\Arduino15\packages\Infineon\hardware\arm\1.2.1\cores\xmc_lib\CMSIS\NN\Source\ConvolutionFunctions\arm_nn_mat_mult_kernel_q7_q15_reordered.c:57:58: warning: unused parameter 'pOut' [-Wunused-parameter]

                                                   q7_t * pOut)

                                                          ^
[... 209 lines cut ...]
                                                ^

C:\Users\Worker\AppData\Local\Arduino15\packages\Infineon\hardware\arm\1.2.1\cores\xmc_lib\CMSIS\NN\Source\ConvolutionFunctions\arm_depthwise_separable_conv_HWC_q7_nonsquare.c: In function 'arm_depthwise_separable_conv_HWC_q7_nonsquare':

C:\Users\Worker\AppData\Local\Arduino15\packages\Infineon\hardware\arm\1.2.1\cores\xmc_lib\CMSIS\NN\Source\ConvolutionFunctions\arm_depthwise_separable_conv_HWC_q7_nonsquare.c:92:66: warning: unused parameter 'bufferA' [-Wunused-parameter]

                                                          q15_t * bufferA, 

                                                                  ^

C:\Users\Worker\AppData\Local\Arduino15\packages\Infineon\hardware\arm\1.2.1\cores\xmc_lib\CMSIS\NN\Source\ConvolutionFunctions\arm_depthwise_separable_conv_HWC_q7_nonsquare.c:93:65: warning: unused parameter 'bufferB' [-Wunused-parameter]

                                                          q7_t * bufferB)

C:\Users\Worker\AppData\Local\Arduino15\packages\Infineon\hardware\arm\1.2.1\cores\new.cpp:38:6: warning: the program should also define 'void operator delete(void*, unsigned int)' [-Wsized-deallocation]

 void operator delete(void* ptr)

      ^

C:\Users\Worker\AppData\Local\Arduino15\packages\Infineon\hardware\arm\1.2.1\cores\new.cpp:43:6: warning: the program should also define 'void operator delete [](void*, unsigned int)' [-Wsized-deallocation]

 void operator delete[](void* ptr)

`

techpaul, 2019-07-14:
I can see you are using library version 1.2.1, but what would be more helpful is to know

1. Which board you are compiling for
2. which libraries you included
3. Which special function you are using

These errors are not seen when doing an empty project test, so it suggests it is related to a specific configuration I can not determine from those errors.

mikrocoder, 2019-07-15:
Sorry.

a) XMC11002 XMC2Go,
b) serial
c) none

Testsketch is:
`void setup(void) {
  Serial.begin(250000);
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop(void) {
  
  heartbeat(500);

}

// ****** Funktionen ******
void heartbeat (unsigned int interval)    
{
  static unsigned long last_ms = 0;
  static bool state = LOW;
  
  unsigned long ms = millis();
  
  if (ms - last_ms >= interval) {
    last_ms = ms;
    state = !state;
    digitalWrite(LED_BUILTIN, state);
  }
}
`

Are all the information available?

techpaul, 2019-07-15:
Just made sure I was on V1.2.1 of XMC-for Arduino and noticed that blank sketch gives same errors I need to spend a few hours down the git blame trail and looking at a couple of files to see what someone has done. This was working fine before as empty sketch test, I did a major overhaul on this a year ago to correct the majority of the cores.

Unfortunately this might be in a few days as I am snowed under with paying work on other projects at the moment.

The standard tes I do and I wish was done by the automated checking mechanism on each new pull and merge is
  1. Configure for a board
  2. Create a new empty sketch
  3. Set preferences for warings ALL
  4. See if it clean compiles with NO warnings
  5. Repeat for all boards supported

I did not merge or test the latest release.

This test has been done before by me, even on the Arduino AVR and SAM cores and found problems that have been incorporated by them over the years.

mikrocoder, 2019-07-15:
I have a problem interpreting your answer correctly. If I understand you correctly, the warnings have been known for some time. You just don't have time to correct them?

techpaul, 2019-07-15:
1.  I do ***not*** work for Infineon this is an open source pltaform, I am ***one*** of the  contributor who has fixed many bugs that get ioften incorporated.

 2. This is  ***not*** my job, I am a part time contributor who has delved into the innards often

  3. I was trying to get more information as to what the area of problem is.

 4. I can only see the warnings in v1.2.1 and it will take time to back track where they came from. The library has been Warning free for some time, I will have to see what was added in V1.2.1

 5. As a ***part time*** contributor I have other work on many platforms

 6. As open source platform does not mean there are ***NOT*** lots of people waiting for your questions and able to drop everything to work on a problem. This is true of the majority of open source projects. Also even ones supported by manufacturers with dedicated support staff will put any issue into a queue if it involves fixing internals and not just explaining a mistake or mis-understanding. That is life as it is. Wait to see how long people like Microsoft, Apple, Adobe and many others fix bugs, there is a video somewhere of someone celebrating a 10 year old in one database software.

mikrocoder, 2019-07-15:
Then my guess is based on a misunderstanding. I thought the board support came directly from Infineon. The name of the github branch was clear to me. Sorry for confusing the two.

And yes I have time and can wait. I just wanted to report it so someone could take care of it.

ArkajyotiChatterjee, 2019-07-17:
Hello @mikrocoder,
The compiler warnings that you received can be categorized into two parts:
1.  Compiler warnings because of CMSIS NN Libraries
2. Compiler warnings because of new.cpp

Now if you check the warnings (part1): The compiler is throwing unused parameter warning in the files inside CMSIS\NN\Source folder (link: https://github.com/Infineon/XMC-for-Arduino/tree/master/arm/cores/xmc_lib/CMSIS/NN/Source). Now this CMSIS section, as a matter of fact, was authored by ARM(link: https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/NN/Source).  And we are using their codebase. So, even if I can solve this issue I would rather leave this to the attention of engineers from ARM as they would know better than me as to what all they meant when they wrote the program like that and what all can go wrong if someone just update the code keeping in mind the compiler warnings.

As for part2(new.cpp), there are only 2 compiler warnings, warning us about unsized deallocation function(to know more about this error(-Wsized-deallocation) please visit here: https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html).
Now there is a fair reason behind this (as this is a default in GCC6). What we would do is, not just suppress this warning but rather add the necessary function and update the codebase.
Thank You for pointing this out for us. We would try to take care of this as soon as possible. Please do comment if I have not clarified any point. And as you mentioned in your initial comment, we do take flawless quality seriously.
Thanks again,
Arka

mikrocoder, 2019-07-18:
Hello, everything's fine. I just wanted to report it because it was just too many warnings and I thought someone should take care of it here. Also, I thought at first the code came directly from Infineon. Which made me even more thoughtful.
Sorry.
The main thing is that it is reported and someone can fix it. Thanks for taking care of it.

ArkajyotiChatterjee, 2019-07-18:
Hello mikrocoder,
The board support for XMC in Arduino is coming from Infineon. But the exact library that you pointed out(CMSIS NN), that support is coming from ARM.
Hope this helps.
Regards,
Arka

techpaul, 2019-09-13:
Having checked things again after pull #106 I have noticed this is NOT a CMSIS problem it is a platforms.txt issue and definition of what each processor supports that has obviously changed MEANING in newer versions of CMSIS. So WRONG or MISSING defines describing the processors is the main issue.

The errors are for high end DSP functions that may or may not be included correctly for XMC4700 (or not used in XMC-for-Arduino normally), however they are also being included for ALL XMC1 series which have NO hardware DSP.

The lines in platform.txt that worry me are for ALL processors (not in boards.txt for specific processors)
~~~
compiler.nn_include.path={compiler.cmsys_rc2_include.path}/CMSIS/NN/Include
compiler.dsp_include.path={compiler.cmsys_rc2_include.path}/CMSIS/DSP/Include
~~~
and
~~~
compiler.arm.cmsis.path=  "-I{compiler.cmsys_rc2_include.path}/XMCLib/inc"  "-I{compiler.dsp_include.path}" "-I{compiler.nn_include.path}" "-I{compiler.cmsys_include.path}"  "-I{compiler.cmsys_rc2_include.path}/LIBS"  "-I{build.variant.path}"  "-I{build.variant.config_path}"
~~~
Due to number and nature of warnings many NN and DSP files are being compiled unnecessarily for XMC1 series and some parts that should not for XMC4

Wish the standard compilation for ALL people was show ALL warnings and errors then act upon them for NO warnings at all

techpaul, 2020-02-10:
Currently using develop branch and PR #126 gives cleaner compiles for warnings on all boards tested

Does not fix #124 USB issues on its own this has highlighted many libraries source will need review on how header files are included

techpaul, 2020-06-29:
Now incorporated into release V1.5.0 so closing compiler warnings Note USB issues unknown status

closed 2020-06-29 by techpaul