Infineon/XMC-for-Arduino #103: IFX9201.cpp minor bug

techpaul, 2019-09-09:
This bug is referenced in #68

Basically if analogWriteResolution has been set to anything other 8 bits in main parts of code, incorrect values will be sent to analogWrite, reducing range of duty cycle

Previous comment

The comment about Duty Cycle and default values was told in June

There is minor bug in IFX9201.cpp
~~~
uint8_t IFX9201::DutyCycleToanlogWriteValue(uint8_t duty_cycle)
{
	uint8_t ret = ( (duty_cycle >= 100u) ? 255u : (uint8_t)((double)duty_cycle * 2.55) );
	return ret;
}
~~~

The value used shoulkd be uint16_t to match AnalogWrite function

The 255 values should be replaced with values determined by analogWriteResolution and NOT assume it is 8 bit as the max value could be 255, 1023, 4095, 6553, or other vallues. Sending 8 bit values when the resolution has changed to 12 bit will give nearly no response never getting above 16.67%

This also needs either class members or helper functions in wiring_analog.c to get what the current resolution is.

Been on back burner for me and not had a board to test on as well does not help

ArkajyotiChatterjee, 2019-09-09:
Hello techpaul,
To summarise, what you are suggesting is this :

- Don't assume that analog Write Resolution is only 8 bits

- Get the current resolution from wiring_analog.c

- Update the DutyCycletoanalogWriteValue from uint8_t to uint16_t so that it can handle other resolution values

- Update the function so that it can handle the duty cycle to analog value conversion based on the current resolution

If the points mentioned above looks okay, then we would update the IFX9201 . You can test it and close this issue then.
Thanks,
Arka

techpaul, 2019-09-09:
Yes but there are no global variables or public class members or helper functions for getting number of bits or max values for read or write resolution from wiring_analog.c

I CANNOT test this on hardware  I do not have a IFX9201 or HBride2GO

I can see the errors in the code and not had time to implement a set of fixes, needs someone else to test on hardware.

Files affected 
# IFX9201.cpp
# wiring_analog.c
# wiring_analog.h

If I get time I will sync up my repository and put some suggested fixes up in a few days, somewhat busy at the moment.

dizi1, 2019-09-09:
What a joke.. one coder writes libraries without owning the $13 Hbridge2go board.. another coder "fixes" that bugged code with lots of syntax errors, Arduino IDE cant load the zip file library.. no example usage for a first time install..

Now more suggestions on how to fix a bug WITHOUT AGAIN HAVING A $13 BOARD TO TEST ON IT !

WHAT A JOKE PROJECT.

Infineon Technologies is making a JOKE of their products.. instead of assigning an *engineer* to fix the code/implementation.

ArkajyotiChatterjee, 2019-09-09:
Hello dizl1,
First of all, techpaul is an independent contributor who is a member of our Infineon Github because of his contributions.
Secondly, every codebase has some or other bugs. We appreciate people pointing out the errors, but what we don't appreciate is impolite rude people.
Thirdly, to work with IFX9201 library, you don't need to install any zip file. When you have the XMC package for Arduino installed the IFX9201 Library is installed by default. Now, I can understand you are a novice /beginner and if you have asked nicely, I would have shown you how to work with this.
Fourthly, there is a whole lot of information on how to work with the libraries targetted towards first-time users. It's your problem that you can't find or understand what is written there.
Finally, we(me and my colleagues) are finding time out of our schedule to distribute the software in open source communities targeted for makers. Every one of us is extremely capable" engineers" and we do assign bugs and issues to relevant people based on their particular expertise. But, it's futile explaining these things to someone as obnoxious as you.
Please refrain from posting such comments on this platform.
Arka

dizi1, 2019-09-09:
I guess its a wake up call how ridiculous your "engineers" skills are implemented here ..   writing a code "on the air" wasting anyone who read this time.

My RANT is with Infinieon.. celebrating "an arduino version" on their marketing page while its NOT a working code.

Peace!

techpaul, 2019-09-09:
There are many many bugs in many of the Arduino/Genuino libraries and IDEs, as well as many other manufacturers.

There is no such thing as perfect software, there is very good nd all the rest.

You have the ability to make mdoficiations and suggestions yourself and be a contributor, which is more than Microsoft, Apple and others will allow.

ArkajyotiChatterjee, 2019-09-09:
Stop commenting and polluting the issue thread.
Next time you will be blocked.

jaenrig-ifx, 2019-09-09:
Dear dizi1, 

I am afraid to inform you that respect and good manners are within our "nowhere stated" code of conduct. And this is "nowhere stated" because normally applied to any fruitful interaction between human beings in any life context. Specially if you want something from them.

As a troll, you are doing good. Congratulations :) As a software engineering who provides constructive negative feedback, I must say you have quite some room for improvement. While you refine such skills, I will delete your comment and give you another change to point out bugs and issues as a good-tempered skilled engineer.

Best Regards,
Juan A

dizi1, 2019-09-10:
Hi All.

I apologize to anyone who felt i was rude. 

I wish Infinieon would contribute few free boards and send them to this project's contributes/engineers - rather than let them guess the outcome of their code-changes.

Having said that, i will do my best to post bugs in ill-tempered way.. ;)

Here goes.. from a user perspective:
1. I installed version 1.2.1 on Arduino IDE 1.8.21  - restarted Arduino - heading to file->examples does not show any examples for hbrdige2go board (i,e: motocontrol.ino on last version)

looking at Sketch->Include library does not shows any library for hbridge2go thus compiling the default example code from last version gives this error:
exit status 1
IFX9201.h: No such file or directory

Once this is fixed i can test further and report back.

dizi1, 2019-09-10:
ok the example code in SPI mode works - forwards, stop , backwards. 
  IFX9201_HBridge.begin(SPI, CSN, DIR, PWM, DIS);

Stop() does not shut the hbridge 100%,
I still get minimal current flowing 

trying to set duty cycle failed.   ex: IFX9201_HBridge.forwards(100); 
Actually any number shuts OFF the forwards completely (zero current)  
even  IFX9201_HBridge.forwards(50);   turns off current completely.

Yet running  IFX9201_HBridge.forwards(); works (i assume at 50% default duty cycle) 

----

On PWM mode,: IFX9201_HBridge.begin( DIR, PWM, DIS);

forwards , stop (complete shut off current) , backwards - all work well.

Trying to set duty cycle value doesn't affect the current.

Trying to set new frequency also doesn't work.
IFX9201_HBridge.setPWMFreqency(5000);

Thanks

techpaul, 2019-09-10:
Quite a lot of errors can be found without any hardware, by following the flow of the code and library calls, too often people create code that uses data in a way that will causes problems or pass the wrong data.

Checking something will compile can be done by anybody with no boards as the compile is done on the PC host not the board itself. No need to download to target to prove that.

In your case have followed

  Tools -> Board 
     Set to XMC1100 H-Bridge 2Go

  File -> Examples
   -      Lists examples for A:: XMC boards (some examples do not work on all boards check first when loading examples in the code)
  -       Selected   IFX9201 H-Bridge
 -        Selected Motor Control

Note There is no example labelled XMC1100 H-Bridge 2Go there is an example of the add on chip IFX9201

The example loads in a new window

Do a Verify Top left button on tool bar
or type CTRL-R on Windows
or Sketch -> Verify/Compile

Example compile with no errors

When reporting level set to ALL some WARNINGS appear that are due to much lower level ARM libraries beyond Infineon's contro, but I believe has been passed up the chain.

Note Most examples are grouped not by board but functionality, and where possible work on all XMC boards, some involving specific chips like Radar or H-Bridge or specif processor functions like Hibernate ONLY work on specific boards which should be in the top of the example code or when compiling give errors of wrong board selected.

In my case the last motor control I used with an XMC board and then laid out design to make my own PCBs (many of them), was ffor an electric wheelchair, needed to use existing motot controllers with 10A in motion and 30A peak current when starting handling, needing larger motors, batteries and much larger cabling than could fit on an H-Bridge 2Go.

Currently I have no applications for that low a motor control capability, so no motors to drive.

Other than the IFX9201, most of the functionality on the XMC1100 H-Bridge 2Go can be tested on an XMC1100 2Go which I have several of for doing projects with 14 to 18 year olds in their school classes or course work or exam projects.

dizi1, 2019-09-10:
So setting the frequency & duty cycle are what needs to be looked into (i prefer in PWM mode)

Thanks

ArkajyotiChatterjee, 2019-09-10:
> ok the example code in SPI mode works - forwards, stop , backwards.
> IFX9201_HBridge.begin(SPI, CSN, DIR, PWM, DIS);
>
> Stop() does not shut the hbridge 100%,
> I still get minimal current flowing [...]

Hello @dizi1,
Thank You for the questions. Well, the answers are ->
1. To change the duty cycle(like forwards(70) or something will only work in PWM Mode.
2. Changing the PWM frequency does indeed works. What you can do is this : 
in IFX9201.h in place of void setPWMFreqency(uint16_t);  make it 
int16_t setPWMFrequency(uint16_t);
Inside IFX9201.cpp, in the definition of setPWMFrequency change the definition to this:
`int16_t IFX9201::setPWMFreqency(uint16_t freq)
{ 
     int16_t freqReturn = 2 ;
	if(freq <= IFX9201__MAX_PWM_FREQ)
	{
		m_PWMFreq = freq;
		freqReturn = setAnalogWriteFrequency(m_PWM, m_PWMFreq);
	}
     ret freqReturn;
}`
Then from your Arduino Code, read the value that setPWMFrequency returns and print on the serial monitor. If value is 0, that means its working.
Max Value of PWM is 20k (defined inside util/IFX9201_conf.h)
3. When you call stop() in SPI Mode, we write to a register (Check Line 171 in IFX9201.cpp). So yes it does not shuts it down 100%. In PWM, we switch off the MOSFETS completely(Line 177 of IFX9201.cpp) where we are turning Disable pin HIGH.

dizi1, 2019-09-10:
Thank you for your reply.

I just did and tested what you suggested:
In PWM i still can not change the PWM, its always on same default value, i.e: 
IFX9201_HBridge.forwards(1); or IFX9201_HBridge.forwards(90); = same motor speed 

I changed the code for the frequency as you suggested, Arduino IDE gave error:
exit status 1
'class IFX9201' has no member named 'setPWMFreqency'

Also i fixed a syntax error in your suggested code: ret = return

ArkajyotiChatterjee, 2019-09-10:
Hello @dizi1 ,
I made a typo it should be setPWMFrequency and not setPWMFreqency . And it was just a rough idea for you to change the stuff.
Anyway fix the typo in your code and try it again.
Regards,
Arka

dizi1, 2019-09-10:
Thank you Arka, 

Typo changed and when i set a new frequency i get -1 in return.
I assume a frequency change was successful?

If so, one last item to fix is the PWM on PWM mode. Any value gives the same motor speed.

Thanks

ArkajyotiChatterjee, 2019-09-10:
Hello @dizi1,
No. If frequency value change was succesful then you would get 0. BTW have you selected PWM mode in your Arduino Code?
Arka

dizi1, 2019-09-10:
Yes i have changed the selection to  IFX9201_HBridge.begin( DIR, PWM, DIS);

When i set the frequency i get -1  
   RetFreq = IFX9201_HBridge.setPWMFreqency(10000); // full is = 20k
   Serial.println(RetFreq); 

I assume the MAX is 20k (though on DAVE's example by Infineon its 30K which worked fine for me)
What's the minimum?

techpaul, 2019-09-10:
If you ever get compilation time error  it will be configuration or incorrect code

setPWMFrequency can fail for **two** reasons

1. Frequency too high or invalid number
2. Invalid pin for PWM

For item 1 check values passed in against the function setAnalogWriteFrequency in wiring_analog.c

For item 2 have you incorporated the pins fix in pull request #102

I would suggest checking that first.

I am looking at adding more error codes to wiring_analog.c  to catch errors like this

techpaul, 2019-09-10:
If you get pull request mods for #83 this improves a lot of wiring analog functions so setAnalogWriteFrequency returns -1 for frequency or pin error and 0 to F to denote a correct prescaler has been able to be selected.

This was only merged yesterday

dizi1, 2019-09-10:
Question: do  pull request #102 & #83 already in version 1.2.1 ?

dizi1, 2019-09-10:
Ok how would you recommend a novice like me to merge pull #102 manually, for testing 

Thanks

techpaul, 2019-09-10:
Pull #102 is fairly simple go to pull request #102 look at the changes Red background removed blue background is added and shows path to file in the package.

Arduino IDE File -> Preferences at bottom has link to where Preferences are stored in a windows explorer window then open folders

1. packages
2. Infineon
3. hardware
4. arm
5. 1.2.1

From there you should be able to see the paths that match and edit  the same files locally to match

The alternative is to go to Code link above download a zip file extract to temporary folder and copy files into paths as described above

Another alternative is to get a git client create a github account and other longer things that we use for managing pull requests and synchronisation of files, which would be a further hill to climb.

dizi1, 2019-09-10:
Thanks, i actually done just that manually replacing the red lines with the blue.
I get 0 in frequency change but no power on forwards and backwards at any value.

so 1.2.1 pin map is ok expect the PWN pin.

Any suggestions? pull #102 write:
#define PWM 	11

yet on this line its not:
  /* 11  */   {XMC_GPIO_PORT2 , 10},  // PWM / PWM1 output (Fixed on PCB) P2.10

I trued setting the above to 11 and the DIS to 10 but no luck.

ArkajyotiChatterjee, 2019-09-10:
Hello @dizi1 ,
I will send you a new package tomorrow.
@MatamRohith  : Can you work with @dizi1 tomorrow and check what all he needs to change?
Thanks,
Arka

dizi1, 2019-09-10:
Thanks, basically i think the issue left is the PWM/Frequency  pin/implementation. 
package 1.2.1 is ok except the PWM/Frequency.

Pull #102 did not work and has conflicting pin numbers?

Thanks

ArkajyotiChatterjee, 2019-09-10:
> Question: do pull request #102 & #83 already in version 1.2.1 ?

Hello @dizi1 ,
I believe you have not updated the XMC-for-Arduino library after tomorrow. Since you cannot properly update your codebase according to #102 , better yet just start afresh (this I am suggestimg because maybe you are not familiar with git, but if you are familiar then you can just pull the latest changes in your local machine).
In any case, best is to just delete all the packages and start afresh.
Because we did try everything locally and everything that I suggested to you was working for all of us.
Arka

dizi1, 2019-09-10:
no problem. fresh package tomorrow then.

ArkajyotiChatterjee, 2019-09-10:
Hello @dizi1,
If it's not a problem, do share your email id with me.
Ark

dizi1, 2019-09-10:
how? i dont see a DM option here

ArkajyotiChatterjee, 2019-09-10:
Mail me at arkajyoti.chatterjee@infineon.com

dizi1, 2019-09-10:
Great. Thanks.

techpaul, 2019-09-13:
Currently after Pull Request #106 which is part of the fixes I have the structure being test compiled at present to
- Add error returns to IFX9201 library functions
- Improve code flow
- Handle duty cycle conversion on ANY analogWriteResolution setting scaled correctly to analogWrite, even if the resolution changes between calls
- Added more error codes in IFX9201
- Split example into separate SPI and PWM mode examples

Currently working on an example of using IFX9201 with error checking and notify on Serial.prints this will probably be on PWM mode only, but may be made into both versions.

I will test this with XMC1200 2G) in PWM mode but will request this pull request be tested by someone else in SPI mode on actual H-Bridge 2GO

techpaul, 2019-09-13:
Draft Pull Request #107 available for improved IFX9201 library

Please test using SPI at least on H-Bridge 2 GO and let me know results for all examples and in your applications. I need external testing before merging.

For those who cannot use git pul/checkout facilities see my profile and request a zip file which I can supply when I have time. The important bit is to copy the libraries/IFX9201 folder and all its contents to overwrite your libraries folder

techpaul, 2019-09-23:
Please see Pull Request #107 for updates

techpaul, 2019-10-16:
Pull request has no been merged

Thanks to @9Volts9er for additional testing

closed 2019-10-16 by techpaul