Infineon/XMC-for-Arduino #104: Error message (and solution) : JLink software is not installed

Willem23, 2019-09-11:
I was getting an error message when trying to flash code to the XMC1100 boot kit after successful compilation. System is Windows 10, 64bit, Arduino 1.8.9, Segger Jlink V650.

The error says:
[Error] Infineon.DebuggerExceptions: It seems that JLink software is not installed please download from www.segger.com and install it. You can specify it by setting java property xmcFlasher.JLink.dllPath

Others have encountered the issue as well. See https://www.infineonforums.com/threads/5697-XMC-Flasher-is-not-working

Tried to install and run the Infineon XMCFlasher to rule out Arduino issues. It resulted in the same error. After much hairpulling and googling it turns out that the XMCFlasher cant find ''JLinkARM.dll''.
You can fix that for the standalone XMCFlasher by modifying the xmcFlasher.bat like so:
-----------------------------------------------------------------------------------------------------
java -DxmcFlasher.JLink.dllPath="C:\Program Files (x86)\SEGGER\JLink_V650\JLinkARM.dll" -jar %~dp0\xmcflasher.jar %* --gui 
-----------------------------------------------------------------------------------------------------
The flasher will start in GUI mode.
Select ''SEGGER'' in XMCFlasher instead of default DAP link and ‘’connect’’ 

You can fix the problem in Arduino IDE by setting an environment value for your Windows account:
-----------------------------------------------------------------------------------------------------
JAVA_TOOL_OPTIONS: "-DxmcFlasher.JLink.dllPath=C:\Program Files (x86)\SEGGER\JLink_V650\JLinkARM.dll"
-----------------------------------------------------------------------------------------------------

Arduino Debug window will now report:
Picked up environment var ... JAVA_TOOL_OPTIONS = "-DxmcFlasher.JLink.dllPath=C:\Program Files etc

Hope this helps others struggling with the same problem.
Moderators: Please add to the Wiki page.

techpaul, 2019-09-11:
Seems to be a common Segger problem on 64 bit systems. Already seen on Linux.

Thank you for the information, will add to my wiki at least

at https://github.com/techpaul/XMC-for-Arduino/wiki

Will leave to wiki maintainer to add there

techpaul, 2019-09-11:
On this wiki as of now

https://github.com/techpaul/XMC-for-Arduino/wiki/Windows-10-64-bit-Flasher

ArkajyotiChatterjee, 2019-09-12:
Hello @techpaul  & @Willem23 ,
Thanks a lot for pointing out the issue as well as solution. We will add it to our wiki and then close this topic.
Regards,
Arka

techpaul, 2019-09-13:
Seen on older 32bit Windows as well and V600d of Segger Jlink

Same fix cures this

Problem occurs if some other programme creates Program Files (x86) alongside Program Files  on 32 bit system, in this case it was ST Microelectronics flashing software for another of my customers and their systems.

hubertlxb, 2021-06-25:
I met the same issue and try to fix it following your suggestion.
But I can find where to set the enviroment value in Arduino IDE.
could you please help? thanks!
"You can fix the problem in Arduino IDE by setting an environment value for your Windows account:"

techpaul, 2021-06-27:
How about searching for "Windows 10 environment variable" that is how to configure Windows and has been around since the early days of Windows.

Loads of sites come up here is one of the many hundreds
https://www.google.co.uk/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=2ahUKEwiwjoSp8rfxAhW0mFwKHSo8BZYQFjAKegQIDRAD&url=https%3A%2F%2Fwww.architectryan.com%2F2018%2F08%2F31%2Fhow-to-change-environment-variables-on-windows-10%2F&usg=AOvVaw2TshLtXFaJFYyJRku3RDBN

techpaul, 2021-07-19:
This is a Wiki item now

closed 2021-07-19 by techpaul