Infineon/XMC-for-Arduino #108: Flashing on Ubuntu 18.04 does not work

odo2063, 2019-09-19:
Hello again!

Flashing does not work on Ubuntu 18.04 64bit.

"ln -s /opt/SEGGER/JLink/libjlinkarm.so /usr/lib/libjlinkarm.so" does not help this time...

neither java11 or java8...

`java -jar /home/teste/.arduino15/packages/Infineon/tools/XMCFlasher/1.2.1/XMCFlasher.jar -p /tmp/arduino_build_369585/vl53l0x.ino.hex -device XMC1100-0064
--------------------------
Infineon XMC Flasher Lite
Copyright Infineon Technologies 2017
--------------------------
Linux
Operating System: Linux

Initialisation
--------------------------
Linux
`

techpaul, 2019-09-23:
Works for me, just went onto an Ubuntu 18.04 laptop installed or already installed

1. Arduino IDE 1.85
2. XMC-for-Arduino V1.2.1 and latest patches
3. Segger Jlink software V6.50b (installed)
4. added link

Did a test compile and download to XMC2GO (XMC1100-064)

First attempt error was Could not connect to device

Restart  system then try again successful upload and serial comms monitor
Upload details
~~~
Sketch uses 8556 bytes (13%) of program storage space. Maximum is 65536 bytes.
Global variables use 2372 bytes of dynamic memory.
Picked up JAVA_TOOL_OPTIONS: 
Loading JLink Library... 
--------------------------
Infineon XMC Flasher Lite
Copyright Infineon Technologies 2017
--------------------------
Linux
Operating System: Linux

Initialisation
--------------------------
Linux
JLink Library Version: V6.50b

Configuring Device
Device Name : XMC1100-0064
Interface   : SWD
Speed       : 4000
File Upload
--------------------------
Uploading /tmp/arduino_build_984576/MotorControl-SPI.ino.hex
... Done
--------------------------
... Finished Succesfully
~~~

My java is openjdk as shown by
~~~
test@CS02:~$ java -version
openjdk version "1.8.0_222"
OpenJDK Runtime Environment (build 1.8.0_222-8u222-b10-1ubuntu1~18.04.1-b10)
OpenJDK 64-Bit Server VM (build 25.222-b10, mixed mode)
test@CS02:~$ 
~~~

I believe shows running on 64 bit Ubuntu 18.0.4

odo2063, 2019-10-09:
Rebooting and changing java does not help... :-/

techpaul, 2019-10-14:
Without more details on the error I cannot see what is or is not working

Can you provide a copy of the output from 
~~~
java -version
~~~
It may be that your old java has not been removed completely

Also what Segger version you have installed for Jlink

As said before my configuration does work as shown above

odo2063, 2019-10-14:
java is same as yours, segger is latest(6.50b), exception is arduino 1.8.10.

it stops on 
```
Picked up JAVA_TOOL_OPTIONS: 
--------------------------
Infineon XMC Flasher Lite
Copyright Infineon Technologies 2017
--------------------------
Linux
Operating System: Linux

Initialisation
--------------------------
Linux
```

techpaul, 2019-10-14:
When I get some time (too many family issues and customer panics at moment) this week I will update my Arduino IDE and see if I get the same problem. Unless there is some other JAVA_TOOL_OPTIONS:  mismatch, it looks like it is trying to run the correct JLink library.

At the moment I am confused and wonder what happens if just try to connect using XMC Flasher (this is supposedly windows but is Java and despite blurb I have had this working even on XP for mainly manipulating BMI on new boards to make them look like boot kits even though actually a complete board doing another function.

techpaul, 2019-10-14:
Link for XMC Flasher is 
https://www.infineon.com/cms/en/product/microcontroller/32-bit-industrial-microcontroller-based-on-arm-cortex-m/#!tools

odo2063, 2019-10-14:
I already tried it with jlink command which seemd to work...can't we just get rid off that java stuff?

techpaul, 2019-10-14:
The whole of Arduino IDE or Eclipse are based on Java for multi-platform
Many tools are based on Java or Python for multi-platform work without recompiling etc

odo2063, 2019-10-17:
do we have the source of the Java-flasher?

techpaul, 2019-10-17:
Jlink is Segger's closed source

Base Arduino IDE is available from the github Arduino repository

The boards and platform files and how the flasher is called are part of the XMC-for-Arduino

XMC-LIB source is included in the XMC-for-Arduino but the maintainers are internal to Infineon only

CMSIS and other Arm parts are part of ARM and other distrubtions are maintained as part of the ARM and others offerings, most maintained off of Github some parts are in your source tree as source.

techpaul, 2020-01-19:
After a very busy couple of months with family issues I rerun the tests as above

**AFTER** updating Arduino IDE to latest 1.8.10 (for other reasons as well)
updating XMC-for-Arduino to V1.3.0 recent;y released

Using latest patched Ubuntu 18.0.4 and previous Jlink software version everything passed again

~~~
Sketch uses 8324 bytes (12%) of program storage space. Maximum is 65536 bytes.
Global variables use 2372 bytes of dynamic memory.
java -jar /home/test/.arduino15/packages/Infineon/tools/XMCFlasher/1.2.1/XMCFlasher.jar -p /tmp/arduino_build_732132/MotorControl-PWM.ino.hex -device XMC1100-0064 
Picked up JAVA_TOOL_OPTIONS: 
--------------------------
Infineon XMC Flasher Lite
Copyright Infineon Technologies 2017
--------------------------
Linux
Operating System: Linux

Initialisation
--------------------------
Linux
Loading JLink Library... 
JLink Library Version: V6.50b

Configuring Device
Device Name : XMC1100-0064
Interface   : SWD
Speed       : 4000
File Upload
--------------------------
Uploading /tmp/arduino_build_732132/MotorControl-PWM.ino.hex
... Done
--------------------------
... Finished Succesfully

~~~

I do not know what else to suggest on this issue as I cannot repeat it

techpaul, 2020-01-23:
Closing issue as unable to replicate

schumar, 2021-01-10:
I can reproduce the issue.

It seems to be caused by `XMCFlasher` doing a recursive search for a file in `/usr/lib/`, but not correctly coping with symbolic links.
On my machine, those were
```
/usr/lib/llvm-6.0/build/Debug+Asserts -> ..
/usr/lib/llvm-7/build/Debug+Asserts -> ..
```
from the packages `llvm-6.0-dev` and `llvm-7-dev` -- but I guess there are many packages that might provide such symlinks.

Found by prefixing the `java` command with `strace -efile -f`, and then seeing an infinite loop with lines like
```
[pid 16489] stat("/usr/lib/llvm-6.0/build/Debug+Asserts/build/Debug+Asserts/build/Debug+Asserts/build/Debug+Asserts/build/Debug+Asserts/build/Debug+Asserts/build/Debug+Asserts/build/Debug+Asserts/build/Debug+Asserts/build/Debug+Asserts/build/Debug+Asserts/build/Debug+Asserts/build/Debug+Asserts/build/Debug+Asserts/build/Debug+Asserts/build/Debug+Asserts/build/Debug+Asserts/build/Debug+Asserts/build/Debug+Asserts/build/Debug+Asserts/build/Debug+Asserts/build/Debug+Asserts/build/Debug+Asserts/build/Debug+Asserts/build/Debug+Asserts/build/Debug+Asserts/build/Debug+Asserts/build/Debug+Asserts/build/Debug+Asserts/build/Debug+Asserts/build/Debug+Asserts/build/Debug+Asserts/include/llvm/BinaryFormat/ELFRelocs/Lanai.def", {st_mode=S_IFREG|0644, st_size=542, ...}) = 0
```

schumar, 2021-01-10:
@techpaul , should I open a new issue for this?

techpaul, 2021-01-10:
@schumar

XMCFlasher is not maintained in this repository it is used by the build.

This suggests an order of installation of the livm packages and after effects as well as symlink following issues that need to be brought up with the maintainers of XMCFlasher. I do not personally know who that would be.

mhollfelder, 2021-01-11:
@schumar @techpaul 

Thanks for bringing this up - the problem is related to the recursive search for the SEGGER pre-compiled libraries.
The XMCFlasher is linked against these dynamic libraries to implement the flashing of the XMC devices.

However, on several Linux distros the path changes and/or is set by users. 
So a search was implemented, but it looks like that it has problems to resolve the paths correctly.

I will check what we can do and how to solve this issue - I will come back to you.

Best,

Manuel

mhollfelder, 2021-01-11:
You can also set `IFX_JLINK_PATH` as an env variable with the correct path to the library.
So in principal it should be possible to overwrite the behavior of searching for the library as it can be seen below from the code.

Basically, we have the following implementation for Linux in XMC Flasher:
The full source code is right now not available, but this can already give some ideas how it is implemented:

```Java
public String SearchLib(String jarpath) throws DebuggerExceptions
{
    File x = null;

    Map<String, String> env = System.getenv();
    if (env.containsKey("IFX_JLINK_PATH"))
    {
        String path = env.get("IFX_JLINK_PATH");
        x = finder(path);
    }
    if (x == null)
    {
        x = finder(jarpath + "/etc/linux");
    }
    if (x == null)
    {
        x = finder("/usr/local/lib");
    }
    if (x == null)
    {
        x = finder("/usr/lib");
    }
    if (x == null)
    {
        x = finder("/opt/SEGGER");
    }
    if (x == null)
    {
        throw new DebuggerExceptions("Unable to locate libjlinkarm.so file in /usr/lib, /usr/local/lib, etc/Linux, /opt/SEGGER or variable IFX_JLINK_PATH");
    }

    return x.toString();
}
```

So, we check for a couple of locations to find the respective library called `libjlinkarm.so` with the following function:

```Java
public File finder(String dirName)
{
    File root = new File(dirName);
    File[] list = root.listFiles();

    if (list == null)
        return null;

    for (File f : list)
    {
        if (f.isDirectory())
        {
            File ret = finder(f.getAbsolutePath());
            if (ret != null)
            {
                return ret;
            }
        }
        else
        {
            if (f.getName().equals("libjlinkarm.so"))
            {
                return f;
            }
        }
    }
    return null;
}
```

As it can be seen, the search is not completely safe as it does not have a check of the recursive level.
I will check and implement some changes as soon as I have time.

Best regards,

Manuel

techpaul, 2021-01-12:
Hmmm me thinks set IFX_JLINK_PATH as an env variable with the correct path to the library will still get the problem as it will go to the same directory that the symlinks point to which is itself

mhollfelder, 2021-01-12:
Yes, you are right if the other path has the symlink as well - the `listFiles()` returns `null` if you provide a file to it so you can only disable the default search behavior for the paths by providing a path to a folder without the symlinks, but not provide the complete path directly.

So, this approach does not solve the problem if you point it to the same directory with the symlinks as it would also search in nevertheless.

closed 2022-01-06 by jaenrig-ifx