Infineon/XMC-for-Arduino #112: Arduino official servo library support

danielstuart14, 2020-01-23:
Currently, the official arduino servo library (https://github.com/arduino-libraries/Servo) doesn't support xmc boards.
When including the Servo.h header, an error will be thrown by the compiler:
'''
error: #error "This library only supports boards with an AVR, SAM, SAMD, NRF52 or STM32F4 processor."
'''
This can be replicated by using the example file on arduino IDE:
Examples -> StarterKit_BasicKit ->ServoMoodIndicator

After some research, seems like the servo library needs a plataform specific implementation for the timers used to control the servo. https://github.com/arduino-libraries/Servo/blob/8cb2d33b89d7766657454699666930b0ccf84eb9/src/Servo.h#L77

luis951, 2020-01-23:
Yeah, I have the same issue

pandayswarnam, 2020-01-23:
Infineon has its own Servo library https://github.com/Infineon/ServoC

ArkajyotiChatterjee, 2020-01-23:
Hello,
Just to add in to what  Swarnam said, the Servo Library that is in Infineon's Github supports XMC4700 Relax Kit, XMC1100 Bootkit and XMC1100 2GO as well as NodeMCU and Arduino Uno.
As you can find out in the readme that this is a generic library and does not used any thing platform or MCU specific.
Test it with your board and if there is any issue, raise a ticket here [https://github.com/Infineon/ServoC/issues](url) .
Regards,
Arka

pandayswarnam, 2020-01-23:
Hi, @danielstuart14  & @luis951  this will help you in case if there is still any problem feel free to raise the issue.

techpaul, 2020-01-23:
All the Servo implementations and others (like softwareserial) in base Arduino are hardware specific most assume AVR some assume only Arduino's only boards, as such they are very prone to timing issues as many software implementations are a hinderance to a complete application. Accuracy and jitter can be ruined by simple Serial.println of many long lines of text.

However it is very EASY to create your own Servo control (as in radio control type), part of one project I am working on for one customer has TWO servo outputs using the Hardware PWM on an XMC1300 in MY case. Depending on your application and which processor/board you are using depends on number of servo channels you can make available.

To make an RC Servo output to better resolution, you need to initialise a PWM to

1.  16 bit resolution on PWM analogWriteResolution
2.  Set timebase to 50Hz (20ms) setAnalogWriteFrequency( <pin>, 50 )
3.  Send  a duty cycle any time you want to CHANGE it setAnalogWrite( <pin>, <uint16_t value>  ) value is unsigned 16 bits

The duty cycle value is a counter value than can be EASILY calculated

For a timebase of 50Hz (20 ms) the maximum count value is 40000

This means a count is 0.5 micro seconds, so your resolution is 0.0005 ms so your RC Servo must be
1500 micro seconds (us) for idle/centre and range of 500 us to 2500 us for ABSOLUTE maximum range for an RC servo most work around the 1000 to 2000 us range, (check what your servo will do).

So passing in value of 3000 gives you idle, 5000 or 1000 gives absolute max and min

Everything else depends on your application, board being used how many servo channels you need.

I have done FPGA solutions for 16 RC or PWM channels and 6 quadraure encoder feedback channels before.

Longer term I would like to make more of the boards have more Hardware PWM o/ps and serial I/O  if possible to be MORE useable, bt like everything else it is time available to do this.

techpaul, 2020-01-23:
Please be aware the SOFTWARE only ones like ServoC rely on timing delays so your timebase may not always be 20ms in your application as your software is doing other things so cannot guarantee the repritition frequency.

Software servo driven by delay() and other delays for microseconds running all the time will be messed up by the rest of your application.

Using Hardware PWM is set it and forget it, until you need to change it

techpaul, 2020-01-23:
Software driven ServoC may not be the solution for YOUR application on an XMC board so comments above

closed 2020-01-23 by techpaul