Infineon/XMC-for-Arduino #116: Discussion required - Weird architecture statements in library properties

techpaul, 2020-01-24:
There are other things I am SLOWLY reviewing currently XMC libraries library.properties files as many seem to use weird definitions of archtectures that will cause issues best example is

          BGT24LTR11-Pulsed-Doppler-Radar

Which claims the architecture is

       architectures=cortex-m4

Shame it actually is arm as that is the branch under Infineon that the IDE and builder use.

Other classics are DSP that claims any architecture but the actual files say only XMC4700

A suggestion that needs resolution to be make things NOT clash with other Arm systems is it would be better if the builder and other parts of the board package did **NOT** use simple arm as its architecture as this is CPU ONLY description. Realistically standard arcitecture definitions like

    ARDUINO_ARCH_xxxx

Should be something like 
either
    ARDUINO_ARCH_XMC
or 
    ARDUINO_ARCH_ARM_XMC

This means if needed later XMC1 and XMC4 could be split, or other conflicting architectures easily added, as well as better IDE integration for examples and library choices

There are many other arm architectures out there which ared described as SAM, SAMD1 STM32 etc.. etc.. For compilers and build packages the architecture is the architecture of the whole chip/board not just the CPU core.

Other arch defines are
    ARDUINO_ARCH_MEGAAVR
    ARDUINO_ARCH_AVR
    ARDUINO_ARCH_SAM
    ARDUINO_ARCH_SAMD1
    ARDUINO_ARCH_STM32F4
    ARDUINO_ARCH_NRF52

Note all but he ones ending in AVR are actually ARM

techpaul, 2020-06-30:
@jaenrig-ifx 

Can I request in light of the last part above and about to release other external libraries that for next release the following change confusing architecture of 'arm' to one of the following

- XMC
- XMC-arm

One of the first external libraries will be a series of libraries to enable correct RC servo support which standard Arduino library does not support so needs to be architecture aware.

This is known to effect
- package_infineon_index.json
- .travis.yml
- existing libraries that have arm specified.

These libraries are external as not all users will even need them but some do

jaenrig-ifx, 2020-06-30:
Hi @techpaul,

Yes, that could be "xmc".  That would mean a major version change release as external libraries (out of this project) would not be compatible with the new changes (only after updating the corresponding architecture dependencies) or?

techpaul, 2020-06-30:
Yes but it will become more of a problem, just like changing where package_infineon_index.json is for users.

Yes internal libraries have to change

Are there many XMC specific external libraries yet that you know of ?

Search and replace in places for ARDUINO_ARCH_ARM tests/testdefines I can quickly modify to accept either. The important thing is to agree on what it is **first**, then libraries etc can accept either in advance, then afterwards change to xmc only

jaenrig-ifx, 2020-07-01:
Hi @techpaul,

Some changes are now in [this branch](https://github.com/Infineon/XMC-for-Arduino/tree/xmc_arch). I still wonder if we would have to change the version of each library to major or keep the current one... The backward compatibility is there from a code perspective, but not from arduino toolchain... 

Beside that, I actually find no usage of the "ARDUINO_ARCH_ARM" define other than i couple of test files: 

[image]

From our side, some external libraries are bound to the xmc-for-arduino particularities. But I am not sure if explicitly mentioned in the library.properties. I would need to check more thoroughly. External to our organization I am not aware.

techpaul, 2020-07-01:
This is NOT a major change to the XMC-for-Arduino code, it is a small change to integration, as such it is important to forewarn developers and make a note of when a change finally happens happens.

To help people now and in future I would suggest a small set of changes, first _readme.md_ to add a few lines near bottom to say someting like

This ti be a STAGED deployment first warn people and say what will work now, during changeover, and what to do after changeover.. Settings in library.properties architecure setting should be
1. Currently _arm_
2. Now and changeover period _xmc,arm_
3. After change   _xmc_

This way there is less hassle for other developers and users

This needs a Wiki page for contributors and 3rd party libraries to assist folks, I will try to create a few readme type files and Wiki page to put warnings up, as preparation for architecture change, which could be a minor release, before testing done

Note _.travis.yml will probably need to have an extra line at least once
~~~
before_initiall:
....
#         - sudo rm /home/travis/.arduino15/packages/Infineon/hardware/arm/* -rf
#         - sudo rm /home/travis/.arduino15/packages/Infineon/hardware/xmc/* -rf
~~~
Change all other references from arm to xmc

techpaul, 2020-07-01:
Some preparation for changes are in PR #154 so the details can be seen in advance of testing xmc_arch branch

Need to get my head around the .travis.yml and any other potential issues first.

jaenrig-ifx, 2020-07-02:
Nice. The .travis.yml is commented because the output log is too long and Travis CI stops with error.  We have to implement that part more silently...

closed 2022-01-06 by jaenrig-ifx