Infineon/XMC-for-Arduino #120: V1.4 release not available

techpaul, 2020-01-28:
Been checking on a few platforms the release of V1.4.0 and waiting for patches has not worked

package.json still says V1.3.0

index.json still says V1.30 as last version,

Something went wrong with the release, might need redoing as not being picked in Arduino IDE Board Manager as any version newer than V1.3.0

techpaul, 2020-01-29:
@jaenrig-ifx @mhollfelder 

Can I suggest the following some of which I do not have access to

1.  Merge latest fixes to/from develop into master
2.  Uodate package.json to show **V1.4.1**
3.  Add necessary part to https://github.com/Infineon/Assets/releases/download/current/package_infineon_index.json
4.  Any other release steps
5.  Create new version V1.4.1 here

Class V1.4.0 as a version no longer available from IDE

jaenrig-ifx, 2020-01-29:
Since the current version is not availalble anyway. I will delete the tag and release it again.

techpaul, 2020-01-29:
Deleting the tag may help, but be aware V1.30 in IDE appears to be offering currently quite a lot of V1,4 files so V1,3 potentially be a bit of multi version depending on when loaded.

Please make sure the JSON files are updated.

jaenrig-ifx, 2020-01-29:
Should be now ready. Definitively we need some release automation @mhollfelder :)

techpaul, 2020-01-29:
Currently visible and updatable here will close later after a test

techpaul, 2020-01-29:
Correctly picking up V1.4.0 now

closed 2020-01-29 by techpaul